MarlinFirmware/Marlin #22397: [BUG] status_screen_DOGM.cpp fails to compile if STATUS_COMBINE_HEATERS is enabled (LPC1769)

NeoMod, 2021-07-19:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When compiling Marlin from both the 2.0.x and the bugfix-2-0-x branch with `#define STATUS_COMBINE_HEATERS`  enabled, the compilation fails with the following error:

```
Compiling .pio\build\LPC1769\src\src\lcd\menu\menu_advanced.cpp.o
Marlin\src\lcd\dogm\status_screen_DOGM.cpp: In function 'void _draw_hotend_status(heater_id_t, bool)':
Marlin\src\lcd\dogm\status_screen_DOGM.cpp:274:26: error: 'STATUS_HOTEND_BYTEWIDTH' was not declared in this scope; did you mean 'STATUS_HEATERS_BYTEWIDTH'?
  274 |                     bw = STATUS_HOTEND_BYTEWIDTH(heater_id);
      |                          ^~~~~~~~~~~~~~~~~~~~~~~
      |                          STATUS_HEATERS_BYTEWIDTH
Marlin\src\lcd\dogm\status_screen_DOGM.cpp:260:34: error: 'status_hotend_a_bmp' was not declared in this scope; did you mean 'status_heaters_bmp'?
  260 |       #define HOTEND_BITMAP(N,S) status_hotend_a_bmp
      |                                  ^~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\dogm\status_screen_DOGM.cpp:283:76: note: in expansion of macro 'HOTEND_BITMAP'
  283 |           u8g.drawBitmapP(hx, STATUS_HEATERS_Y, bw, STATUS_HEATERS_HEIGHT, HOTEND_BITMAP(TERN(HAS_MMU, active_extruder, heater_id), isHeat));
```

Disabling `#define STATUS_COMBINE_HEATERS` makes the build successful.

### Bug Timeline

New

### Expected behavior

Build should be complete without errors.

### Actual behavior

Build fails, see the error described above.

### Steps to Reproduce

1. Select LCP1769 as env
2. Select MKS_SGEN_L_V2 as board _(not sure it matters, anyway)_
3. Enable `#define STATUS_COMBINE_HEATERS` in Configuration_adv.h
4. Try to build 

### Version of Marlin Firmware

Marlin 2.0.9.1 

### Printer model

Custom

### Electronics

MKS SGEN L v2, TMC2209

### Add-ons

MKS LCD12864B

### Your Slicer

Prusa Slicer

ellensp, 2021-07-19:
default Config (ramps atmega2560) + REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER + STATUS_COMBINE_HEATERS does it also. From the error it probably is platform independent

ellensp, 2021-07-19:
Quick fix. in Marlin/src/lcd/dogm/dogm_Statusscreen.h
add `#include "status/hotend.h"` after the other two #includes

NeoMod, 2021-07-19:
Thanks a lot. I just tested it, and it worked.

In the end, it also helped to add `#include "status/bed.h"` to avoid other errors. The build succeeded, and the machine is operational, although the LCD isn't displaying anything. I'm not sure if this is related, probably not, so I think this could be closed.

ellensp, 2021-07-20:
128x64 lcds have two different pin outs. They are the same, but they key is rotated 180 degrees (ie effectively the pins are rotated 180 degrees)
Check that gnd goes to gnd and 5v goes to 5v.

ellensp, 2021-07-20:
The code has changed since I last looked at it..  the above fix is not correct. looking deeper

ellensp, 2021-07-20:
Ive created a PR to fix it properly

ellensp, 2021-07-21:
This has been fixed

closed 2021-07-21 by ellensp