MarlinFirmware/Marlin #22400: [BUG] xyz move breaks after G29 if mode is `AUTO_BED_LEVELING_LINEAR`

thebestnom, 2021-07-19:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

On ender 3 v2 with bltouch
After fiddling with `AUTO_BED_LEVELING_LINEAR` after running G29 I can't control the xyz (tested with lcd control)
changing it to  `AUTO_BED_LEVELING_BILINEAR` fixed that
as far as I guess changing this shouldn't affect the after effects of G29

### Bug Timeline

Tested only on new bugfix-2.0.x

### Expected behavior

after G29 in `AUTO_BED_LEVELING_LINEAR` you should be able to control xyz

### Actual behavior

After G29 can't control z

### Steps to Reproduce

1. compile with `AUTO_BED_LEVELING_LINEAR` uncommented
2. run auto level
3. try moving xyz in control

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

Creality Ender 3 v2

### Electronics

Stock 

### Add-ons

BLTouch

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

The configuration here is with `AUTO_BED_LEVELING_BILINEAR` which works, only 
changing this to `AUTO_BED_LEVELING_LINEAR` breaks it
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6843215/Configuration.zip)

thisiskeithb, 2021-07-19:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/22128

closed 2021-07-19 by thisiskeithb

thebestnom, 2021-07-19:
Oh my god, I search for this for ages! sorry for opening duplicate😓