MarlinFirmware/Marlin #22404: [BUG] Can't compile: U8glib-HAL.h: No such file or directory

SinisterRj, 2021-07-20:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I can't compile this latest Marlin (2.0.9.1) and actual bugfix branch on VSCode+PIO in Windows 10. No problems with an older bugfix, from march, 23th. I tried to manually download U8glib-HAL from here: https://github.com/MarlinFirmware/U8glib-HAL and the compilation was successfull, but when inserted the SD card on my printer the LCD stucks blank (no boot). I had to reflash it again with the older version.

Follow terminal info:

`Processing STM32F103RET6_creality (platform: ststm32@~12.0; board: genericSTM32F103RE; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RE.html
PLATFORM: ST STM32 (12.0.0) > STM32F103RE (64k RAM. 512k Flash)       
HARDWARE: STM32F103RET6 72MHz, 64KB RAM, 512KB Flash
DEBUG: Current (jlink) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32-maple 3.10000.201129 (1.0.0)
 - tool-stm32duino 1.0.2
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 28 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SoftwareSerialM> 1.0.0
|-- <U8glib-HAL> 0.4.5
|   |-- <Wire> 1.0
|-- <STM32ADC> 1.0
|-- <EEPROM>
|-- <USBComposite for STM32F1> 0.99
|-- <Wire> 1.0
|-- <Servo(STM32F1)> 1.1.2
Building in release mode

warning: Calling missing SConscript without error is deprecated.
Transition by adding must_exist=0 to SConscript calls.
Missing SConscript 'buildroot\share\PlatformIO\scripts\STM32F103RET6_creality.py'
File "C:\Users\Rodrigo\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 185, in <module>
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduinoVariant\board.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduinoVariant\wirish\boards.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduinoVariant\wirish\boards_setup.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduinoVariant\wirish\start.S.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduinoVariant\wirish\start_c.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduinoVariant\wirish\syscalls.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\HardwareTimer.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\Print.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\WString.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\avr\dtostrf.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\cxxabi-compat.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\ext_interrupts.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\hooks.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\itoa.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\adc.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\adc_f1.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\bkp_f1.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\dac.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\dma.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\dma_f1.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\exc.S.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\exti.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\exti_f1.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\flash.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\fsmc_f1.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\gpio.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\gpio_f1.c.o
[... 119 lines cut ...]
2.0.9.1

### Printer model

Creality Ender 3

### Electronics

Creality 4.2.2 (32 bit bstock board)

### Add-ons

Run out sensor + Bl touch

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[config.zip](https://github.com/MarlinFirmware/Marlin/files/6844598/config.zip)

ellensp, 2021-07-20:
Cannot replicate
The library is there

```
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing U8glib-HAL @ ~0.5.0
Library Manager: Warning! More than one package has been found by U8glib-HAL @ ~0.5.0 requirements:
 - marlinfirmware/U8glib-HAL @ 0.5.0
 - thinkyhead/U8glib-HAL @ 0.4.5
Library Manager: Please specify detailed REQUIREMENTS using package owner and version (showed above) to avoid name conflicts
Library Manager: U8glib-HAL @ 0.5.0 has been installed!
```

And if you follow the link, it shows you platfomio knows about this library 
https://platformio.org/lib/search?query=header:U8glib-HAL.h

qrstu, 2021-07-20:
check the features.ini there the "HAS_MARLINUI_U8GLIB                    = U8glib-HAL@~0.5.0" has changed.
I had the same issues. Then it was working
Regards
Marton

ellensp, 2021-07-20:
yea but that is definitely in bugfix, which users says also didn't work.

ellensp, 2021-07-20:
> ### Did you test the latest `bugfix-2.0.x` code?
> Yes, and the problem still exists.
> 
> ### Bug Description
> I can't compile this latest Marlin (2.0.9.1) and actual bugfix branch on VSCode+PIO in Windows 10. No problems with an older bugfix, from march, 23th. I tried to manually download U8glib-HAL from here: https://github.com/MarlinFirmware/U8glib-HAL and the compilation was successfull, but when inserted the SD card on my printer the LCD stucks blank (no boot). I had to reflash it again with the older version.
> 
> Follow terminal info:
> 
> `Processing STM32F103RET6_creality (platform: ststm32@~12.0; board: genericSTM32F103RE; framework: arduino) ----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
> CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RE.html
> PLATFORM: ST STM32 (12.0.0) > STM32F103RE (64k RAM. 512k Flash)
> HARDWARE: STM32F103RET6 72MHz, 64KB RAM, 512KB Flash
> DEBUG: Current (jlink) External (blackmagic, cmsis-dap, jlink, stlink)
> PACKAGES:
> 
> * framework-arduinoststm32-maple 3.10000.201129 (1.0.0)
> * tool-stm32duino 1.0.2
> * toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
>   Converting Marlin.ino
>   LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
>   LDF Modes: Finder ~ chain, Compatibility ~ soft
>   Found 28 compatible libraries
>   Scanning dependencies...
>   Dependency Graph
>   |--  1.0.0
>   |--  0.4.5
>   |   |--  1.0
>   |--  1.0
>   |-- 
>   |--  0.99
>   |--  1.0
>   |-- <Servo(STM32F1)> 1.1.2
>   Building in release mode
> 
> warning: Calling missing SConscript without error is deprecated.
> Transition by adding must_exist=0 to SConscript calls.
> Missing SConscript 'buildroot\share\PlatformIO\scripts\STM32F103RET6_creality.py'
> File "C:\Users\Rodrigo.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 185, in
> Compiling .pio\build\STM32F103RET6_creality\FrameworkArduinoVariant\board.cpp.o
> Compiling .pio\build\STM32F103RET6_creality\FrameworkArduinoVariant\wirish\boards.cpp.o
> Compiling .pio\build\STM32F103RET6_creality\FrameworkArduinoVariant\wirish\boards_setup.cpp.o
> Compiling .pio\build\STM32F103RET6_creality\FrameworkArduinoVariant\wirish\start.S.o
> Compiling .pio\build\STM32F103RET6_creality\FrameworkArduinoVariant\wirish\start_c.c.o
> Compiling .pio\build\STM32F103RET6_creality\FrameworkArduinoVariant\wirish\syscalls.c.o
> Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\HardwareSerial.cpp.o
> Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\HardwareTimer.cpp.o
> Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\IPAddress.cpp.o
> Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\Print.cpp.o
> Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\Stream.cpp.o
> Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\WString.cpp.o
> Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\avr\dtostrf.c.o
> Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\cxxabi-compat.cpp.o
> Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\ext_interrupts.cpp.o
> Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\hooks.c.o
> Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\itoa.c.o
> Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\adc.c.o
> Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\adc_f1.c.o
> Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\bkp_f1.c.o
> Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\dac.c.o
> Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\dma.c.o
> Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\dma_f1.c.o
> Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\exc.S.o
> Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\exti.c.o
> Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\exti_f1.c.o
> Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\flash.c.o
> Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\fsmc_f1.c.o
> Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\gpio.c.o
> Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\gpio_f1.c.o
> Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\i2c.c.o
> Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\i2c_f1.c.o
[... 102 lines cut ...]
> 
> ### Steps to Reproduce
> 1. Download marlin source code;
> 2. Compile on Visual Studio code.
> 
> ### Version of Marlin Firmware
> 2.0.9.1
> 
> ### Printer model
> Creality Ender 3
> 
> ### Electronics
> Creality 4.2.2 (32 bit bstock board)
> 
> ### Add-ons
> Run out sensor + Bl touch
> 
> ### Your Slicer
> Cura
> 
> ### Host Software
> OctoPrint
> 
> ### Additional information & file uploads
> [config.zip](https://github.com/MarlinFirmware/Marlin/files/6844598/config.zip)

ellensp, 2021-07-20:
provided log shows incomplete list of libraries  (nor names)

```
Dependency Graph
|-- 1.0.0
|-- 0.4.5
| |-- 1.0
|-- 1.0
|--
|-- 0.99
|-- 1.0
|-- <Servo(STM32F1)> 1.1.2
```

but from this your clearly using 0.4.5 U8glib, which is still called U8glib not  U8glib-HAL as it it become in 0.5.0

So you have miss matched code and build environment,

closed 2021-07-20 by ellensp

SinisterRj, 2021-07-20:
Sorry, I was using an old platformio.ini. With the right one (just changed this line: default_envs = STM32F103RET6_creality) now I have a different error:

```
Library Manager: Installing U8glib-HAL @ ~0.5.0
Library Manager: Warning! More than one package has been found by U8glib-HAL @ ~0.5.0 requirements:
 - marlinfirmware/U8glib-HAL @ 0.5.0
 - thinkyhead/U8glib-HAL @ 0.4.5
Library Manager: Please specify detailed REQUIREMENTS using package owner and version (showed above) to avoid name conflicts
Library Manager: U8glib-HAL @ 0.5.0 has been installed!
Found 13 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.5.0
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <SPI> 1.0
|-- <Servo> 1.1.2
|-- <EEPROM> 2.0.1
|-- <SoftwareSerialM> 1.0.0
|-- <IWatchdog> 1.0.0
|-- <Wire> 1.0
Building in release mode

...
[Terminal.txt](https://github.com/MarlinFirmware/Marlin/files/6848225/Terminal.txt)
[Terminal.zip](https://github.com/MarlinFirmware/Marlin/files/6848231/Terminal.zip)

Compiling .pio\build\STM32F103RET6_creality\libe5d\IWatchdog\IWatchdog.cpp.o
In file included from .pio\libdeps\STM32F103RET6_creality\SoftwareSerialM\HAL_softserial_STM32.cpp:27:
.pio\libdeps\STM32F103RET6_creality\SoftwareSerialM\HAL_softserial_STM32.h:37:36: error: variable or field 'SoftSerial_Handler' declared void
   37 | extern "C" void SoftSerial_Handler(stimer_t *htim);
      |                                    ^~~~~~~~
.pio\libdeps\STM32F103RET6_creality\SoftwareSerialM\HAL_softserial_STM32.h:37:36: error: 'stimer_t' was not declared in this scope; did you mean 'timer_t'?
   37 | extern "C" void SoftSerial_Handler(stimer_t *htim);
      |                                    ^~~~~~~~
      |                                    timer_t
.pio\libdeps\STM32F103RET6_creality\SoftwareSerialM\HAL_softserial_STM32.h:37:46: error: 'htim' was not declared in this scope; did you mean 'tm'?
   37 | extern "C" void SoftSerial_Handler(stimer_t *htim);
      |                                              ^~~~
      |                                              tm
.pio\libdeps\STM32F103RET6_creality\SoftwareSerialM\HAL_softserial_STM32.cpp:69:1: error: 'stimer_t' does not name a type; did you mean 'timer_t'?
   69 | stimer_t SSTimerHandle;
      | ^~~~~~~~
      | timer_t
.pio\libdeps\STM32F103RET6_creality\SoftwareSerialM\HAL_softserial_STM32.cpp: In function 'void HAL_softSerial_init()':
.pio\libdeps\STM32F103RET6_creality\SoftwareSerialM\HAL_softserial_STM32.cpp:74:3: error: 'SSTimerHandle' was not declared in this scope
   74 |   SSTimerHandle.timer = SS_TIMER_DEV;
      |   ^~~~~~~~~~~~~
compilation terminated due to -fmax-errors=5.
In file included from .pio\libdeps\STM32F103RET6_creality\SoftwareSerialM\HAL_softserial.h:28,
                 from .pio\libdeps\STM32F103RET6_creality\SoftwareSerialM\SoftwareSerial.cpp:40:
.pio\libdeps\STM32F103RET6_creality\SoftwareSerialM\HAL_softserial_STM32.h:37:36: error: variable or field 'SoftSerial_Handler' declared void
   37 | extern "C" void SoftSerial_Handler(stimer_t *htim);
      |                                    ^~~~~~~~
.pio\libdeps\STM32F103RET6_creality\SoftwareSerialM\HAL_softserial_STM32.h:37:36: error: 'stimer_t' was not declared in this scope; did you mean 'timer_t'?
   37 | extern "C" void SoftSerial_Handler(stimer_t *htim);
      |                                    ^~~~~~~~
      |                                    timer_t
.pio\libdeps\STM32F103RET6_creality\SoftwareSerialM\HAL_softserial_STM32.h:37:46: error: 'htim' was not declared in this scope; did you mean 'tm'?
   37 | extern "C" void SoftSerial_Handler(stimer_t *htim);
      |                                              ^~~~
      |                                              tm
.pio\libdeps\STM32F103RET6_creality\SoftwareSerialM\HAL_softserial_STM32.h:35:71: error: variable or field 'SoftSerial_Handler' declared void
   35 | #define HAL_SOFTSERIAL_TIMER_ISR() extern "C" void SoftSerial_Handler(stimer_t *htim)
      |                                                                       ^~~~~~~~
.pio\libdeps\STM32F103RET6_creality\SoftwareSerialM\SoftwareSerial.cpp:218:1: note: in expansion of macro 'HAL_SOFTSERIAL_TIMER_ISR'
  218 | HAL_SOFTSERIAL_TIMER_ISR() {
      | ^~~~~~~~~~~~~~~~~~~~~~~~
.pio\libdeps\STM32F103RET6_creality\SoftwareSerialM\HAL_softserial_STM32.h:35:71: error: 'stimer_t' was not declared in this scope; did you mean 'timer_t'?
   35 | #define HAL_SOFTSERIAL_TIMER_ISR() extern "C" void SoftSerial_Handler(stimer_t *htim)
      |                                                                       ^~~~~~~~
.pio\libdeps\STM32F103RET6_creality\SoftwareSerialM\SoftwareSerial.cpp:218:1: note: in expansion of macro 'HAL_SOFTSERIAL_TIMER_ISR'
  218 | HAL_SOFTSERIAL_TIMER_ISR() {
      | ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pio\build\STM32F103RET6_creality\libec7\SoftwareSerialM\HAL_softserial_STM32.cpp.o] Error 1
*** [.pio\build\STM32F103RET6_creality\libec7\SoftwareSerialM\SoftwareSerial.cpp.o] Error 1
====================================================================
```

ellensp, 2021-07-20:
delete the `Marlin/.pio/` directory

ellensp, 2021-07-20:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

SinisterRj, 2021-07-20:
Ok, sorry.