MarlinFirmware/Marlin #22406: Add support for Big Tree Tech TFT 35 V3.0 Touchscreen

Mightywren, 2021-07-20:
### Are you looking for hardware support?

BTT SKR Mini E3 V2 needs to be able to connect to the BTT TFT35 V3 .0 Touchscreen which is not listed in configeration.h in the Marlin 2.0.x Bug fix as it is becoming a popular brand

### Describe the feature you want

BTT TFT35 V3 touchscreen should be included in the Graphical TFTs list

ellensp, 2021-07-20:
RTFM https://github.com/bigtreetech/BTT-TFT35-E3-V3.0/blob/master/Hardware/BTT%20TFT35-E3%20V3.0%20User%20Guide.pdf

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-07-20 by ellensp

ellensp, 2021-07-20:
It is a supported screen on that controller.
Touch requires a serial port only 
The LCD emulates twp other tyres  of LCD, you select the appropriate emulated display