MarlinFirmware/Marlin #22408: [BUG] SERIAL_ECHOLNPAIR no longer reliable on Archim when printer is busy

marciot, 2021-07-20:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I often use `SERIAL_ECHOLNPAIR` for debugging code, in fact, it is pretty much the only way I have of debugging things in Marlin. `SERIAL_ECHOLNPAIR` used to be pretty reliable for this, but now I've noticed that these messages are not showing up when the printer is printing. Instead, the messages show up normally while the printer is doing something, but as soon as I start a print all I see is "echo:busy: processing" in the serial console.

I have a UI debugging mode that causes a lot of messages to be printed using `SERIAL_ECHOLNPAIR` anytime the user interact with the UI. Now I am able to interact with the UI, but I don't see those messages getting print at all when the printer is actively printing. When I cancel the print, the messages start to show up again. It almost seems as if the behavior has changed to only allow serial messages through when the printer isn't very busy.

This behavior pretty much makes it impossible for me to debug anything that happens when a print is ongoing. Is there anyway to restore the old behavior? Or is there a lower level print routine that is guaranteed to work?

### Expected behavior

Calling `SERIAL_ECHOLNPAIR` should reliably print the serial console.

### Actual behavior

Calling `SERIAL_ECHOLNPAIR` only happens when a print isn't ongoing.

### Version of Marlin Firmware

bugfix-2.0.x

### Electronics

Ultimachine Archim 2 board

rhapsodyv, 2021-07-21:
Hum... if we are somehow buffering SERIAL_ECHOes, that would explain some random serial issues some users have reported, because it will overflow sometime...

thinkyhead, 2021-07-30:
Serial printing in most contexts will go to the default serial port, which might be the screen on your machines. Serial printing inside of G-code handling will go to the serial port that sent the G-code command. I believe we also set it up at some point to default the echo of serial output in the normal context to go to ALL serial ports. So, do some testing to see if the serial port index can be initialized at startup so it points to the serial console or ALL instead, and after that it should always fall back to that output outside of G-code handlers.

thinkyhead, 2021-07-30:
…Also, take a look at the behavior of the SERIAL REDIRECT stuff to make sure it is always being restored properly after usage. It is supposed to always be restored at the end of the current block.

closed 2021-10-08 by github-actions[bot]