MarlinFirmware/Marlin #22410: [BUG] Marlin 2.0.9.1 M32 no longer working

roel8032, 2021-07-21:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

M32 does no longer jump into sub program. Always worked in 2.07 and before (not tested 2.08).

Program who is calling M32 stops at the point it should jump into sub program.

### Bug Timeline

New

### Actual behavior

Stopping main program, where it should jump into sub program.

### Version of Marlin Firmware

2.0.9.1

### Printer model

Infitary M508

### Your Slicer

Cura

roel8032, 2021-07-29:
Hello, Is somebody working on the problem?

thinkyhead, 2021-07-30:
M32 is pretty brittle and has been worked on many times, sometimes working and then sometimes breaking. Please scrub through some old versions of Marlin and test them to see if you can determine the Marlin version and then the date range where the bug first appeared.

roel8032, 2021-07-30:
> M32 is pretty brittle and has been worked on many times, sometimes working and then sometimes breaking. Please scrub through some old versions of Marlin and test them to see if you can determine the Marlin version and then the date range where the bug most recently appeared.

In Marlin 2.0.7 it is working. I skipped 2.0.8, so i don't know if it works in this version.

thinkyhead, 2021-07-30:
So, okay, somewhere in the last year or so it got broken. To narrow it down further, we can try testing Marlin from 6 months ago. If it is still broken at 6 months ago, then we test Marlin from 9 months ago. But if it is not broken at 6 months ago, then we check Marlin from 3 months ago. By continuing to bisect in this manner, after about 3 to 5 tests we will just about know the exact day and hour when the problem first appears.

roel8032, 2021-08-01:
> So, okay, somewhere in the last year or so it got broken. To narrow it down further, we can try testing Marlin from 6 months ago. If it is still broken at 6 months ago, then we test Marlin from 9 [...]

Dit you find the problem already?

roel8032, 2021-09-03:
> So, okay, somewhere in the last year or so it got broken. To narrow it down further, we can try testing Marlin from 6 months ago. If it is still broken at 6 months ago, then we test Marlin from 9 [...]

Is the problem solved already?

DerAndere1, 2021-09-04:
@roel8032 : thinkyhead had asked YOU to check different versions and report your findings. We are all volunteers and rarely find the motivation/ time to narrow down a problem when even You  do not have that motivation to help even though you suffer from the problem most. Also, we cannot test this as we do not have the hardware.

roel8032, 2021-09-04:
> @roel8032 : thinkyhead had asked YOU to check different versions and report your findings. We are all volunteers and rarely find the motivation/ time to narrow down a problem when even You do not have that motivation to help although you suffer from the problem most.

He wrote: WE can try, not YOU can try!

roel8032, 2021-09-14:
Little change in behavior with latest Marlin bug version when calling subprogram with M32: Starts a move in X+ direction, while a Y+ move is the first move in the subprogram.

roel8032, 2022-03-03:
Not solved yet.

thisiskeithb, 2022-03-03:
@roel8032 Please follow thinkyhead’s instructions and report back: https://github.com/MarlinFirmware/Marlin/issues/22410#issuecomment-889626359

roel8032, 2022-03-03:
I do not understand how to do this. I do not have al that versions.

thisiskeithb, 2022-03-03:
You can download previous versions of Marlin from https://github.com/MarlinFirmware/Marlin/releases

closed 2022-05-14 by github-actions[bot]