MarlinFirmware/Marlin #22413: [BUG] BOARD_MKS_ROBIN_NANO_V3 and MKS_TS35_V2_0 with TFT_LVGL_UI compile error

ellensp, 2021-07-21:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

The following fails to compile

#define SERIAL_PORT 3
#define MOTHERBOARD BOARD_MKS_ROBIN_NANO_V3
#define SDSUPPORT
#define MKS_TS35_V2_0
#define TFT_LVGL_UI
#define TOUCH_SCREEN

### Bug Timeline

new

### Expected behavior

It should compile

### Actual behavior

It errors with
```
Compiling .pio/build/mks_robin_nano_v3/src/src/lcd/extui/mks_ui/gb2312_puhui16.cpp.o
In file included from Marlin/src/lcd/extui/mks_ui/../../../inc/MarlinConfigPre.h:37,
                 from Marlin/src/lcd/extui/mks_ui/draw_ready_print.cpp:22:
Marlin/src/lcd/extui/mks_ui/draw_ready_print.cpp: In function 'void lv_draw_ready_print()':
Marlin/src/lcd/extui/mks_ui/draw_ready_print.cpp:137:24: error: 'mks_test_flag' was not declared in this scope
  137 |   if (TERN0(SDSUPPORT, mks_test_flag == 0x1E)) {
```

### Steps to Reproduce

1) use provided config
2) compile
3) note that it errors

### Version of Marlin Firmware

latest bugfix

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6853948/Configuration.zip)

Related to issue: https://github.com/MarlinFirmware/Marlin/issues/21069

ellensp, 2021-07-21:
The following seems to fix it, but m not confident its th correct thing to do. 

```DIFF
diff --git a/Marlin/src/lcd/extui/mks_ui/draw_ready_print.cpp b/Marlin/src/lcd/extui/mks_ui/draw_ready_print.cpp
index 6f24d81b49..78e01842b2 100644
--- a/Marlin/src/lcd/extui/mks_ui/draw_ready_print.cpp
+++ b/Marlin/src/lcd/extui/mks_ui/draw_ready_print.cpp
@@ -126,7 +126,7 @@ void lv_draw_ready_print() {
   ZERO(disp_state_stack._disp_state);
   scr = lv_screen_create(PRINT_READY_UI, "");

-  if (TERN0(SDSUPPORT, mks_test_flag == 0x1E)) {
+  if (TERN0(MKS_TEST, mks_test_flag == 0x1E)) {
     // Create image buttons
     buttonTool = lv_imgbtn_create(scr, "F:/bmp_tool.bin", event_handler, ID_TOOL);
```

ellensp, 2021-07-21:
Seems to work ok. when MKS_TEST is not defined, but is quite broken when it is.

thisiskeithb, 2021-07-21:
https://github.com/MarlinFirmware/Marlin/pull/22414 was merged. Closing.

closed 2021-07-21 by thisiskeithb