MarlinFirmware/Marlin #22415: [BUG] Random z-offset without ABL on Ender 3 V2 with BTT SKR Mini E3 V2.0 (GD chip) and DWIN Display

blazewicz, 2021-07-21:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

My printer isn't consistent in printing first layer, it randomly adds tiny z-offset at the beginning of the print. Sometimes it's clearly visible, like 5 mm, and sometimes its just enough to ruin the first layer. I managed to capture one event when it added maybe 0.2 mm to the z-offset.

This is how it should look, and sometimes does:

[image]

And this is how it looks when it goes crazy:

[image]

On both videos it is the same printer and same g-code, same fragment. These where 2 consecutive attempts to print. I removed the filament from the nozzle so the gap is better visible.

There is no problem with z-height accuracy otherwise. If I adjust bed height on the beginning of the print it prints well. 

I don't have any ABL, I use stock Z-Stop switch type. It doesn't seem to be issue with the Z-Stop, because G28 always ends with nozzle gently touching the bed.

It can be seen just after the first line following initial G28 and happens on all files, no matter the slicer, here is my start G-code:

```gcode
M117 Start G-code ...

G90    ; Absolute Positioning
G92 E0 ; Reset Extruder

G28                             ; Home all axes
G0 Z2.0 F3000                   ; Move Z Axis up little to prevent scratching of Heat Bed
G0 X0.1 Y20 Z0.3 F5000.0        ; Move to start position
G1 X0.1 Y200.0 Z0.3 E15 F1500.0 ; Draw the first line
G0 X0.4 Y200.0 Z0.3 F5000.0     ; Move to side a little
G1 X0.4 Y20 Z0.3 E30 F1500.0    ; Draw the second line
G92 E0                          ; Reset Extruder
G0 Z2.0 F3000                   ; Move Z Axis up little to prevent scratching of Heat Bed
G0 X5 Y20 Z0.3 F5000.0          ; Move over to prevent blob squish

M117 Printing ...
```

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

Ender 3 V2

### Electronics

BTT SKR Mini E3 V2.0 (GD32F103RET6 chip)

### Host Software

Other (explain below)

### Additional information & file uploads

Happens when printing from SD card and from Octoprint.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6856711/Configuration.zip)

DerAndere1, 2021-07-22:
Thanks for the report. If you downloaded your firmware before 20th July 2021, please test again with current Marlin bugfix-2.0.x (after commit https://github.com/MarlinFirmware/Marlin/commit/154decfc66c30808cfe7320e542fcf90427d7176 ).

blazewicz, 2021-07-24:
Ok, I think I got this. I've bathed the z-stop in IPA and then sprayed it with preparation for electical contants and it seems that the issue is gone.

It probably was only z-homing issue, and my observations that homing worked at all times were probably false.

closed 2021-07-24 by blazewicz