MarlinFirmware/Marlin #22416: [BUG] Z axis stops without deceleration even if max acceleration is set

Frictionpowered, 2021-07-21:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

I'm tuning Z-hops on my Prusa clone and noticed that the Z steppers sound "suspicious" when i change the Z acceleration and max speed different from the factory defaults. So started experimenting and it looks like the same behaviour as described here:
https://github.com/MarlinFirmware/Marlin/issues/15797
Basically, instead of accelerating and decelerating gently, the Z steppers accelerate but stop very suddenly, without any deceleration at all.

My current settings are:
M201 X500 Y500 Z10	;set max acceleration (default=1400)
M203 Z4
M204 S500
M205 X7	

Having looked into the source code, i *suspect* the problem might be in planner.cpp, if the move is short and doesn't reach the top speed (see plateau_steps < 0, intersection_distance).

### Bug Timeline

probably the same as https://github.com/MarlinFirmware/Marlin/issues/15797

### Expected behavior

Z moves (including short Z hops) should not stop suddenly if acceleration is set; they should accelerate AND decelerate as configured, just like X and Y moves do.

### Version of Marlin Firmware

Marlin V1

### Printer model

CTC Prusa I3 Pro B

### Electronics

stock GT2560

### Add-ons

-

### Your Slicer

Cura

### Host Software

Cura

### Additional information & file uploads

[AAA Beer crate Mod-ZHopBug.zip](https://github.com/MarlinFirmware/Marlin/files/6857581/AAA.Beer.crate.Mod-ZHopBug.zip)

Print this file and watch/listen to the Z stepper motors. It's pretty obvious.

XDA-Bam, 2021-07-27:
Hi. Two things:
- Please add your config files in a ZIP, otherwise debugging won't be possible
- You write "Marlinn V1" - Do you mean v1.1.9? If so, that version is deprecated and bugs won't be fixed anymore. Please check if the bug is still present in bugfix-2.0.x

thinkyhead, 2021-07-30:
Is the entire file really needed, or is the issue demonstrated with just this…?
```gcode
G0 F12000 Z3.300000
```

thinkyhead, 2021-07-30:
What is your Z Jerk setting like? How does changing the Z jerk affect the results?

(Note: Only testing Marlin bugfix-2.0.x of course.)

Frictionpowered, 2021-07-31:
Thanks, i'll do some more testing. And more importantly, try to whittle down the gcode to as small as possible, because i agree surely not the whole thing is necessary. :) 
In the meantime I'm also looking into how to build the correct Marlin config for my printer and how to backup the existing firmware because tbh i don't like the possibility of messing up my printer that otherwise works pretty well.

Frictionpowered, 2021-07-31:
Ok i made a much smaller test object; it's only about 10 layers of two rectangles, hopping back and forth between the two. I decreased the Z acceleration and increased the Z speed to slow the problem more clearly, making Z moves audibly and visibly very smooth on the z steppers EXCEPT when it does the sharp stop, which is even more noticeable because of the higher top speed.
On my Marlin, the Z lifts before the moves from the left to the right block clearly do produce the sharp stops, but the lifts before the right to left moves accelerate and decelerate gently as expected. Pretty interesting.

(I'm probably going to replace my stepper driver with a quiet TMC2208 like i did on the X/Y already so wanted to do this while i can hear it so sharply with the standard driver.)

Here's the gcode file
[Z-hop test 4.zip](https://github.com/MarlinFirmware/Marlin/files/6911563/Z-hop.test.4.zip)

...and here are my settings as returned by the M115 and M503 from the start gcode.

< FIRMWARE_NAME:Marlin V1; Sprinter/grbl mashup for gen6 FIRMWARE_URL:http://firmware.ultimaker.com PROTOCOL_VERSION:1.0 MACHINE_TYPE:I3 Pro B  EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
< echo:Steps per unit:
< echo:  M92 X78.90 Y78.90 Z2560.00 E95.00
< echo:Maximum feedrates (mm/s):
< echo:  M203 X400.00 Y400.00 Z4.00 E45.00
< echo:Maximum Acceleration (mm/s2):
< echo:  M201 X600 Y600 Z15 E80000
< echo:Acceleration: S=acceleration, T=retract acceleration
< echo:  M204 S500.00 T5000.00
< echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
< echo:  M205 S0.00 T0.00 B20000 X7.00 Z0.10 E5.00
< echo:Home offset (mm):
< echo:  M206 X0.00 Y0.00 Z0.00
< echo:PID settings:
< echo:   M301 P19.86 I1.00 D98.93