MarlinFirmware/Marlin #22429: [BUG] simulator not build on windows10

sl1pkn07, 2021-07-24:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

simulator for windows(10) not build

### Bug Timeline

new

### Expected behavior

build ok

### Actual behavior

fail build

### Steps to Reproduce

1 -> download marlin souurces from git and checkout bugfix-2.0.x branch
2 -> set `default_envs = simulator_windows` in the `platformio.ini`
3 -> set `BOARD_LINUX_RAMPS` in `canfiguration.h`
4 -> install msys2 and perform system upgrade. and install the simulator dependiences
4 -> run `pio run`

### Version of Marlin Firmware

git

### Printer model

simulator

### Electronics

simulator

### Add-ons

none

### Additional information & file uploads

~~~
PS C:\Users\sL1pKn07\Documents\marlin-orig> pio run
Processing simulator_windows (platform: native; framework: )
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ off
Library Manager: Installing git+https://github.com/p3p/MarlinSimUI.git
git version 2.32.0.windows.2
Cloning into 'C:\Users\sL1pKn07\.platformio\.cache\tmp\pkg-installing-1hzwg65w'...
Library Manager: MarlinSimUI @ 0.0.1+sha.0e047cc has been installed!
Library Manager: Installing dependencies...
Library Manager: Installing git+https://github.com/p3p/imgui.git#pio_docking
git version 2.32.0.windows.2
Cloning into 'C:\Users\sL1pKn07\.platformio\.cache\tmp\pkg-installing-hm90q3bz'...
Library Manager: imgui @ 1.81.1+sha.0ab0689 has been installed!
Library Manager: Installing git+https://github.com/p3p/implot.git#pio_master
git version 2.32.0.windows.2
Cloning into 'C:\Users\sL1pKn07\.platformio\.cache\tmp\pkg-installing-3bijdz0i'...
Library Manager: implot @ 0.8.1+sha.d26e0f5 has been installed!
Library Manager: Installing dependencies...
Library Manager: Installing https://github.com/p3p/Adafruit_NeoPixel/archive/marlin_sim_native.zip
Downloading...
Library Manager: Adafruit NeoPixel @ 1.5.0 has been installed!
Library Manager: Installing https://github.com/p3p/LiquidCrystal/archive/master.zip
[... 779 lines cut ...]
      |      ^~~~~~~~~~~~~~~~~~~
Marlin\src\gcode\config\../../inc/../core/serial.h:299:6: note: candidate: 'void serial_echopair_PGM(const char*, long int)'
  299 | void serial_echopair_PGM(PGM_P const s_P, long v);
      |      ^~~~~~~~~~~~~~~~~~~
Marlin\src\gcode\config\../../inc/../core/serial.h:300:6: note: candidate: 'void serial_echopair_PGM(const char*, float)'
  300 | void serial_echopair_PGM(PGM_P const s_P, float v);
      |      ^~~~~~~~~~~~~~~~~~~
Marlin\src\gcode\config\../../inc/../core/serial.h:301:6: note: candidate: 'void serial_echopair_PGM(const char*, double)'
  301 | void serial_echopair_PGM(PGM_P const s_P, double v);
      |      ^~~~~~~~~~~~~~~~~~~
Marlin\src\gcode\config\../../inc/../core/serial.h:302:6: note: candidate: 'void serial_echopair_PGM(const char*, unsigned char)'
  302 | void serial_echopair_PGM(PGM_P const s_P, unsigned char v);
      |      ^~~~~~~~~~~~~~~~~~~
Marlin\src\gcode\config\../../inc/../core/serial.h:303:6: note: candidate: 'void serial_echopair_PGM(const char*, unsigned int)'
  303 | void serial_echopair_PGM(PGM_P const s_P, unsigned int v);
      |      ^~~~~~~~~~~~~~~~~~~
Marlin\src\gcode\config\../../inc/../core/serial.h:304:6: note: candidate: 'void serial_echopair_PGM(const char*, long unsigned int)'
  304 | void serial_echopair_PGM(PGM_P const s_P, unsigned long v);
      |      ^~~~~~~~~~~~~~~~~~~
Marlin\src\gcode\config\../../inc/../core/serial.h:305:13: note: candidate: 'void serial_echopair_PGM(const char*, bool)'
      |             ^~~~~~~~~~~~~~~~~~~
*** [.pio\build\simulator_windows\debug\src\src\gcode\config\M302.o] Error 1
~~~

[platformio.zip](https://github.com/MarlinFirmware/Marlin/files/6872509/platformio.zip)

sl1pkn07, 2021-07-24:
as note. same steps in WSL2(archlinux) the build is OK (but fails about GL things)

p3p, 2021-07-24:
https://github.com/p3p/Marlin/tree/pr_sim_win_fix is hacky but should fix the build for now as long as you set the sdl lib path to wherever yours is.

There is still also an issue with sdl_net not being able to open a port for listening though which will stop the sim starting, you need to comment out `net_serial.listen_on_port(8099);`  in`.pio\libdeps\simulator_windows\MarlinSimUI\src\MarlinSimulator\main.cpp` to disable the network serial port.

also added a missing lib to the pacman example.

sl1pkn07, 2021-07-24:
> https://github.com/p3p/Marlin/tree/pr_sim_win_fix is hacky but should fix the build for now as long as you set the sdl lib path to wherever yours is.

thanks. now i can able to build the sim.

greetings

PS: is possible use pkgconfig for get the SDL include path instead hardcoded one?

thinkyhead, 2021-07-30:
It looks like `pr_sim_win_fix` will get submitted as a PR soon. So if you want to close this, that's fine.

sjasonsmith, 2021-08-05:
I posted a PR to the simulator repo to resolve the problem with net_serial.listen_on_port, and crashes which occurred when it failed.

https://github.com/p3p/MarlinSimUI/pull/3

sjasonsmith, 2021-08-09:
@sl1pkn07 with the latest updates to Marlin and MarlinSimulator this now builds and launches cleanly. I'm going to go ahead and close this, let us know if you still have trouble with it.

sjasonsmith, 2021-08-09:
@thinkyhead I also verified that your example in the Configs repo builds cleanly and can launch on Windows.

closed 2021-08-09 by sjasonsmith