MarlinFirmware/Marlin #22432: [FR] TMC 5160 integrated motion controller support

jetdream, 2021-07-25:
### Is your feature request related to a problem? Please describe.

No

### Are you looking for hardware support?

TMC 5160 - integrated motion controller

### Describe the feature you want

Currently Marlin does not utilize TMC 5160's integrated motion controller.
It can greatly unload microcontroller from many computations and make motions smoother - it performs acceleration/deceleration logic with up to 256 microsteps which is almost impossible on the most microcontrollers without interpolation.
 

### Additional context

TMC 5160 became widespread with reasonable prices. I think it's time to start using the full power of one of the best stepper drivers on the market.

teemuatlut, 2021-07-29:
This was actually explored before with TMC5130 drivers in the TRAMS platform.
In the end it became evident that it changes so much in the very core of Marlin and how the firmware fundamentally works and it would be a lot of development time to implement as well as to maintain it in the future and frankly for very little benefit.
The main argument before was that by offloading we can reduce the load on the AVR chip but that has already been solved by cheap 32bit platforms.

thinkyhead, 2021-07-30:
I explored the TRAMS fork of Marlin at one point and the code for a standard Cartesian seemed pretty straightforward, but it basically replaces the entire Stepper ISR, and it is not clear how we can tune accelerations, get steppers to start and arrive at the same times, do different types of acceleration profiles, decouple the E axis for Linear Advance, or do other advanced things without creating a whole separate Stepper class for the motion controller. IOW it is a lot of work.

I don't mind leaving the Feature Request up, but it is important to remember there is no team standing by plowing through the feature request list all day. This issue is just a request for _someone, somewhere in the world_ to please implement this feature, and this is a relatively obscure location if you really want someone to jump on it. For best results, you should seek out help and pay someone to implement the feature or port TRAMS in some basic form to get the ball rolling.

We have a Job Postings section on the Marlin Firmware Discord channel for just this purpose.