MarlinFirmware/Marlin #22436: [FR] (feature summary)

Slow-Bear, 2021-07-25:
### Is your feature request related to a problem? Please describe.

No.
I want to have a swappable tool head ( CNC spindle / laser diode / 3d printer head )

That means dedicated menu screens

### Are you looking for hardware support?

It would be nice to run on a Arduino

### Describe the feature you want

I want to have a swappable tool head ( CNC spindle / laser diode / 3d printer head )

Marlin already suport 2 at the same time ( 3d printer and laser/spindle but not hot swappable)

thinkyhead, 2021-07-30:
How much are you able to accomplish with the current `SWITCHING_TOOLHEAD` feature?

Slow-Bear, 2021-07-31:
I am currently working on this option.
But so far not much - having trouble to "trick" the firmware in believing the CNC spindle is a 3d printer head

However that option is work in progress.

Slow-Bear, 2021-08-05:
Hello 

I have failed miserably with the SWITCHING_TOOLHEAD --- no correct menus / no related operations in the menu 

The silver lining thou is the tool sensor pin which I can hardwire for tool head identification

thinkyhead, 2021-08-09:
Everything is a work-in-progress. Hopefully nerds with tool-changers will contribute more code this year.