MarlinFirmware/Marlin #22437: [BUG] ( syntaxis error: extra tokens at end of #ifdef directive)

npmomchev, 2021-07-25:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

In file "pins_BTT_SKR_PRO_common.h" on line 263 compiler generate error : extra tokens at end of #ifdef directive

#ifdef TEMP_SENSOR_PROBE && !defined(TEMP_PROBE_PIN)              Original syntax
#if defined(TEMP_SENSOR_PROBE) && !defined(TEMP_PROBE_PIN)     I think this will correct the error

### Bug Timeline

not shure

### Version of Marlin Firmware

Recv: FIRMWARE_NAME:Marlin bugfix-2.0.9.1

thisiskeithb, 2021-07-25:
Thanks for the report! I fixed it in https://github.com/MarlinFirmware/Marlin/pull/22438.

thisiskeithb, 2021-07-28:
https://github.com/MarlinFirmware/Marlin/pull/22438 Has been merged. Closing.

closed 2021-07-28 by thisiskeithb