MarlinFirmware/Marlin #22440: [BUG] TFT_GENERIC INTERFACE_FSMC didn't build missing SRAM_HandleTypeDef and FSMC_NORSRAM_DEVICE

rsdcampos, 2021-07-26:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

First of all, thanks for your great work. The bug occurs when I try to compile the marlin with a generic TFT with a FSMC interface because same missing stuff. The error just occur with FSMC interface, with SPI build well.
The error log is above. and the config files is attached.
[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/6874754/Configurations.zip)

_"In file included from Marlin\src\HAL\STM32\tft\tft_fsmc.cpp:28:
Marlin\src\HAL\STM32\tft\tft_fsmc.h:55:12: error: 'SRAM_HandleTypeDef' does not name a type; did you mean 'UART_HandleTypeDef'?"_

_In file included from Marlin\src\HAL\STM32\tft\tft_fsmc.cpp:28:
Marlin\src\HAL\STM32\tft\tft_fsmc.h:100:12: error: 'FSMC_NORSRAM_DEVICE' was not declared in this scope; did you mean 'FSMC_NORSRAM_TYPEDEF'?_

### Bug Timeline

It was the first time I tried it, I never hed seen working

### Expected behavior

Compile the marlin for generic TFT with FSMC interface to test my TFT generic board with Marlin.

### Actual behavior

Error and stops the compilation.

### Steps to Reproduce

1- Replace the configuration.h and Configuration_adv.h files in the last marlin bugfix folder for the ones I provided here; 
2- Open VS Code;
3- Open the Auto Build Marlin;
4- Open the last marlin bugfix folder;
5- Click in build button in Auto Build Marlin;
6- Wait for the error.

### Version of Marlin Firmware

Marlin-bugfix-2.0.9.1

### Printer model

Prusa clone

### Electronics

Fysetc S6 mainboard

### Add-ons

Parallel FSMC intefaced 2.4" TFT with touch

### Additional information & file uploads

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/6874754/Configurations.zip)

thisiskeithb, 2021-07-26:
Did you modify any of the motherboard pins files to connect your TFT? Which pins did you use?

I don't see any called out in your configs and there aren't any pins assigned to generic TFTs in the various S6 pins files.

rsdcampos, 2021-07-26:
First of all I tried just built the firmware to then use it, so I changed the pin file for design the fsmc pins and same pins I used was busy I know, but it was not the error, the error is the missing typedef. I looked for this type in the Marlin files and I can't fond it in any where.
For example, I found the definition of the SRAM_HandleTypeDef in stm32f4xx_hal_sram.h and FSMC_NORSRAM_DEVICE in stm32f4xx_ll_fsmc.h files, in a project made direct in CubeMX just for test this TFT, but I dont know how to put it in Marlin. I tried to import all file, but than it call another tons of dependencies...

tpruvot, 2021-07-26:
try to add -DHAL_SRAM_MODULE_ENABLED but i dont think the https://wiki.fysetc.com/FYSETC_S6/ has a FSMC ribbon... seems to use EXT ports...

thisiskeithb, 2021-07-26:
Please attach your modified pins file and any other files you modified.

tpruvot, 2021-07-26:
the tiny FPC cable is Serial Interface (DWIN like) cf https://wiki.fysetc.com/FYS_Touch_Panel/

rsdcampos, 2021-07-26:
@thisiskeithb, this is the modified pins file, [pins_FYSETC_S6.zip](https://github.com/MarlinFirmware/Marlin/files/6880964/pins_FYSETC_S6.zip), but the error didn't seems to be about pins. I tried other boards with same issue, for example, I have a STM32F407VET6 (without Shield) and a MKS Robin E2 and I tried to compile to it and in both I get the same issue, so there ate 3 boards with same issue here.
My main goal for now is just make it to show anything in TFT without using in the printer at first time, just after it I will try use it with my printer.

thisiskeithb, 2021-07-26:
> This is the modified pins file, pins_FYSETC_S6.zip, but the error didn't seems to be about pins

Thanks for providing the pins. This is true, but it's helpful to include everything that was modified so we're all working from the same code and can more easily verify the issue.

rsdcampos, 2021-07-26:
@tpruvot, thanks for your answer to. I tried your suggestion but same issue. I see that the FSMC pins are assigned at the tft_fsmc.h and there are this "FSMC_NORSRAM_DEVICE" that I don't know exactly what it is, but in other project (no Marlin) it is defined as FSMC_BANK1_1.

tpruvot, 2021-07-26:
FSMC is a 16bit interface = 16 data wires ribbon (minimum) like a parallel port if you prefer

rsdcampos, 2021-07-26:
Nope, the mine TFT is 8bit FSMC interface. because of it I tried using `#define TFT_INTERFACE_FSMC` and `#define TFT_INTERFACE_FSMC_8BIT` the 8bit type didn't work, because Marlis think it is SPI and call for the SPI pins instead of the FSMC ones (I don't know why), so I tried using both, `TFT_INTERFACE_FSMC `and  `TFT_INTERFACE_FSMC_8BIT `(this last is just for the FSMC.h file understood that just need 8 data wires). So the error was the same missing datatypes.

I forget to say before, but sorry my bad English, I'm brazilian and my english is nos as good as I liked. So, if there are any mistake in my text, please, sorry and fell free to ask for better explication if need.

jmz52, 2021-07-27:
@rsdcampos

> I see that the FSMC pins are assigned at the tft_fsmc.h and there are this "FSMC_NORSRAM_DEVICE" that I don't know exactly what it is, but in other project (no Marlin) it is defined as FSMC_BANK1_1.

FSMC bank is defined by TFT_CS_PIN in https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/HAL/STM32/tft/tft_fsmc.cpp
```
  uint32_t NSBank = (uint32_t)pinmap_peripheral(digitalPinToPinName(TFT_CS_PIN), PinMap_FSMC_CS);
...
  controllerAddress = FSMC_BANK1_1;
  #ifdef PF0
    switch (NSBank) {
      case FSMC_NORSRAM_BANK2: controllerAddress = FSMC_BANK1_2 ; break;
      case FSMC_NORSRAM_BANK3: controllerAddress = FSMC_BANK1_3 ; break;
      case FSMC_NORSRAM_BANK4: controllerAddress = FSMC_BANK1_4 ; break;
    }
  #endif
```

>mine TFT is 8bit FSMC interface

This is the root cause of your problems. FSMC IO functions were never implemented for 8-bit mode (due to lack of actual hardware to test code on).
Could you provide more details how TFT and touch screens are connected to your board?

rsdcampos, 2021-07-27:
@jmz52, Thanks for your answer. Unfortunately there are almost none information in the web about using generic TFT in marlin, what I could found is proprietary ones, like MKS and others and all in SPI, I found none in FSMC, so I should work in trial and error to try make it to work, I just opened this issue because I didn't found this needed typedefs in Marlin code, and as FSMC need it, I think it is a Bug even if I made any mistake in my code.
I'll try same changes in my code based in your informations to see if there are any progress in build this.

> Could you provide more details how TFT and touch screens are connected to your board?

Yes, this is a MAR2406 module with a ILI9341 IC driver and FSMC 8bits data bus and have SD card and touch.
This is exactly the board: http://www.lcdwiki.com/2.4inch_Arduino_Display
This is the user manual: [2.4inch_Arduino_8BIT_Module_MAR2406_User_Manual_EN.pdf](https://github.com/MarlinFirmware/Marlin/files/6886718/2.4inch_Arduino_8BIT_Module_MAR2406_User_Manual_EN.pdf)
This is in a shield for arduino Uno/Mega, but I just use it in my STM32F407Vet6.
If I could make it to work with Marlin, I'll make a shield to minimize the wire mess.

Thanks again for your help and forgive me for my bed English if any mistake.

thinkyhead, 2021-07-30:
The Marlin Fysetc S6 variant is missing a `hal_conf_extra.h` or `hal_conf_custom.h` with `HAL_TIM_MODULE_ENABLED` defined. Adding this file to our variant should do the trick.

thinkyhead, 2021-07-30:
Hmm… Nevermind, there seems to be more to it. We have been messing with the fysetc_s6 lately, so this is possibly a side-effect.

thinkyhead, 2021-07-30:
I get the sense that adding FSMC to a board that doesn't have some native support in the platform is not a trivial thing.

jmz52, 2021-07-30:
Problem's root cause is that STM32F446 as well as STM32F429, STM32F7 and STM32H7 MCUs do not have FSMC (Flexible Static Memory Controller).
FSMC was replaced by FMC (Flexible Storage Manager) which has extended functionality and improved performance.
So this issue  it not really a bug but a feature request to add support for FMC hardware in MCUs listed above.

I was able to compile Marlin with no errors using crude fix with redefining missing FSMC entities with similar FMS entities, but I have no clue if resulting coide will work at all.
Modified code is available at https://github.com/jmz52/Marlin/tree/FYSETC_S6

@rsdcampos, I've changed TFT_RS_PIN from PC7 (invalid) to PE2 (FSMC_A23). Pin is available in Z2 driver's connector, make sure to change your wiring if you want to test this code.
Also I've changed display style from TFT_CLASSIC_UI (upscaled 128x64 UI) to TFT_COLOR_UI (implementation of MarlinUI designed for color TFT displays)

rsdcampos, 2021-07-31:
Thanks @jmz52, I'll take a carefully look in your modified code, I agree that this boards have FMC instead of FSMC, but as I say before, I tried with other boards like STM32F407Vet6 that is FSMC capable. Anyway it still may been any mistake from me.

thinkyhead, 2021-07-31:
> I tried with other boards…

Is this not an issue of the platform HAL failing to support FSMC in certain cases, or is this entirely within Marlin's purview?

thisiskeithb, 2021-07-31:
The Anet ET4 and ET4P motherboards support FSMC TFTs (it's all that Anect connects to them) and those boards are STM32F407VGT6:
https://github.com/MarlinFirmware/Marlin/blob/d9ab20ec27599d99f0a7fa901086687a6d9139d5/ini/stm32f4.ini#L102-L122

rsdcampos, 2021-08-02:
@jmz52, thanks again for your effort in help me with this Bug and/or FR. I tried your mod and it built, but the screen just was full white. I don't know if the fact that the FMC pins where designed for step motors in this board, or just for my low knowledge, but didn't work.
As you think it is not a bug I don't see any reason to let it open, I just opened because I was sure was miss same typesdefs.
It still don't build for me in FSMC boards even with your mods. Maybe my lacke of knowledge about FSMC/FMC interface and Marlin structure.

Thanks again for every buddy for all your care and effort.

jmz52, 2021-08-03:
@rsdcampos, I've added two FMC-specific parameters to initialization code.
https://github.com/jmz52/Marlin/tree/FYSETC_S6
Where the LCD_RST pin is connected to?
Which pin you are using for TFT_RS_PIN?

tpruvot, 2021-08-04:
"10P FPC for Serial Touch display" https://wiki.fysetc.com/FYSETC_S6/

imo you are wrong....

jmz52, 2021-08-04:
"10P FPC for Serial Touch display" is a fancy name for UART and it is hardwired to UART1 connector, so you can't get MarlinUI there.

@rsdcampos, I've checked board's schematics and there is a problem.
FMS lines are used to control stepper motor drivers and to free them for TFT screen you need to sacrifice 3 (or 4) of 6 stepper motors, and you need at least 4 stepper motor drivers for 3D printer.
Also 3 lines are used for 'Enable' pins and have pull-up resistors which *might* affect FMC.
```
{PD_14, FSMC_NORSRAM_DEVICE, FSMC_PIN_DATA}, // FSMC_D00 // Z-STEP
{PD_15, FSMC_NORSRAM_DEVICE, FSMC_PIN_DATA}, // FSMC_D01 // Z-EN
{PD_0,  FSMC_NORSRAM_DEVICE, FSMC_PIN_DATA}, // FSMC_D02 // LCD_D6
{PD_1,  FSMC_NORSRAM_DEVICE, FSMC_PIN_DATA}, // FSMC_D03 // LCD_D7
{PE_7,  FSMC_NORSRAM_DEVICE, FSMC_PIN_DATA}, // FSMC_D04 // X-CS
{PE_8,  FSMC_NORSRAM_DEVICE, FSMC_PIN_DATA}, // FSMC_D05 // X-RX
{PE_9,  FSMC_NORSRAM_DEVICE, FSMC_PIN_DATA}, // FSMC_D06 // X-EN
{PE_10, FSMC_NORSRAM_DEVICE, FSMC_PIN_DATA}, // FSMC_D07 // X-DIR
{PD_4,  FSMC_NORSRAM_DEVICE, FSMC_PIN_DATA}, // FSMC_NOE // E0-EN
{PD_5,  FSMC_NORSRAM_DEVICE, FSMC_PIN_DATA}, // FSMC_NWE // E0-STEP

{PD_11, FSMC_RS(16), FSMC_PIN_DATA}, // FSMC_A16 // NC
{PD_12, FSMC_RS(17), FSMC_PIN_DATA}, // FSMC_A17 // Z-RX
{PD_13, FSMC_RS(18), FSMC_PIN_DATA}, // FSMC_A18 // Z-DIR
{PE_3,  FSMC_RS(19), FSMC_PIN_DATA}, // FSMC_A19 // E2-EN
{PE_4,  FSMC_RS(20), FSMC_PIN_DATA}, // FSMC_A20 // E2-DIR
{PE_5,  FSMC_RS(21), FSMC_PIN_DATA}, // FSMC_A21 // E1-EN
{PE_6,  FSMC_RS(22), FSMC_PIN_DATA}, // FSMC_A22 // E1-STEP
{PE_2,  FSMC_RS(23), FSMC_PIN_DATA}, // FSMC_A23 // E2-STEP
```

SPI TFT is not an options either - on this board the SD card is connected to SPI, so you can't have a dedicated SPI for TFT and touch screen. HAL STM32 uses DMA for background data transfer and can't share SPI with other devices like SD card or TMC2130 drivers.

thisiskeithb, 2021-08-04:
Since this board/LCD combination is not possible, I'm going to close this bug/FR.

closed 2021-08-04 by thisiskeithb