MarlinFirmware/Marlin #22445: [BUG] Tiltng the mesh resets mesh in Unified Bed Leveling

bergstar, 2021-07-26:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Running G29 J after G29 P1 resets the mesh, by replicating a single value across the whole mesh.

Same result with G29 J3. Tested and confirmed on 2 different printers. Printers reset, EEPROM initialized, firmware re-flashed etc. Also tested on current 2.0.x with same results.

### Bug Timeline

New

### Expected behavior

Mesh should be tilted.

### Actual behavior

All points in the mesh are replaced with a single value.

### Steps to Reproduce

1. Send ubl.gcode to a board running latest Marlin firmware
2. Observe output where initialized mesh is reset (shown in output.txt).

### Version of Marlin Firmware

2.0.9.1

### Printer model

Creality Ender 3 Pro

### Electronics

4.2.2

### Add-ons

BLTouch

### Your Slicer

Prusa Slicer

### Host Software

Pronterface

### Additional information & file uploads

[output.txt](https://github.com/MarlinFirmware/Marlin/files/6881306/output.txt)
[ubl-gcode.txt](https://github.com/MarlinFirmware/Marlin/files/6881307/ubl-gcode.txt)
[Configuration-h.txt](https://github.com/MarlinFirmware/Marlin/files/6881312/Configuration-h.txt)
[Configuration-adv.txt](https://github.com/MarlinFirmware/Marlin/files/6881311/Configuration-adv.txt)

bergstar, 2021-07-26:
Potentially related issues:
https://github.com/MarlinFirmware/Marlin/issues/21226
https://github.com/MarlinFirmware/Marlin/issues/19621

DerAndere1, 2021-07-26:
Also: duplicate of issue https://github.com/MarlinFirmware/Marlin/issues/22175 . This issue can be closed as it is fixed in Marlin bugfix-2.0.x

bergstar, 2021-07-26:
> 
> 
> Also: duplicate of issue #22175 . Did you test with bugfix 2.0.x after commit [154decf](https://github.com/MarlinFirmware/Marlin/commit/154decfc66c30808cfe7320e542fcf90427d7176) (downloaded after 19th July)?

sry didn't see that one! I will test tomorrow - but yes, I'm sure it is fixed.

thisiskeithb, 2021-07-27:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/22175

thisiskeithb, 2021-07-27:
This was fixed in https://github.com/MarlinFirmware/Marlin/commit/154decfc66c30808cfe7320e542fcf90427d7176.

closed 2021-07-27 by thisiskeithb