MarlinFirmware/Marlin #22446: [BUG] Marlin not compiling with 5 Linear Axes and 0 Extruders

joshmaceachern, 2021-07-27:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Trying to compile latest Marlin for SKR 2 with 5 linear axes and 0 extruders. Marlin compilation fails when it gets to planner.cpp, and it looks like the planner is breaking with having 0 extruders defined. Already tried the bugfix version with the same issues. I have only changed pin definitions in the appropriate pins.h and have also edited configuration.h and configuration_adv.h

Attached is zip file containing configuration.h and configuration_adv.h

Error output is here:
`Compiling .pio\build\BIGTREE_SKR_2\src\src\module\stepper\trinamic.cpp.o
Compiling .pio\build\BIGTREE_SKR_2\src\src\module\temperature.cpp.o
Marlin\src\module\planner.cpp: In static member function 'static bool Planner::_populate_block(block_t*, bool, const abce_long_t&, feedRate_t, uint8_t, const_float_t)':
Marlin\src\module\planner.cpp:2687:33: error: 'de' was not declared in this scope; did you mean 'dm'?
 2687 |     const float extra_xyjerk = (de <= 0) ? TRAVEL_EXTRA_XYJERK : 0;
      |                                 ^~
      |                                 dm
*** [.pio\build\BIGTREE_SKR_2\src\src\module\planner.cpp.o] Error 1
In file included from Marlin\src\module\../inc/MarlinConfigPre.h:37,
                 from Marlin\src\module\../inc/MarlinConfig.h:28,
                 from Marlin\src\module\settings.h:28,
                 from Marlin\src\module\settings.cpp:46:
Marlin\src\module\settings.cpp: In static member function 'static void MarlinSettings::reset()':
Marlin\src\module\settings.cpp:2540:48: error: 'xyze_pos_t' {aka 'struct XYZEval<float>'} has no member named 'e'
 2540 |     TERN_(HAS_CLASSIC_E_JERK, planner.max_jerk.e = DEFAULT_EJERK);
      |                                                ^
Marlin\src\module\../inc/../core/macros.h:558:26: note: in definition of macro 'THIRD'
  558 | #define THIRD(a,b,c,...) c
      |                          ^
Marlin\src\module\../inc/../core/macros.h:204:29: note: in expansion of macro '___TERN'
  204 | #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
      |                             ^~~~~~~
Marlin\src\module\../inc/../core/macros.h:203:29: note: in expansion of macro '__TERN'
  203 | #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
      |                             ^~~~~~
Marlin\src\module\../inc/../core/macros.h:202:29: note: in expansion of macro '_TERN'
  202 | #define TERN_(O,A)          _TERN(_ENA_1(O),,A)     // OPTION ? 'A' : '<nul>'
      |                             ^~~~~
Marlin\src\module\settings.cpp:2540:5: note: in expansion of macro 'TERN_'
 2540 |     TERN_(HAS_CLASSIC_E_JERK, planner.max_jerk.e = DEFAULT_EJERK);
      |     ^~~~~
Marlin\src\module\settings.cpp:2540:52: error: 'DEFAULT_EJERK' was not declared in this scope; did you mean 'DEFAULT_JJERK'?
 2540 |     TERN_(HAS_CLASSIC_E_JERK, planner.max_jerk.e = DEFAULT_EJERK);
      |                                                    ^~~~~~~~~~~~~
Marlin\src\module\../inc/../core/macros.h:558:26: note: in definition of macro 'THIRD'
  558 | #define THIRD(a,b,c,...) c
      |                          ^
Marlin\src\module\../inc/../core/macros.h:204:29: note: in expansion of macro '___TERN'
  204 | #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
      |                             ^~~~~~~
Marlin\src\module\../inc/../core/macros.h:203:29: note: in expansion of macro '__TERN'
  203 | #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
      |                             ^~~~~~
Marlin\src\module\../inc/../core/macros.h:202:29: note: in expansion of macro '_TERN'
  202 | #define TERN_(O,A)          _TERN(_ENA_1(O),,A)     // OPTION ? 'A' : '<nul>'
      |                             ^~~~~
Marlin\src\module\settings.cpp:2540:5: note: in expansion of macro 'TERN_'
 2540 |     TERN_(HAS_CLASSIC_E_JERK, planner.max_jerk.e = DEFAULT_EJERK);
      |     ^~~~~
Marlin\src\module\settings.cpp: In static member function 'static void MarlinSettings::report(bool)':
Marlin\src\module\settings.cpp:3160:52: error: 'xyze_pos_t' {aka 'struct XYZEval<float>'} has no member named 'e'
 3160 |           , SP_E_STR, LINEAR_UNIT(planner.max_jerk.e)
      |                                                    ^
Marlin\src\module\../inc/../core/macros.h:579:24: note: in definition of macro 'EVAL1'
  579 | #define EVAL1(V...)    V
      |                        ^
Marlin\src\module\../inc/../core/macros.h:578:30: note: in expansion of macro 'EVAL1'
  578 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                              ^~~~~
[... 168 lines cut ...]
Marlin\src\module\../inc/../core/serial.h:227:44: note: in expansion of macro '_SELP_N_P'
  227 | #define SERIAL_ECHOLNPAIR_P(V...) do{ EVAL(_SELP_N_P(TWO_ARGS(V),V)); }while(0)
      |                                            ^~~~~~~~~
Marlin\src\module\settings.cpp:3143:5: note: in expansion of macro 'SERIAL_ECHOLNPAIR_P'
 3143 |     SERIAL_ECHOLNPAIR_P(
      |     ^~~~~~~~~~~~~~~~~~~
Marlin\src\module\settings.cpp:3160:23: note: in expansion of macro 'LINEAR_UNIT'
 3160 |           , SP_E_STR, LINEAR_UNIT(planner.max_jerk.e)
      |                       ^~~~~~~~~~~
*** [.pio\build\BIGTREE_SKR_2\src\src\module\settings.cpp.o] Error 1
==================================================================== [FAILED] Took 93.74 seconds ====================================================================

Environment    Status    Duration
-------------  --------  ------------
BIGTREE_SKR_2  FAILED    00:01:33.744
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/6882722/configs.zip)
`

### Bug Timeline

First time using this board, first time issue

### Version of Marlin Firmware

2.0.9.1

joshmaceachern, 2021-07-27:
Further info: This issue only happens when classic jerk is enabled. Switching to junction deviation lets it compile with no issues.

ellensp, 2021-07-27:
This should fix it

```DIFF
diff --git a/Marlin/src/inc/Conditionals_LCD.h b/Marlin/src/inc/Conditionals_LCD.h
index 7e2239b3f6..1174beb3fe 100644
--- a/Marlin/src/inc/Conditionals_LCD.h
+++ b/Marlin/src/inc/Conditionals_LCD.h
@@ -1053,7 +1053,7 @@
 #endif

 // E jerk exists with JD disabled (of course) but also when Linear Advance is disabled on Delta/SCARA
-#if ENABLED(CLASSIC_JERK) || (IS_KINEMATIC && DISABLED(LIN_ADVANCE))
+#if ENABLED(CLASSIC_JERK) && EXTRUDERS > 0 || (IS_KINEMATIC && DISABLED(LIN_ADVANCE))
   #define HAS_CLASSIC_E_JERK 1
 #endif

diff --git a/Marlin/src/module/planner.cpp b/Marlin/src/module/planner.cpp
index e48d05b09f..f6c55a7e53 100644
--- a/Marlin/src/module/planner.cpp
+++ b/Marlin/src/module/planner.cpp
@@ -2684,7 +2684,12 @@ bool Planner::_populate_block(block_t * const block, bool split_move,
     #ifndef TRAVEL_EXTRA_XYJERK
       #define TRAVEL_EXTRA_XYJERK 0
     #endif
-    const float extra_xyjerk = (de <= 0) ? TRAVEL_EXTRA_XYJERK : 0;
+
+    #if EXTRUDERS > 0
+      const float extra_xyjerk = (de <= 0) ? TRAVEL_EXTRA_XYJERK : 0;
+    #else
+      const float extra_xyjerk = 0;
+    #endif

     uint8_t limited = 0;
     TERN(HAS_LINEAR_E_JERK, LOOP_LINEAR_AXES, LOOP_LOGICAL_AXES)(i) {

```

commit 99a53e2c (2021-07-30): 🐛 Fix 5-axis no extruder compile

Fixes #22446
commit 74b0133b (2021-07-30): 🐛 Fix 5-axis no extruder compile

Fixes #22446
closed 2021-07-30 by thinkyhead