MarlinFirmware/Marlin #22451: [BUG] Z_PROBE_SERVO_NR crashes the Ender 3 v2

dwery, 2021-07-28:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Enabling Z_PROBE_SERVO_NR leads to a complete crash of the Ender 3 v2. The display is totally garbled and the loading bar can be seen over and over.

Configuration is Ender 3 v2 template + attached changes

### Bug Timeline

New bug. dd4990252e891cdfe56cb7d6e3bbe1e6289be649 was working.

### Version of Marlin Firmware

296d0d495cad942cccd611b74a6b8a414a089a65

### Printer model

Ender 3 v2

### Additional information & file uploads

[probe.txt](https://github.com/MarlinFirmware/Marlin/files/6890540/probe.txt)

ellensp, 2021-07-28:
That seems rather unlikely
In marlin is 
```CPP
#if ENABLED(BLTOUCH)
  #ifndef Z_PROBE_SERVO_NR
    #define Z_PROBE_SERVO_NR 0 
```
So if you enable it or not Marlin always enables it...

ellensp, 2021-07-28:
and also Z_SERVO_ANGLES are also bypassed
```CPP
  #ifdef Z_PROBE_SERVO_NR
    #if ENABLED(BLTOUCH)
      #include "../feature/bltouch.h"
      #undef Z_SERVO_ANGLES
      #define Z_SERVO_ANGLES { BLTOUCH_DEPLOY, BLTOUCH_STOW }
    #endif

where
#define BLTOUCH_DEPLOY          10
#define BLTOUCH_STOW            90
```

dwery, 2021-07-28:
@ellensp if it help, my probe is not a BL Touch, so I guess there is something else involved.

ellensp, 2021-07-28:
Please attach your full configuration files.

thinkyhead, 2021-07-30:
What can we do to get folks to upload their configurations? I thought I made it pretty explicit in the new bug report template.

thisiskeithb, 2021-07-30:
> What can we do to get folks to upload their configurations? I thought I made it pretty explicit in the new bug report template.

Honestly, some kind of bot to check for attachments would be the best path forward. It's getting pretty tiring asking for them.

thinkyhead, 2021-07-30:
Fun with actions. There goes our salvage, guys.

dwery, 2021-07-30:
@thinkyhead I thought that attaching a delta diff from the stock Ender3v2 config was enough. I’ll upload ASAP.

thinkyhead, 2021-07-30:
> attaching a delta diff from the stock Ender3v2 config was enough

As you can imagine, it is quite simple and pleasant to drop the complete config files into place, hit "Build," and see what you are seeing in the shake of a lamb's tail. Any other format is a hassle.

dwery, 2021-07-31:
[config.tar.gz](https://github.com/MarlinFirmware/Marlin/files/6911824/config.tar.gz)
Her you go

thinkyhead, 2021-08-10:
For isolation testing… try commenting out these lines in `MarlinCore.cpp`…
```cpp
  #if HAS_Z_SERVO_PROBE
    SETUP_RUN(probe.servo_probe_init());
  #endif
```

thinkyhead, 2021-08-10:
If that still crashes, then also comment out the lines just above it…
```cpp
  #if HAS_SERVOS
    SETUP_RUN(servo_init());
  #endif
```

dwery, 2021-08-16:
@thinkyhead tried both, still crashes.

dwery, 2021-09-07:
@thinkyhead is there anything else I can test?

dwery, 2021-11-14:
A few more data points:

BLTOUCH enabled and my non bltouch probe connected: crashes.
BLTOUCH enabled and my non bltouch probe disconnected: no crash until I connect the probe at runtime.
Z_PROBE_SERVO_NR and probe connected: crashes
Z_PROBE_SERVO_NR and probe disconnected: crashes

dwery, 2021-11-15:
@thinkyhead 
Disabling ENDSTOP_INTERRUPTS_FEATURE solves this issue. I guess the handler is called in an unexpected way when this particular type of probe is enabled.

closed 2022-01-10 by thinkyhead