MarlinFirmware/Marlin #22454: [BUG] SLOW Probe fail! - Triggered early.

Fusseldieb, 2021-07-29:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When I try to do a "G29" to do the bed calibration of my CNC machine, it goes to the first point at the far bottom right, touches down, stops, repeats and when it touches, it just stops there with the message "SLOW Probe fail! - Triggered early.". I already tried to adjust various parameters and even the Z position to no avail.

### Bug Timeline

Since I had Marlin

### Expected behavior

It to go to all points and level the bed

### Actual behavior

It just measures the first, does it again slowly and then stops once it touches the second time, with an error.

### Steps to Reproduce

1. Do a G28
2. Do a G29
3. Watch

### Version of Marlin Firmware

2.0.9.1

### Printer model

Custom CNC

### Electronics

SKR 2.0 REV B

### Host Software

Other (explain below)

### Additional information & file uploads

```
CNCjs 1.9.22 [Marlin]
Conectado à porta COM3 com taxa de transferência de 250000
FIRMWARE_NAME:Marlin 2.0.9.1 (Jul 23 2021 00:13:05) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:PAREN_COMMENTS:1
Cap:SERIAL_XON_XOFF:0
Cap:BINARY_FILE_TRANSFER:0
Cap:EEPROM:1
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_POS:0
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:1
Cap:RUNOUT:0
Cap:Z_PROBE:1
Cap:LEVELING_DATA:1
Cap:BUILD_PERCENT:0
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:0
Cap:HOST_ACTION_COMMANDS:0
Cap:PROMPT_SUPPORT:0
Cap:SDCARD:0
Cap:REPEAT:0
Cap:SD_WRITE:0
Cap:AUTOREPORT_SD_STATUS:0
Cap:LONG_FILENAME:0
Cap:THERMAL_PROTECTION:0
Cap:MOTION_MODES:1
Cap:ARCS:1
Cap:BABYSTEPPING:0
Cap:CHAMBER_TEMPERATURE:0
Cap:COOLER_TEMPERATURE:0
Cap:MEATPACK:0
> M111S32
echo:DEBUG:LEVELING
ok
> G28
>>> G28  X0.00 Y0.00 Z273.20
Machine Type: Cartesian
Probe: FIX_MOUNTED_PROBE
Probe Offset X10.00 Y10.00 Z0.00 (Right-Back & Same Z as Nozzle)
Auto Bed Leveling: BILINEAR (disabled)
>>> homeaxis(Z)
Home Fast: 409.80mm
>>> do_homing_move  X0.00 Y0.00 Z273.20
...(Z, 409.80, [25.00])
<<< do_homing_move  X0.00 Y0.00 Z273.20
Move Away: -5.00mm
>>> do_homing_move  X0.00 Y0.00 Z273.20
...(Z, -5.00, [25.00])
<<< do_homing_move  X0.00 Y0.00 Z273.20
Re-bump: 10.00mm
>>> do_homing_move  X0.00 Y0.00 Z273.20
...(Z, 10.00, 6.25)
<<< do_homing_move  X0.00 Y0.00 Z273.20
>>> set_axis_is_at_home(Z)
Axis Z home_offset = 0.00 position_shift = 0.00
> home_offset[Z] = 0.00
  current_position= X0.00 Y0.00 Z273.20 :
<<< set_axis_is_at_home(Z)
  current_position= X0.00 Y0.00 Z273.20 : sync_plan_position
  current_position= X0.00 Y0.00 Z273.20 : > AFTER set_axis_is_at_home
<<< homeaxis(Z)
Raise Z (before homing) by 5.00
>>> do_blocking_move_to  X0.00 Y0.00 Z268.20
>  X0.00 Y0.00 Z268.20
echo:busy: processing
<<< do_blocking_move_to  X0.00 Y0.00 Z268.20
  current_position= X0.00 Y0.00 Z268.20 : sync_plan_position
>>> do_blocking_move_to  X0.00 Y0.00 Z268.20
>  X-2677.50 Y-3757.50 Z268.20
<<< do_blocking_move_to  X-2677.50 Y-3757.50 Z268.20
>>> homeaxis(X)
Home Fast: -2677.50mm
>>> do_homing_move  X0.00 Y0.00 Z268.20
...(X, -2677.50, [66.67])
<<< do_homing_move  X0.00 Y0.00 Z268.20
Move Away: 5.00mm
>>> do_homing_move  X0.00 Y0.00 Z268.20
...(X, 5.00, [66.67])
<<< do_homing_move  X0.00 Y0.00 Z268.20
Re-bump: -10.00mm
>>> do_homing_move  X0.00 Y0.00 Z268.20
...(X, -10.00, 8.33)
echo:busy: processing
<<< do_homing_move  X0.00 Y0.00 Z268.20
>>> set_axis_is_at_home(X)
Axis X home_offset = 0.00 position_shift = 0.00
> home_offset[X] = 0.00
  current_position= X0.00 Y0.00 Z268.20 :
<<< set_axis_is_at_home(X)
  current_position= X0.00 Y0.00 Z268.20 : sync_plan_position
  current_position= X0.00 Y0.00 Z268.20 : > AFTER set_axis_is_at_home
<<< homeaxis(X)
>>> homeaxis(Y)
Home Fast: -3757.50mm
>>> do_homing_move  X5.00 Y0.00 Z268.20
...(Y, -3757.50, [66.67])
<<< do_homing_move  X5.00 Y0.00 Z268.20
Move Away: 5.00mm
>>> do_homing_move  X5.00 Y0.00 Z268.20
...(Y, 5.00, [66.67])
<<< do_homing_move  X5.00 Y0.00 Z268.20
Re-bump: -10.00mm
>>> do_homing_move  X5.00 Y0.00 Z268.20
...(Y, -10.00, 8.33)
<<< do_homing_move  X5.00 Y0.00 Z268.20
>>> set_axis_is_at_home(Y)
Axis Y home_offset = 0.00 position_shift = 0.00
> home_offset[Y] = 0.00
  current_position= X5.00 Y0.00 Z268.20 :
<<< set_axis_is_at_home(Y)
  current_position= X5.00 Y0.00 Z268.20 : sync_plan_position
  current_position= X5.00 Y0.00 Z268.20 : > AFTER set_axis_is_at_home
<<< homeaxis(Y)
  current_position= X5.00 Y5.00 Z268.20 : sync_plan_position
X:5.00 Y:5.00 Z:268.20 E:0.00 Count X:1280 Y:0 Z:172989
<<< G28  X5.00 Y5.00 Z268.20
ok
feeder> G91
ok
feeder> G0 Z-100
ok
feeder> G90
ok
feeder> G91
ok
feeder> G0 Z-20
ok
feeder> G90
ok
feeder> G91
ok
feeder> G0 Z-10
ok
feeder> G90
ok
> G29
>>> G29  X5.00 Y5.00 Z138.20
Machine Type: Cartesian
Probe: FIX_MOUNTED_PROBE
Probe Offset X10.00 Y10.00 Z0.00 (Right-Back & Same Z as Nozzle)
Auto Bed Leveling: BILINEAR (disabled)
  current_position= X5.00 Y5.00 Z138.20 : Probe::set_deployed
deploy: 1
Probe::do_z_raise(10.00)
>>> do_blocking_move_to  X5.00 Y5.00 Z138.20
>  X5.00 Y5.00 Z138.20
<<< do_blocking_move_to  X5.00 Y5.00 Z138.20
>>> do_blocking_move_to  X5.00 Y5.00 Z138.20
>  X5.00 Y5.00 Z138.20
<<< do_blocking_move_to  X5.00 Y5.00 Z138.20
reset_bed_level
>>> Probe::probe_at_point  X5.00 Y5.00 Z138.20
...(1735.00, 50.00, raise, 0, probe_relative)
  current_position= X5.00 Y5.00 Z138.20 :
>>> do_blocking_move_to  X5.00 Y5.00 Z138.20
>  X1725.00 Y40.00 Z138.20
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
<<< do_blocking_move_to  X1725.00 Y40.00 Z138.20
  current_position= X1725.00 Y40.00 Z138.20 : Probe::set_deployed
deploy: 1
>>> Probe::run_z_probe  X1725.00 Y40.00 Z138.20
>>> Probe::probe_down_to_z  X1725.00 Y40.00 Z138.20
>>> do_blocking_move_to  X1725.00 Y40.00 Z138.20
>  X1725.00 Y40.00 Z15.00
echo:busy: processing
echo:busy: processing
<<< do_blocking_move_to  X1725.00 Y40.00 Z15.00
  current_position= X1725.00 Y40.00 Z117.69 : sync_plan_position
<<< Probe::probe_down_to_z  X1725.00 Y40.00 Z117.69
>>> do_blocking_move_to  X1725.00 Y40.00 Z117.69
>  X1725.00 Y40.00 Z122.69
echo:busy: processing
<<< do_blocking_move_to  X1725.00 Y40.00 Z122.69
>>> Probe::probe_down_to_z  X1725.00 Y40.00 Z122.69
>>> do_blocking_move_to  X1725.00 Y40.00 Z122.69
>  X1725.00 Y40.00 Z-20.00
echo:busy: processing
<<< do_blocking_move_to  X1725.00 Y40.00 Z-20.00
  current_position= X1725.00 Y40.00 Z117.67 : sync_plan_position
<<< Probe::probe_down_to_z  X1725.00 Y40.00 Z117.67
SLOW Probe fail! - Triggered early.
<<< Probe::run_z_probe  X1725.00 Y40.00 Z117.67
  current_position= X1725.00 Y40.00 Z117.67 : Probe::set_deployed
deploy: 0
>>> do_blocking_move_to  X1725.00 Y40.00 Z117.67
>  X1725.00 Y40.00 Z117.67
<<< do_blocking_move_to  X1725.00 Y40.00 Z117.67
Error:Probing Failed
<<< Probe::probe_at_point  X1725.00 Y40.00 Z117.67
  current_position= X1725.00 Y40.00 Z117.67 : Probe::set_deployed
deploy: 0
  current_position= X1725.00 Y40.00 Z117.67 : > probing complete
X:1725.00 Y:40.00 Z:117.67 E:0.00 Count X:441600 Y:10240 Z:75896
<<< G29  X1725.00 Y40.00 Z117.67
ok
>
```

thisiskeithb, 2021-07-29:
Please attach your configuration files.

Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

Fusseldieb, 2021-08-01:
After intensive research I found out what caused the Z probe to "trigger early".

CNCs, unlike printers, home going with their spindle **UPWARDS**, while most 3D printers go up with their **BEDS**. While a 3D printer needs to know Z-MIN, a CNC needs Z-MAX. 
My first thought were that the CNC would find Z-MIN on its own by _probing_. However, when probing it always stated _TRIGGERED EARLY_.
Early when? On the slow second descent? 
_No._ On the whole thing. 
Let me explain:

Marlin already does know the Z-MIN when it does homing towards the Z-MAX! How?
When you define the Z max travel in your configuration file, you basically say it: From Z-MAX you may ONLY have _this much_ of room until you reach **Z_MIN**!
So, when probing, Marlin slowly descends the probe, but already knows (or thinks he knows) where the bed is, according to your Z max travel that you typed in. If the probe _triggers before_ it reaches Z-MIN (that it calculated based on that Z max travel), you're going to get the _triggered early_ error.

Now that this is sorted out, I can enable UBL and everything is working fine :)

closed 2021-08-01 by Fusseldieb