MarlinFirmware/Marlin #22456: [BUG] EXPERIMENTAL_I2CBUS + BOARD_RADDS ---> no connection to board after upload

2MSL, 2021-07-29:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

used the latest "bugfix-2.0.x" with auto build marlin.
When setting BOARD_RADDS and EXPERIMENTAL_I2CBUS i cannot connect to the board after uploading.
Without EXPERIMENTAL_I2CBUS is disabled there are no issues connecting to the board.
I need the i2c functionality to control external hardware.
Has anyone an idea whats going on?

### Steps to Reproduce

#define MOTHERBOARD BOARD_RADDS
#define EXPERIMENTAL_I2CBUS
Build and Upload with Auto Build Marlin
Open Monitor in Auto Build Marlin

### Version of Marlin Firmware

12:41:34.383 > FIRMWARE_NAME:Marlin bugfix-2.0.x (Jul 29 2021 12:40:01) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff

thinkyhead, 2021-07-30:
Where are your configuration files? We can't check for conflicts without those.

2MSL, 2021-07-30:
Hi thinkyhead,

here are my config files. Only thing i changed was:

  #define MOTHERBOARD BOARD_RADDS in configuration.h
and
#define EXPERIMENTAL_I2CBUS

greetings and thanks for your help!
2MSL

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/6906658/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/6906659/Configuration_adv.txt)

thinkyhead, 2021-07-31:
Thanks! Try defining `DEBUG_TWIBUS` and `MARLIN_DEV_MODE` in your configuration for some extra output in the logs. Maybe that will tell us where the feature or the board boot is falling down.

Does that board reboot itself when a serial connection is opened to it? It should be easier to get the log output if that is the case. If not, you might be able to just hit the Reset button after establishing a serial connection to see the startup output.