MarlinFirmware/Marlin #22457: [BUG] Wrong Home Position after manual AUTO_BED_LEVELING_3POINT on delta

VICLER, 2021-07-29:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

First of all, there is a bug by compiling marlin if AUTO_BED_LEVELING_3POINT and PROBE_MANUALLY are enabled. 
`Marlin\src\gcode\bedlevel\abl\G29.cpp: In static member function 'static void GcodeSuite::G29()':
Marlin\src\gcode\bedlevel\abl\G29.cpp:571:50: error: ambiguous overload for 'operator=' (operand types are 'xy_pos_t' {aka 'XYval<float>'} and 'vector_3')
  571 |         abl.probePos = points[abl.abl_probe_index];
      |                                                  ^
In file included from Marlin\src\gcode\bedlevel\abl\../../../inc/MarlinConfig.h:47,
                 from Marlin\src\gcode\bedlevel\abl\G29.cpp:27:
Marlin\src\gcode\bedlevel\abl\../../../inc/../core/types.h:306:16: note: candidate: 'XYval<T>& XYval<T>::operator=(const XYZval<T>&) [with T = float]'       
  306 |   FI XYval<T>& operator= (const XYZval<T>  &rs)         { set(rs.x, rs.y); return *this; }
      |                ^~~~~~~~
Marlin\src\gcode\bedlevel\abl\../../../inc/../core/types.h:244:8: note: candidate: 'constexpr XYval<float>& XYval<float>::operator=(const XYval<float>&)'    
  244 | struct XYval {
      |        ^~~~~
Marlin\src\gcode\bedlevel\abl\../../../inc/../core/types.h:244:8: note: candidate: 'constexpr XYval<float>& XYval<float>::operator=(XYval<float>&&)'`
I fixed it with a cast: 
`#elif ENABLED(AUTO_BED_LEVELING_3POINT)

      // Probe at 3 arbitrary points
      if (abl.abl_probe_index < abl.abl_points) {
        abl.probePos = (XYval<float>)points[abl.abl_probe_index];`
This compile error doesnt exist with AUTO_BED_LEVELING_LINEAR, only with 3POINT

After manual 3-Point leveling procedure on delta and then homing, the home position is wrong. 
For examle: 
DELTA_HEIGHT 289.15
I send g29, the print head moves to the first poin(X100 Y0 Z1.5), but my host programm(Repetier Host) reports still the home position(X0 Y0 Z289.15), so I have to send M114 each time after G29 to adjust the hight with Repetier Host moving arrows. 
After all 3 points are done I send g28 and marlin reports X0 Y0 Z-0.48 at home position
Here is a debug report after adjusting the last point and sending g28:

Machine Type: Delta
Probe: PROBE_MANUALLY
Auto Bed Leveling: 3POINT (enabled)
ABL Adjustment X+300.66 Y+204.06 Z+229.46
current_position= X-50.00 Y0.00 Z0.00 : Leveling ON
current_position= X-50.00 Y0.00 Z0.00 : ...Now OFF
current_position= X-50.00 Y0.00 Z0.00 : sync_plan_position
>>> home_delta  X-50.00 Y0.00 Z0.00
current_position= X0.00 Y0.00 Z0.00 : sync_plan_position
echo:busy: processing
echo:busy: processing
homeaxis(X)
Home Fast: 433.72mm
>>> do_homing_move  X0.00 Y0.00 Z299.15
...(X, 433.72, [60.00])
<<< do_homing_move  X0.00 Y0.00 Z299.15
adjDistance:-0.99
>>> do_homing_move  X0.00 Y0.00 Z299.15
...(X, -0.99, 6.00)
<<< do_homing_move  X0.00 Y0.00 Z299.15
<<< homeaxis(X)
>>> homeaxis(Y)
Home Fast: 433.72mm
>>> do_homing_move  X0.00 Y0.00 Z299.15
...(Y, 433.72, [60.00])
<<< do_homing_move  X0.00 Y0.00 Z299.15
<<< homeaxis(Y)
>>> homeaxis(Z)
Home Fast: 433.72mm
>>> do_homing_move  X0.00 Y0.00 Z299.15
...(Z, 433.72, [60.00])
echo:busy: processing
<<< do_homing_move  X0.00 Y0.00 Z299.15
adjDistance:-0.83
>>> do_homing_move  X0.00 Y0.00 Z299.15
...(Z, -0.83, 6.00)
<<< do_homing_move  X0.00 Y0.00 Z299.15
<<< homeaxis(Z)
>>> set_axis_is_at_home(X)
current_position= X0.00 Y0.00 Z299.15 :
<<< set_axis_is_at_home(X)
>>> set_axis_is_at_home(Y)
current_position= X0.00 Y0.00 Z299.15 :
<<< set_axis_is_at_home(Y)
>>> set_axis_is_at_home(Z)
current_position= X0.00 Y0.00 Z289.15 :
<<< set_axis_is_at_home(Z)
current_position= X0.00 Y0.00 Z289.15 : sync_plan_position
<<< home_delta  X0.00 Y0.00 Z289.15
**current_position= X0.00 Y0.00 Z289.15 : Leveling OFF
current_position= X0.00 Y0.00 Z-0.48 : ...Now ON**
current_position= X0.00 Y0.00 Z-0.48 : sync_plan_position
X:0.00 Y:0.00 Z:-0.48 E:0.00 Count A:149373B:150110C:150343
<<< G28  X0.00 Y0.00 Z-0.48

The same behaviour with wrong home position is also with AUTO_BED_LEVELING_LINEAR + PROBE_MANUALLY. 

### Expected behavior

The home Z position with LEVELING ON should be near DELTA_HEIGHT

### Actual behavior

The home Z position with LEVELING ON is near 0

### Steps to Reproduce

Configuration.h:
#define CONFIG_EXAMPLES_DIR "delta/Anycubic/Kossel Linear Plus"
#define PROBE_MANUALLY
#define AUTO_BED_LEVELING_3POINT

### Version of Marlin Firmware

Marlin-2.0.9.1

### Printer model

Anycubic Kossel Linear Plus

### Electronics

BTT SKR 1.3

### Your Slicer

Cura

### Host Software

Repetier Host

thinkyhead, 2021-07-30:
Looks like this is a regression from #19112. The type conversion used to happen automagically. Although the cast can work, it may be best here to just copy the fields separately.

thinkyhead, 2021-07-30:
The three-point issue is pretty obvious, but I'm not able to see where `AUTO_BED_LEVELING_LINEAR` + `PROBE_MANUALLY` is being affected by `vector_3` changes. Perhaps it's `matrix_3x3` being affected in that case.

VICLER, 2021-07-30:
@thinkyhead thank you for response!
I tried a cast first and it fixed compiling error and there were no errors in XY points by probing, so I let this cast. But this does not solve the wrong Z height correction. 
Is there maybe a min- and max-endstop reversed in leveling procedure by delta?

thinkyhead, 2021-07-31:
The line at the end of probing is supposed to be seen by Repetier Host and applied to its reckoning of the current position:
```
X:0.00 Y:0.00 Z:-0.48 E:0.00 Count A:149373B:150110C:150343
```

ChinChillaSPb, 2021-10-05:
i confirm this bug again on my kossel linear plus. it works ok in 2.0.9.1 but returned in 2.0.9.2

DerAndere1, 2021-12-05:
I think this should stay open until we have a confirmation that it works in bugfix-2.0.x

closed 2022-02-14 by github-actions[bot]