MarlinFirmware/Marlin #22458: [BUG] G29 J doesn't respect X/Y_MIN_POS

bergstar, 2021-07-29:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Running G29 J on a printer with UBL mesh, will try to probe outside of allowed build area.

Nozzle-to-Probe offsets
`#define NOZZLE_TO_PROBE_OFFSET { -43, -9, -2.50 }`

Travel limits
`#define X_MIN_POS 5
#define Y_MIN_POS 5`

AUTO_BED_LEVELING_UBL
`#define MESH_INSET 5              // Set Mesh bounds as an inset region of the bed`

### Expected behavior

G29 J should either take into account X/Y_MIN_POS and / or MESH_INSET in order to probe within allowed envelope. 

### Actual behavior

First point where G29 J will try to probe, will be X49. and Y9. cousing probe to measure at bed X0. Y0.
This causes a crash as probe is outside build area, which starts at X5. Y5.

### Steps to Reproduce

1. Setup X/Y_MIN_POS and / or MESH_INSET 
2. Create a mesh
2. Run G29 J and observe X/Y coordinates.

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

Creality Ender 3 Pro

### Electronics

4.2.2

### Add-ons

BLTouch

### Your Slicer

Prusa Slicer

### Host Software

Pronterface

### Additional information & file uploads

[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/6902352/Configuration_adv.h.txt)
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/6902353/Configuration.h.txt)

ManuelMcLure, 2021-07-29:
`X_MIN_POS`/`X_MAX_POS` should _NEVER_ be positive. If the nozzle cannot reach the front corner of the bed then the point it can reach should be defined to be 0,0 and the bed size should be reduced accordingly.

bergstar, 2021-07-29:
> 
> 
> `X_MIN_POS`/`X_MAX_POS` should _NEVER_ be positive. If the nozzle cannot reach the front corner of the bed then the point it can reach should be defined to be 0,0 and the bed size should be reduced accordingly.

I'm sorry I don't understand. In my config:
`
#define X_MIN_POS 5
#define Y_MIN_POS 5
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE + 10
#define Y_MAX_POS Y_BED_SIZE + 10
`

With this I wanted to indicate:

- Print bed starts at +5mm offset from the position of endstops. 
- Printer han extended travel limit of +10mm in X/Y direction 
- With extending X/Y_MAX_POS I can probe additional row in UBL

If I understand you correctly, it should be:
`
#define X_MIN_POS -5
#define Y_MIN_POS -5
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE + 10
#define Y_MAX_POS Y_BED_SIZE + 10
`
Or did I misunderstood?

It is extremly confusing since every other function of the printer works as expected.

ManuelMcLure, 2021-07-29:
Correct. `X_MIN_POS`/`Y_MIN_POS` determine the position of the end stops with respect to 0,0, which is defined as the front left corner of the printable area (which may or may not be the whole bed). If you need to move 5mm to the right and back to reach the front left corner of the printable area, then `X_MIN_POS` and `Y_MIN_POS` should be -5. Take a look at https://manuelmclure.github.io/ConfiguringLeveling.html which is my attempt to clarify this.

bergstar, 2021-07-30:
Confirming working - this is not a bug. The usual case of user error and persistent RTFM.

closed 2021-07-30 by bergstar