MarlinFirmware/Marlin #22460: [BUG] SKR v1.3 and 2209s: Extruder stepper completely unresponsive

GHbubba77, 2021-07-29:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

SKR v1.3 and 2209s on a Ender 4 platform

X, Y and Z axes work perfectly, homing and moving as expected.  Extruder stepper completely unresponsive to manual movements.

All hardware checked: 
- extruder stepper responds if temporarily connected to X/Y/Z, cabling confirmed working
- 2209's work if moved in alternate X/Y/Z/E sockets, all jumper setting all correct.

Numerous suggested combinations tried:

- SQUARE_WAVE_STEPPING,  spreadcycle -NIL
- SQUARE_WAVE_STEPPING,  spreadcycle, linear advance -NIL
- both of above & #define MINIMUM_STEPPER_PULSE 2 - NIL

Can the heater warning be an issue, even if you confirm and move to the movement interface?

Any possible combination that's been overlooked?

### Bug Timeline

New

### Expected behavior

Extruder should move!

### Actual behavior

Extruder utterly unresponsive!

### Version of Marlin Firmware

Bugfix 2.0.x (July 2021)

### Printer model

Ender 4

### Electronics

SKR 1.3

### Your Slicer

Cura

### Host Software

SD Card (headless)

roel8032, 2021-07-29:
You can not manualy turn the extruder motor, if the hotend themperature is to low.

See also file configuration.h 

/**
 * Prevent extrusion if the temperature is below EXTRUDE_MINTEMP.
 * Add M302 to set the minimum extrusion temperature and/or turn
 * cold extrusion prevention on and off.
 *
 * *** IT IS HIGHLY RECOMMENDED TO LEAVE THIS OPTION ENABLED! ***
 */
#define PREVENT_COLD_EXTRUSION
#define EXTRUDE_MINTEMP 170

GHbubba77, 2021-07-29:
I was just disabling those when I read your reply!  Now the stepper responds as expected...

That said, what's the point of having an manual move override choice/warning if it does nothing??

thisiskeithb, 2021-07-29:
As noted above, disable/comment out `PREVENT_COLD_EXTRUSION` if you want to move your extruder when your hotend is cold.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-07-29 by thisiskeithb