MarlinFirmware/Marlin #22462: [BUG] Manual stepper-control "Hotend tool cold" warning screen pointless and misleading

GHbubba77, 2021-07-29:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

The manual stepper-control "Hotend tool cold" warning screen pointless and misleading.

I just spent many hours chasing a "fault" that did not exist!  If you try to manually control the extruder stepper you get a helpful, valid warning:

"Hotend tool cold"

with TWO choices:

"BACK" and "Proceed"

"Proceed" allows to the user to "control" the stepper with the zero motion!  WTF is the point of "Proceed" if it does nothing??

I assumed (as would any normal person) that "Proceed" is a confirmation by the user:

"Yes, I understand this warning, but I have a need to manually control this stepper..."  

At this point, hotend temperature is irrelevant; the user has made a clear and obvious choice and may be troubleshooting, or simply building a printer and would like to ensure all axes are moving as expected.

### Bug Timeline

OLD

### Expected behavior

I expect an explicit warning override to allow the function.

### Actual behavior

Function is blocked regardless of explicit warning override.

### Version of Marlin Firmware

2.0.x bugfix

### Printer model

Ender 4

### Electronics

SKR 1.3

### Your Slicer

Cura

thisiskeithb, 2021-07-29:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/22460

thisiskeithb, 2021-07-29:
Please don't open multiple issues for the same thing.

closed 2021-07-29 by thisiskeithb