MarlinFirmware/Marlin #22463: [BUG] BLTouch getting inconsistent readings.

EternalLionZeke, 2021-07-30:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I've been trying to get my BLTouch to function properly for the past few weeks, and I've been in and out of forums trying just about anything that could help but I'm at wit's end here asking for help.

I followed Teaching Tech's (https://www.youtube.com/watch?v=neS7lB7fCww&ab_channel=TeachingTech) BLTouch instalation guide while upgrading my stock ender 3 to a 4.2.7 main board, with the BLTouch unit, all of the electronics are functioning correctly as far as I can tell until I get to the G29 command itself.

I increased the probe amount from 9 total points to 25 in hopes of it sorting itself out, instead:

Running {G29 T1} outputs this after probing:
`Bed Height Topography:
Recv:    +--- BACK --+
Recv:    |           |
Recv:  L |    (+)    | R
Recv:  E |           | I
Recv:  F | (-) N (+) | G
Recv:  T |           | H
Recv:    |    (-)    | T
Recv:    |           |
Recv:    O-- FRONT --+
Recv:  (0,0)
Recv:  +1.38780 +1.23530 +1.11780 +1.18280 +1.23030
Recv:  +1.21530 +0.88280 +0.74280 +0.63280 +0.36030
Recv:  +0.24530 +0.24030 +0.25280 +0.30530 +0.17280
Recv:  +0.05780 -0.27720 -0.57970 -0.83970 -1.10720
Recv:  -2.02220 -1.87970 -1.69470 -1.46470 -1.39720
Recv: 
Recv: X:219.98 Y:218.67 Z:8.76 E:0.00 Count X:17600 Y:17483 Z:4074`

When I run [G29 T1] again, IMEDIATELY after the first set of probes with no changes, this is what is output:
`Recv: Bed Height Topography:
Recv:    +--- BACK --+
Recv:    |           |
Recv:  L |    (+)    | R
Recv:  E |           | I
Recv:  F | (-) N (+) | G
Recv:  T |           | H
Recv:    |    (-)    | T
Recv:    |           |
Recv:    O-- FRONT --+
Recv:  (0,0)
Recv:  +1.43730 +1.28730 +1.16230 +1.07230 +1.12230
Recv:  +1.26230 +0.93230 +0.62480 +0.51730 +0.24480
Recv:  +0.13730 +0.12730 +0.14480 +0.19230 +0.05980
Recv:  -0.05020 -0.39270 -0.68770 -0.78520 -1.05520
Recv:  -1.64520 -1.50020 -1.47020 -1.40020 -1.33770
Recv: 
Recv: X:219.97 Y:218.66 Z:8.03 E:0.00 Count X:17599 Y:17483 Z:3710`

As you can see, there is no consistency in how the ABL system is reporting.
This result is observed in Bilinear and Linear ABL modes.

Help? T_T

### Version of Marlin Firmware

2.0.x Bugfix Branch

### Printer model

Ender 3

### Electronics

4.2.7 Main Board Upgrage

### Add-ons

BLTouch

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/6905145/Configs.zip)

thinkyhead, 2021-07-30:
What does `M48` tell you?

EternalLionZeke, 2021-07-30:
Working on getting an `M48` output. Noticed a weird thing, after enabling `Z_MIN_PROBE_REPEATABILITY_TEST` it caused my `G28` to not be homing to the center of the bed anymore, is this expected behavior?

EternalLionZeke, 2021-07-30:
Send: M48
Recv: M48 Z-Probe Repeatability Test
[...]
Recv: Finished!
Recv: Mean: 1.501750 Min: 0.640 Max: 2.237 Range: 1.597
Recv: Standard Deviation: 0.485504
Recv: 
Recv: X:65.00 Y:96.50 Z:10.00 E:0.00 Count X:5200 Y:7720 Z:4000
Recv: ok
[...]

thinkyhead, 2021-07-30:
Your probe is highly inaccurate according to `M48`. Do you have another probe that you can test?

EternalLionZeke, 2021-07-30:
Unfortunately I do not.

jimberg98, 2021-07-30:
Slow down your probing speed. The accuracy of the probe is based on the maximum distance the probe must be depressed before triggering which is 0.005 mm. It's just a switch with a hair trigger. It doesn't return numbers. The firmware moves the bed toward the probe while it's polling to see if the probe triggers. Once it detects that trigger, it notes where it was at Z in relation to the last z home while moving at the time of detecting a trigger. If the bed is moving too fast, additional steps could pass until it detects the trigger. I cut the speed of my probing in half and it it is within spec when I run the repeatability test. I get really consistent results when I run G29. My mainboard is much faster than the Ender 3 mainboard, so  you will have to experiment.

My Z_PROBE_FEED_RATE_FAST is  set to 2*60. The drawback is that it takes longer to probe the bed. Considering how long printing takes, I'd rather be accurate than fast when it comes to ABL.

thinkyhead, 2021-07-31:
Closing, as this is not actionable.

closed 2021-07-31 by thinkyhead

EternalLionZeke, 2021-07-31:
Slowing it down somehow made the results worse.

EternalLionZeke, 2021-07-31:
Send: M48
Recv: M48 Z-Probe Repeatability Test
[...]
Recv: Finished!
Recv: Mean: 1.917000 Min: 0.640 Max: 2.875 Range: 2.235
Recv: Standard Deviation: 0.731576
Recv: 
Recv: X:65.00 Y:96.50 Z:10.00 E:0.00 Count X:5200 Y:7720 Z:4000
Recv: ok

thinkyhead, 2021-07-31:
Did your BLTouch come with a spare pin?

jimberg98, 2021-07-31:
> Did your BLTouch come with a spare pin?

What do you mean by "spare pin"? My connector has 5 pins and there is a dedicated port on the main board. Or do you mean the probing pin? Do you know which version you have? Is it the v3.1?

It's very possible that your bltouch could be defective. It doesn't move freely for some reason. If the bed is moving slower, it may have less momentum to trigger it against some sort of friction. See if speeding it up improves things. If it does, You should probably just get a new sensor.

thinkyhead, 2021-07-31:
Some BLTouch probes come with a spare probe pin, in case the pin gets damaged from crashing into the bed. Older BLTouch probes had a metal pin, but newer ones use a plastic pin.