MarlinFirmware/Marlin #22465: [BUG] (bug summary) Marlin                                   2.0.9.1 Fysect Dgus

andrefcarvalho, 2021-07-30:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

if you compile the marlin with (//#define DGUS_LCD_UI_FYSETC) the result is successful, however if you remove the // it gives an error when compiling

### Bug Timeline

new, no problem with marlin 020005

### Expected behavior

successfully compiled

### Actual behavior

Compiling .pio\build\FYSETC_F6\src\src\libs\stopwatch.cpp.o
Marlin\src\lcd\extui\dgus\fysetc\DGUSScreenHandler.cpp: In static member function 'static void DGUSScreenHandler::DGUSLCD_SD_FileSelected(DGUS_VP_Variable&, void*)':
Marlin\src\lcd\extui\dgus\fysetc\DGUSScreenHandler.cpp:47:22: error: 'filelist' was not declared in this scope
     if (touched_nr > filelist.count()) return;
                      ^~~~~~~~
Marlin\src\lcd\extui\dgus\fysetc\DGUSScreenHandler.cpp:47:22: note: suggested alternative: 'filepos_t'
     if (touched_nr > filelist.count()) return;
                      ^~~~~~~~
                      filepos_t
Marlin\src\lcd\extui\dgus\fysetc\DGUSScreenHandler.cpp:48:10: error: 'filelist' was not declared in this scope
     if (!filelist.seek(touched_nr)) return;
          ^~~~~~~~
Marlin\src\lcd\extui\dgus\fysetc\DGUSScreenHandler.cpp:48:10: note: suggested alternative: 'filepos_t'
     if (!filelist.seek(touched_nr)) return;
          ^~~~~~~~
          filepos_t
Marlin\src\lcd\extui\dgus\fysetc\DGUSScreenHandler.cpp:50:9: error: 'filelist' was not declared in this scope
     if (filelist.isDir()) {
         ^~~~~~~~
Marlin\src\lcd\extui\dgus\fysetc\DGUSScreenHandler.cpp:50:9: note: suggested alternative: 'filepos_t'
     if (filelist.isDir()) {
         ^~~~~~~~
         filepos_t
Marlin\src\lcd\extui\dgus\fysetc\DGUSScreenHandler.cpp:59:55: error: 'filelist' was not declared in this scope
       dgusdisplay.WriteVariable(VP_SD_Print_Filename, filelist.filename(), VP_SD_FileName_LEN, true);
                                                       ^~~~~~~~
Marlin\src\lcd\extui\dgus\fysetc\DGUSScreenHandler.cpp:59:55: note: suggested alternative: 'filepos_t'
       dgusdisplay.WriteVariable(VP_SD_Print_Filename, filelist.filename(), VP_SD_FileName_LEN, true);
                                                       ^~~~~~~~
                                                       filepos_t
Marlin\src\lcd\extui\dgus\fysetc\DGUSScreenHandler.cpp: In static member function 'static void DGUSScreenHandler::DGUSLCD_SD_StartPrint(DGUS_VP_Variable&, void*)':
Marlin\src\lcd\extui\dgus\fysetc\DGUSScreenHandler.cpp:69:10: error: 'filelist' was not declared in this scope
     if (!filelist.seek(file_to_print)) return;
          ^~~~~~~~
Marlin\src\lcd\extui\dgus\fysetc\DGUSScreenHandler.cpp:69:10: note: suggested alternative: 'filepos_t'
     if (!filelist.seek(file_to_print)) return;
          ^~~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pio\build\FYSETC_F6\src\src\lcd\extui\dgus\fysetc\DGUSScreenHandler.cpp.o] Error 1
======================================================================================================================================= [FAILED] Took 23.46 seconds =======================================================================================================================================
Environment    Status    Duration
-------------  --------  ------------
FYSETC_F6      FAILED    00:00:23.464
================================================================================================================================== 1 failed, 0 succeeded in 00:00:23.464 ==================================================================================================================================PS E:\13-3D_PRINTER\soft\Marlin-2.0..9.1\Marlin-2.0.x> echo "done" >C:\Users\andre\AppData\Local\Temp\ipc
PS E:\13-3D_PRINTER\soft\Marlin-2.0..9.1\Marlin-2.0.x>

### Steps to Reproduce

built the marlin with #define DGUS_LCD_UI_FYSETC uncoment

### Version of Marlin Firmware

2.0.9.1

### Printer model

custom

### Electronics

Fysetc f6 and Fys touch  lcd

### Host Software

SD Card (headless)

ellensp, 2021-07-30:
@andrefcarvalho The  error: 'filelist' was not declared in this scope is fixed in my latest PR. Which was aimed at fixing up DGUS_LCD_UI_ORIGIN, but also fixed several DGUS_LCD_UI_FYSETC and DGUS_LCD_UI_HIPRECY issues
https://github.com/MarlinFirmware/Marlin/pull/22464

andrefcarvalho, 2021-07-30:
How to resolve this issue?

ellensp, 2021-07-30:
apply the PR, or wait a few days till its added to Marlin bugfix

Or add the missing code you need from https://github.com/MarlinFirmware/Marlin/pull/22464/files#diff-884fc0d0371fb11b304eb95cb5fa6fc00804de791fc9963d9594d9f7d72144df

most of which is cosmetic except adding  "static ExtUI::FileList filelist;" 
 to Marlin/src/lcd/extui/dgus/fysetc/DGUSScreenHandler.cpp

andrefcarvalho, 2021-07-30:
now its give me this error

`Marlin\src\lcd\extui\dgus\fysetc\DGUSScreenHandler.cpp: In static member function 'static void DGUSScreenHandler::DGUSLCD_SD_ResumePauseAbort(DGUS_VP_Variable&, void*)':
Marlin\src\lcd\extui\dgus\fysetc\DGUSScreenHandler.cpp:86:20: error: 'MKSLCD_SCREEN_PAUSE' was not declared in this scope
         GotoScreen(MKSLCD_SCREEN_PAUSE);
                    ^~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\extui\dgus\fysetc\DGUSScreenHandler.cpp:86:20: note: suggested alternative: 'DGUSLCD_SCREEN_UNUSED'
         GotoScreen(MKSLCD_SCREEN_PAUSE);
                    ^~~~~~~~~~~~~~~~~~~
                    DGUSLCD_SCREEN_UNUSED
*** [.pio\build\FYSETC_F6\src\src\lcd\extui\dgus\fysetc\DGUSScreenHandler.cpp.o] Error 1
======================================================================================================================================= [FAILED] Took 7.77 seconds ======================================================================================================================================= 

Environment    Status    Duration
-------------  --------  ------------`

ellensp, 2021-07-30:
yip...   I don't have access to that lcd so haven't had a chance to do much with it...

replace with MKSLCD_SCREEN_PAUSE DGUSLCD_SCREEN_SDPRINTMANIPULATION in Marlin\src\lcd\extui\dgus\fysetc\DGUSScreenHandler.cpp line 86 

and see whats next

andrefcarvalho, 2021-07-30:
> yip... I don't have access to that lcd so haven't had a chance to do much with it...
> 
> replace MKSLCD_SCREEN_PAUSE with DGUSLCD_SCREEN_SDPRINTMANIPULATION in Marlin\src\lcd\extui\dgus\fysetc\DGUSScreenHandler.cpp line 86
> 
> and see whats next

success, another fix for marlin, i going to upload the code and test everything

ellensp, 2021-07-30:
I suspect It will be stuck on bootscreen. if it is dont panic, as the DGUS_LCD_UI_ORIGIN had same issue. but can try same fix if it does

andrefcarvalho, 2021-07-30:
yes it is stuck on bootcreen

ellensp, 2021-07-30:
ok hold on...

andrefcarvalho, 2021-07-30:
ok

ellensp, 2021-07-30:
you need to make these changes https://github.com/MarlinFirmware/Marlin/pull/22464/commits/aa50ddaf94ada3c5504cc24b13fc13617767d42d

ellensp, 2021-07-30:
I'm working on bugfix, so line numbers are out 1 or 2, but hopefully you can work out what to add

andrefcarvalho, 2021-07-30:
passed the boot screen, but part of the lcd functions do not work, for example, moving axes, setting temperatures or using SD files

[image]

andrefcarvalho, 2021-07-30:
I've been comparing the DGUSDisplayDef.h file with the old one and everything seems to be the same

ellensp, 2021-07-30:
Sadly without access to that lcd I can't assist further...   I've added the patches so far to my current PR

andrefcarvalho, 2021-07-30:
thanks, i'll try to see where the problem comes from, in this built i added the communication by uart for the tmc2208, it could be the cause of the problem, i'll have to test

ellensp, 2021-07-30:
If you get a chance, could you take screen shots and put boxes around areas that don't work. That would be useful to know

andrefcarvalho, 2021-07-30:
i see this on fysect wiki, but i thing marlin it´s diferent files now 

https://wiki.fysetc.com/How_to_Configure_TMC2208_on_FYSETC-F6-Board/

andrefcarvalho, 2021-07-30:
probably a board problem https://forum.fysetc.com/d/4-fysetc-f6-with-tmc-2208-marlin2-0-uart-and-extruder-not-working/8

thinkyhead, 2021-07-31:
The patch PR was merged. Please close once all the Marlin-caused issues are fixed.

closed 2021-07-31 by andrefcarvalho