MarlinFirmware/Marlin #22469: [BUG] Sensorless Homing TMC5160 SPI Double Core XY/Hbot

AidanHunter, 2021-07-31:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I Have 4 motors, double x and double y , with inverted directions on x2 and y2 , using TMC5160s with SPI. 

When I try to home , the toolhead continuously grinds against the limits of the frame. I've tried a wide array of sensitivities for the homing, and none seem to change the behavior. What is more strange is that when I try to home in just X, it will first go to the limit in the X direction, but then starts to move towards the Y direction, continuously running into the frame at the Y extremity. 

I've reviewed loads of forums and none of the solutions listed there have helped, when I enable double endstops, the toolhead doesnt really move.

### Bug Timeline

New big, as soon as I tried to do a sensorless home

### Version of Marlin Firmware

bugfix-2.0.x (can't currently check, but downloaded in last 6 months)

### Printer model

H-bot/custom

### Electronics

SKR 1.4 Turbo, TMC 5160 Stepper Drivers

### Host Software

Pronterface

### Additional information & file uploads

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/6910553/Configurations.zip)

thinkyhead, 2021-07-31:
Have you tried playing around with the `*_CURRENT_HOME` values? Using a different level of current during homing can often remedy this sort of thing.

AidanHunter, 2021-07-31:
> Have you tried playing around with the `*_CURRENT_HOME` values? Using a different level of current during homing can often remedy this sort of thing.

Yup, I've tried an array of values, I tried 1/3, 1/2 ,even tried setting the homing current of X2, and Y2 to zero

thinkyhead, 2021-07-31:
Have you tried `IMPROVE_HOMING_RELIABILITY`? Have you tried playing around with your belt tension?

AidanHunter, 2021-07-31:
I've played around with belt tension but haven't tried homing reliability, I will try it thanks, would there be anything with dual x and Y motors that could be causing the issue as well?

thinkyhead, 2021-07-31:
Yes, I'm sure that sensorless homing is more challenging with dual steppers. I've contemplated adding some kind of sensorless homing calibration to Marlin, but it would be a days-long project. The general idea is to have the axes move at different accelerations while monitoring the stallguard state, bump the ends, etc., and use the collected data to pick the best threshold values.

thisiskeithb, 2021-09-22:
Duplicate of #22012

closed 2021-09-22 by thisiskeithb