MarlinFirmware/Marlin #22472: G29 S0 causes reboot on Ender 3 Max

KodeMunkie, 2021-07-31:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Using new Ender 3 Max with "3D Touch" BTouch clone.
G29 S0 reboots on the Ender 3 Max (symptoms same as #20449, #20510 does not fix). I am currently using the bugfix-2.0.x branch. 

### Bug Timeline

Installed probe for first time 2021-07-31

### Expected behavior

Mesh data to be returned.

### Actual behavior

Printer reboots and disconnects from Octoprint.

### Steps to Reproduce

G29
G500
G29 S0

### Version of Marlin Firmware

bugfix-2.0.x (NB. not run M115 yet, git cloned 2017-07-30 - see add. comments below)

### Printer model

Creality Ender 3 Max

### Electronics

Stock - 4.2.2 mainboard with clone BLTouch "3D Touch"

### Host Software

OctoPrint

### Additional information & file uploads

Configurations and logs to follow in about 4-6 hours (it's 8am here, I've had 4 hours sleep...).
If there's anything else you'd like me to provide from the printer or try too please add it now and I'll include it when I include the logs.
(I'm a dev so can tweak source if pointed in right direction).

KodeMunkie, 2021-07-31:
On a related note... although it should not reboot part of this may be user error.
Because I previously used manual levelling I have been using G29 S0 to retrieve the mesh report as per [https://marlinfw.org/docs/gcode/G029-mbl.html](https://marlinfw.org/docs/gcode/G029-mbl.html)
G29 S0 on bilinear will set the travel speed instead as per [https://marlinfw.org/docs/gcode/G029-abl-bilinear.html(https://marlinfw.org/docs/gcode/G029-abl-bilinear.html)
Is "M420 T0" the preferred way to get the mesh topology?

KodeMunkie, 2021-08-02:
Is it possible to triage this bug please @thinkyhead (somehow I managed to add it without the bug label)?

thinkyhead, 2021-08-04:
I have pushed a patch to `bugfix-2.0.x` that should prevent `S0` from trashing the probing feedrate. Please test to see if it fixes the crashing issue.

KodeMunkie, 2021-08-11:
This fixes the crashing issue but starts an ABL from the point the probe is at resulting in the probe crashing off the end of the bed if (for example) z homed in the middle before G29 S0 called.

closed 2022-04-12 by github-actions[bot]