MarlinFirmware/Marlin #22481: [BUG] Auto bed leveling overcompensating resulting in squished prints.

MakeMake00, 2021-08-01:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hello, I have been using Marlin for years without issues. Your work is amazing.

Recently I started experimenting with Bilinear ABL but I'm experiencing some strange behaviors. I'm unable to tell if this is a bug or if I'm doing something wrong.

The leveling procedure works fine and the first layer prints perfectly but then the first 10 mm of the print are squished (object on the right). I disabled `ENABLE_LEVELING_FADE_HEIGHT` and this error propagated to the whole height of the object.

Considering that my bed is pretty flat already I was able to print the same object skipping the `G29` and it printed perfectly (object on the left) ruling out any mechanical or other configuration issues.

I spent days looking everywhere about this but I found nothing.

[image]

[config.zip](https://github.com/MarlinFirmware/Marlin/files/6912639/config.zip)

### Expected behavior

Considering the small gaps to compensate, no visible deformation of the part.

### Actual behavior

Severe squishing of the first 10mm of every printed parts.

### Version of Marlin Firmware

2.0.9.1 / bugfix-2.0.x

### Printer model

Custom Prusa Like

### Electronics

SKR V1.3

### Add-ons

Inductive Probe on spring steel sheet

### Your Slicer

Cura

### Host Software

SD Card (headless)

thinkyhead, 2021-08-04:
It is strange if _**disabling**_ `ENABLE_LEVELING_FADE_HEIGHT` has that effect. But, I suspect your mesh is strange and may have oversized values due to a bad probe Z offset.

What does the output of `M420 V` look like at the time when you are seeing this issue?

MakeMake00, 2021-08-05:
I may have to clarify. With `ENABLE_LEVELING_FADE_HEIGHT` set to `true`, only the first 10mm of the print are squished. If I disable it, the entirety of the object gets deformed. I think that part is working as intended.

I started a print and run the M420 V command. This is the result. 

```
Bilinear Leveling Grid:
0      1      2
0 -0.093 +0.010 +0.170
1 -0.172 +0.060 +0.235
2 -0.012 +0.102 +0.035
echo:Bed Leveling ON
echo:Fade Height 10.00
```

The bed isn't perfect but I don't think it justifies losing 2 mm of height on a 20mm print.

I thought about the Z probe offset. If it was set incorrectly, shouldn't the extruder be either printing in mid-air or crashing on the bed? The first layer is perfect. Am I misinterpreting this setting?

thinkyhead, 2021-08-08:
I cannot account for layers getting squished due to some flaw in the leveling compensation logic. But, try switching to UBL and see if you get the same effect.

- Is your Z acceleration very high? That can cause the Z axis to fail to move as far as it should when commanded.
- Do you have any settings which would allow the Z stepper motor to go to sleep during the printing of a layer?

MakeMake00, 2021-08-10:
I'll test UBL and I'll review my accelerations as soon as possible and I'll report back. Thank you.

MakeMake00, 2021-08-15:
Hello, I can confirm that there is no bug. The problems were related to thermal issues but considering it always presented itself in the exact same way I didn't think about this possibility.

Thank you for your assistance. I'll close the issue.

closed 2021-08-15 by MakeMake00