MarlinFirmware/Marlin #22488: [FR] Expand multi-axis support to 9 non-extruder axes

DerAndere1, 2021-08-02:
### Is your feature request related to a problem? Please describe.

Currently Marlin supports up to LINEAR_AXES 6. I heared of a custom printer that is built with more than 6 axes for which Marlin cannot be used at the moment. Implementation was started at https://github.com/DerAndere1/Marlin/pull/49 but that is outdated and we could not make it work so far. See  and related issue at https://github.com/DerAndere1/Marlin/issues/47

### Are you looking for hardware support?

Would be great to support all stepper drivers that are supported currently. Support for displays is optional for now.

### Describe the feature you want

 It would be nice to expand this to up to 9 fully fledged non-extruder axes. To see what places have to change in code, see the addition of 6 axis support (https://github.com/MarlinFirmware/Marlin/pull/19112).

ellensp, 2021-08-02:
Can I respectfully request that 6 axis gets finished first?

Ie G2/G3 work as expected with 6 axis
Also machines types other than Cartesian work with 6 axis.  In particular CORE(XY|YZ|YX| etc)

DerAndere1, 2021-08-02:
I explicitly do NOT assign myself to the task of audding 9 axis support. Anyone who Does should indeed thoroughly test and Improve things for 3-6 axes, first.

thinkyhead, 2021-08-04:
Truly man, 'tis a great task.

DerAndere1, 2021-08-20:
7th axis (internally called M_AXIS) is working now with BED_LEVELING_3POINT, so you can now test `XYZUVW A E` or more axes. An interested person can try to rebase the code from my branch https://github.com/DerAndere1/Marlin/tree/9axis_pull onto https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

ellensp, 2022-04-01:
Since this was just merged, we can close this

closed 2022-04-01 by ellensp