MarlinFirmware/Marlin #22490: [FR] multi-axis/6-axis support for CORE platforms

mikkelam, 2021-08-02:
### Is your feature request related to a problem? Please describe.

I love the new linear axes, but it seems to be unimplemented for CORE platforms

### Describe the feature you want

Have `LINEAR_AXES > 4` work with COREXY and so on

Vertabreak, 2021-08-09:
working with mikkelam i got marlin to compile we await testing on his hardware.

thinkyhead, 2021-08-09:
I've just pushed a patch to `bugfix-2.0.x` to get compilation to pass. Please test at your earliest convenience and let us know if more patches need to be made for your particular setup.

Fjederhaek, 2021-08-12:
Working with mikkelam IRL. The patch compiles without errors or warnings, but we are still not able to move our A and B axis.
We are able to read the expected values from our tmc2130 drivers with M122.

Fjederhaek, 2021-08-20:
I have made some changes to planner which fixes the issue on our end.
I have created a pull request