MarlinFirmware/Marlin #22491: [BUG] Compile error if default_envs = mks_robin_nano_v3_usb_flash_drive_msc

McDAK33, 2021-08-02:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

In fact, the essence of the error is in the title. For several weeks I can not compile the firmware for MKS Robin NANO V3 in msc configuration.
[image]

### Bug Timeline

Several weeks.

### Steps to Reproduce

In platformio.ini, define default_envs = mks_robin_nano_v3_usb_flash_drive_msc.

### Version of Marlin Firmware

bugfix-2.0.x (2021-08-02)

### Electronics

MKS Robin NANO V3

ellensp, 2021-08-03:
The error messages you list is normally because you have a serial port set to 0. Further up is the message
"error: #error "SERIAL_PORT must be from 1 to 6. You can also use -1 if the board supports Native USB."

McDAK33, 2021-08-03:
> Сообщения об ошибках, которые вы перечисляете, обычно связаны с тем, что для последовательного порта установлено значение 0 (или другой недопустимый номер). Далее идет сообщение
> «error: #error» SERIAL_PORT должен быть от 1 до 6. Вы также можете использовать -1, если плата поддерживает Native USB ».

I have been using the MKS Robin nano v3 board for over half a year and have always used SERIAL_PORT -1. Now I have the firmware 2.0.9.1 from "2021-06-27" working. The problem appeared in the latest versions.

ellensp, 2021-08-03:
No Configs so having to guess at what you may have set.

ellensp, 2021-08-03:
Give this new build env a try in ini/stm32f4.ini 
It compiles. but cant test at present.

```
[env:mks_robin_nano_v3_usb_flash_drive_msc]
platform          = ${common_stm32.platform}
extends           = env:mks_robin_nano_v3_usb_flash_drive
build_flags       = ${env:mks_robin_nano_v3_usb_flash_drive.build_flags}
                    -DUSBD_USE_CDC_MSC
build_unflags     = -DUSBD_USE_CDC

```

McDAK33, 2021-08-03:
I did not attach the configuration files because the error occurred with minimal changes to the base configuration, concerning only the board type.
After making the changes you suggested in the stm32f4.ini file, the compilation finished without errors. I will try the firmware on my printer soon.
Thank you very much.

closed 2021-08-07 by ellensp