MarlinFirmware/Marlin #22492: [FR] How to ask to add a new board to marlin?

Papipapito, 2021-08-02:
### Is your feature request related to a problem? Please describe.

I would like to ask for the Zonestar manufacturer's ZM3E4 board to be added, I have managed to compile it in marlin 2.0.7, but the manufacturer does not support the latest version of marlin

Sorry for my english

### Are you looking for hardware support?

Zonestar Printer Z9M4 , with Board ZM3E4

Chip STM32F103VC_ZM3E4_USB

Source code marlin 2.0.7

https://github.com/ZONESTAR3D/Z9/blob/main/Z9M4/ZM3E4/Sourcecode/Z9M4_ZM3E4_V1_5_0.zip

### Describe the feature you want

Add support for this board in last version of marlin

### Additional context

Board with 9 drivers, powerful to mount  a Idex printer

ellensp, 2021-08-03:
Can you try my PR?

Papipapito, 2021-08-03:
Hello

Thank you very much, I have verified it, I have the v1 of the ZONESTAR ZM3E4, but it has compiled well, and uploaded,  the board detects the drivers, moves motors, check temperatures, serial port work

If you give me a week I will finish verifying everything, since now I have the printer a little disassembled and I have to connect the motors and extruders

Tx

ant0nyk1ng, 2021-08-04:
> Hello
>
> Thank you very much, I have verified it, I have the v1 of the ZONESTAR ZM3E4, but it has compiled well, and uploaded, the board detects the drivers, moves motors, check temperatures, serial [...]

I have a working version of 2.0.7.2 and 2.0.9.1. 

I contacted the mate that helped me to port the settings from 207 as I couldn't work out with the new ini file layout.

My setup is for and IDEX.

closed 2021-08-05 by thinkyhead