MarlinFirmware/Marlin #22495: [FR] (Please acept distinct_E_factors with Mixing extruder)

rsdcampos, 2021-08-02:
### Is your feature request related to a problem? Please describe.

I need use my diamond nozzle with my printer, but I have distincts step motors and extruders, but Marlis dont let use distinct e factors and mixing extruder in sametime.

### Are you looking for hardware support?

I have a prusa like printer with Fysetc S6 motherboard and a diamond (3 color mixing) nozzle,

### Describe the feature you want

When I tried to build the Marlin with MIXING_EXTRUDER  and DISTINCT_E_FACTORS together, this error appears --> error: #error "MIXING_EXTRUDER can't be used with DISTINCT_E_FACTORS. But you may use SINGLENOZZLE with DISTINCT_E_FACTORS."
I see that I could use Single nozzle with 3 extruders and distinct e factor, but this way I miss the mixing feature that is very useful and I don't know why this can't be used together.

snakesrules94, 2021-08-03:
Hello , this configuration was ok in 2.0.7.2 if i'm right . After this version, indeed , the behaviour is as you decribe .
I used to have this config for my A20M in 2.0.7.2 and now , i have to comment DISTINCT_E_FACTORS and use the same steps for both steppers ( not a big deal as they are the same in my case ).

rsdcampos, 2021-08-03:
So this is not a feature request, but a feature return...
In first time I thought it was a bug, but the error log let clear that it is purposely, because of it I opened a FR.

thinkyhead, 2021-08-03:
This is intentional, to allow the handling of mixing at the steps level instead of at some higher level that would invoke more float maths. So, if a rewrite of this feature is wanted, someone will have to undertake it as a new project.

rsdcampos, 2021-08-04:
Ok, I understood, maybe some day... Thanks anyway.

thinkyhead, 2021-08-06:
Duplicate of #8145

closed 2021-08-06 by thinkyhead