MarlinFirmware/Marlin #22499: [BUG] Commit 0060dbc49e3 incorrectly marks endstops as absent, breaking certain safety checks.

marciot, 2021-08-03:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Commit 0060dbc49e3 changed the definition of HAS_X_MIN, HAS_X_MAX, HAS_Y_MIN, HAS_Y_MAX, HAS_Z_MIN, HAS_Z_MAX, etc so that it only is true when homing to that endstop:

Old definition:

```
#if _HAS_STOP(X,MIN)
  #define HAS_X_MIN 1
#endif
```

New definition:

```
#if BOTH(X_HOME_TO_MIN, USE_XMIN_PLUG) && _HAS_STOP(X,MIN)
  #define HAS_X_MIN 1
#endif
```

This causes problem for certain features. For example, when homing to Z_MIN, if Z_HOMING_HEIGHT is set, the toolhead will be raised prior to being lowered. This can potentially cause the toolhead to hit the opposite axis limit if the toolhead is up top. Prior to this commit, the upwards move would properly be terminated if Z_MAX was triggered; after this commit, the printer behaves as if that endstop does not exist and the axis is allowed to crash. 

### Bug Timeline

Today

### Expected behavior

Endstops should be enabled when present, even if they are not used for homing.

### Actual behavior

Endstops are disable when not used for homing.

### Version of Marlin Firmware

bugfix-2.0.x

marciot, 2021-08-03:
Also, certain UIs, notably FTDI Eve ExtUI, have a screen that shows the state of the printer's endstops. These screens now only show the endstops which are homed to, while previously they would report all endstops which are enabled.

thinkyhead, 2021-08-03:
I have been attempting to get these flags corrected so that no code is compiled for endstop pins that are not assigned to some function, such as interrupting the current move when a defined endstop switch is triggered. This is so that endstop pins may be used for other things without overlapping with endstop checks.

thinkyhead, 2021-08-03:
> Prior to this commit, the upwards move would properly be terminated if Z_MAX was triggered

That is not a formal or intentional behavior, but a side-effect. Either we can add a new option to enable that behavior, or a machine can be wired so that min and max Z endstops both trigger the same pin.

marciot, 2021-08-04:
> I have been attempting to get these flags corrected so that no code is compiled for endstop pins that are not assigned to some function, such as interrupting the current move when a defined endstop switch is triggered.

"Not assigned to some function" is the operative phrase. The way you assign endstop pins the function of acting as an endstops is by defining `USE_{which}_PLUG`. The documentation says this specifically:

```
// Specify here all the endstop connectors that are connected to any endstop or probe.
```

https://github.com/MarlinFirmware/Configurations/blob/354f671b3a853a3faf1664daa5ffdb9dfe499957/config/default/Configuration.h#L766

Therefore, if `USE_{which}_PLUG` is defined, then `HAS_{which}` must be true, because the user has assigned an endstop function to that pin. Right now, as you have written the conditionals, Marlin disregards the user's choice and instead enables the pins only if they are used for homing.

> That is not a formal or intentional behavior, but a side-effect. Either we can add a new option to enable that behavior, or a machine can be wired so that min and max Z endstops both trigger the same pin.

I don't think this is a side effect. My understanding is that if ENDSTOP_INTERRUPTS_FEATURE is enabled, then Marlin will stop motion any time an endstop is triggered, regardless of what is happening. My expectation is that any endstops which have `USE_{which}_PLUG` enabled would then cause the printer to stop, regardless of whether they are used for homing.

marciot, 2021-08-04:
To summarize:

Old behavior:

```
#if _HAS_STOP(X,MIN)
  #define HAS_X_MIN 1
#endif
```

Current behavior:

```
#if BOTH(X_HOME_TO_MIN, USE_XMIN_PLUG) && _HAS_STOP(X,MIN)
  #define HAS_X_MIN 1
#endif
```

* Disregards user's choice
* Breaks ENDSTOP_INTERRUPTS_FEATURE
* 
Proposed behavior:

```
#if ENABLED(USE_XMIN_PLUG) && _HAS_STOP(X,MIN)
  #define HAS_X_MIN 1
#endif
```

* Gives user the choice as to which endstop pins are used as endstops.
* Pins can be freed for other functions by *not* defining `USE_{which}_PLUG`

marciot, 2021-08-05:
@thinkyhead : So I just found out this change breaks the X axis leveling behavior I have on some printers, as it runs the printer into the top of the Z stop and relies on the ENDSTOP_INTERRUPTS_FEATURE to stop the printer from attempting to auto-destruct.

I think this change to longstanding Marlin behavior is bound to cause problems for people who currently depend on it.

InsanityAutomation, 2021-08-05:
I agree with marcio that many machines used this as an over travel to stop extraneous motion and prevent damage in the case of a layer shift or any other adverse event.

thinkyhead, 2021-08-06:
Patched, maybe. Please test the current code.

closed 2021-08-07 by thinkyhead