MarlinFirmware/Marlin #22501: [BUG] CONTROLLER_FAN_PIN in Configuration_adv.h is overridden by CONTROLLER_FAN_PIN in pins_BTT_SKR_PRO_common.h

discip, 2021-08-03:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

After https://github.com/MarlinFirmware/Marlin/commit/eb3ad3e4fe5a914921fb3878ccfaeb1905773d51 the option of asigning a custom **`CONTROLLER_FAN_PIN`** in **`Configuration_adv.h`** is not considered anymore, like it was priviously.

### Version of Marlin Firmware

latest bugfix-2.0.x

### Additional information & file uploads

<details>
  <summary>
    <b>Terminal output</b>
  </summary>

```
Processing BIGTREE_SKR_PRO (platform: ststm32@~12.1; board: marlin_BigTree_SKR_Pro; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/marlin_BigTree_SKR_Pro.html
PLATFORM: ST STM32 (12.1.1) > STM32F407ZG (192k RAM. 1024k Flash)
HARDWARE: STM32F407ZGT6 168MHz, 192KB RAM, 1MB Flash
DEBUG: Current (stlink) External (jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 4.10900.200819 (1.9.0)
 - framework-cmsis 2.50501.200527 (5.5.1)
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 11 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <TMCStepper> 0.7.1      
|   |-- <SoftwareSerial> 1.0
|   |-- <SPI> 1.0
|-- <SPI> 1.0
|-- <Servo> 1.1.2
|-- <EEPROM> 2.0.1
|-- <SoftwareSerial> 1.0
|-- <IWatchdog> 1.0.0
|-- <Wire> 1.0
Building in release mode
Compiling .pio\build\BIGTREE_SKR_PRO\FrameworkArduinoVariant\PeripheralPins.c.o
Compiling .pio\build\BIGTREE_SKR_PRO\FrameworkArduinoVariant\variant.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\FrameworkArduino\HardwareTimer.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\FrameworkArduino\Print.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\FrameworkArduino\RingBuffer.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\FrameworkArduino\USBSerial.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\FrameworkArduino\VirtIOSerial.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\FrameworkArduino\WInterrupts.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\FrameworkArduino\WMath.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\FrameworkArduino\WSerial.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\FrameworkArduino\WString.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\FrameworkArduino\abi.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\FrameworkArduino\avr\dtostrf.c.o
Compiling .pio\build\BIGTREE_SKR_PRO\FrameworkArduino\board.c.o
Compiling .pio\build\BIGTREE_SKR_PRO\FrameworkArduino\hooks.c.o
Compiling .pio\build\BIGTREE_SKR_PRO\FrameworkArduino\itoa.c.o
Compiling .pio\build\BIGTREE_SKR_PRO\FrameworkArduino\main.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\FrameworkArduino\new.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\FrameworkArduino\pins_arduino.c.o
Compiling .pio\build\BIGTREE_SKR_PRO\FrameworkArduino\stm32\OpenAMP\libmetal\device.c.o
[... 326 lines cut ...]
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfigPre.h:56,
                 from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\STM32\eeprom_sdcard.cpp:28:
Marlin\src\HAL\STM32\../../inc/../../Configuration_adv.h:501: note: this is the location of the previous definition
  501 |   #define CONTROLLER_FAN_PIN PE5         // Set a custom pin for the controller fan
      |
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\eeprom_sdcard.cpp:28:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:1912:6: error: #error "You cannot set E0_AUTO_FAN_PIN equal to CONTROLLER_FAN_PIN."
 1912 |     #error "You cannot set E0_AUTO_FAN_PIN equal to CONTROLLER_FAN_PIN."
      |      ^~~~~
*** [.pio\build\BIGTREE_SKR_PRO\src\src\HAL\STM32\eeprom_flash.cpp.o] Error 1
*** [.pio\build\BIGTREE_SKR_PRO\src\src\HAL\STM32\eeprom_sdcard.cpp.o] Error 1
=================================================================================== [FAILED] Took 33.22 seconds ===================================================================================

Environment      Status    Duration
---------------  --------  ------------
BIGTREE_SKR_PRO  FAILED    00:00:33.221
============================================================================== 1 failed, 0 succeeded in 00:00:33.221 ============================================================================== 
```
</details>

[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/6926273/Configuration_adv.txt)

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/6926274/Configuration.txt)

thisiskeithb, 2021-08-03:
Change https://github.com/MarlinFirmware/Marlin/blob/444992c12fa8c6fb6b8fd3f759173259d7875345/Marlin/src/pins/stm32f4/pins_BTT_SKR_PRO_common.h#L303-L305

to:

```cpp
#if ENABLED(USE_CONTROLLER_FAN) && HOTENDS < 2
  #ifndef CONTROLLER_FAN_PIN
    #define CONTROLLER_FAN_PIN              PE6   // Fan2
  #endif
#else
```

discip, 2021-08-03:
@thisiskeithb
That was fast! 😃
Will definitely do!

thisiskeithb, 2021-08-03:
> Should I create a PR for that?

I can put one in if that resolves your issue.

discip, 2021-08-03:
Yes please.

thinkyhead, 2021-08-03:
Patched.

closed 2021-08-03 by thinkyhead

thisiskeithb, 2021-08-03:
This was handled directly by @thinkyhead  in https://github.com/MarlinFirmware/Marlin/commit/6a25e4e56f0863e3abee50cad35e3ef89ed270df.

thisiskeithb, 2021-08-03:
Oops.!