MarlinFirmware/Marlin #22502: [BUG] bug in commit #22377 for BTT GTR board

mlprod35, 2021-08-03:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Without this patch GTR board does not work for me

```
------------------------------ ini/stm32f4.ini -------------------------------
index f9c16cf455..e30dae1d05 100644
@@ -166,6 +166,7 @@ extends            = stm32_variant
 board              = marlin_BigTree_GTR_v1
 board_build.offset = 0x8000
 build_flags        = ${stm32_variant.build_flags} -DSTM32F407IX
+extra_scripts      = ${stm32_variant.extra_scripts}
 
 #
 # Bigtreetech GTR V1.0 (STM32F407IGT6 ARM Cortex-M4) with USB Flash Drive Support
```

### Bug Timeline

after commit 22377 

### Version of Marlin Firmware

2.0

thinkyhead, 2021-08-03:
Strange, because the `extends` line should bring in the `extra_scripts` according to PlatformIO documentation. If that doesn't work, it implies that there is a glitch in PlatformIO.

tpruvot, 2021-08-03:
rmm i remember problems like that too, maybe when you extend at 2 levels.. or maybe a pre: / post: prefix issue ?

tpruvot, 2021-08-03:
may be an old bug, all i seen recently is trying to extend a key which is not in the extended env... breaks the build on start

mlprod35, 2021-08-04:
> Strange, because the `extends` line should bring in the `extra_scripts` according to PlatformIO documentation. If that doesn't work, it implies that there is a glitch in PlatformIO.

I use PlatformIO => Core 5.1.1·Home 3.3.4, if that can help

thinkyhead, 2021-08-06:
Unable to reproduce using PlatformIO 5.2.0a8 or 5.2.0b1. Downgraded to 5.1.1 and also cannot reproduce. The scripts are included and executed in all cases here.

closed 2021-08-07 by thinkyhead