MarlinFirmware/Marlin #22505: [BUG] When Dual Z motors are running off 2 different steppers drivers, 1 motor only turns half as much as the other

Sparhawk402, 2021-08-04:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I have set 
#define X_DRIVER_TYPE  TMC2208
#define Y_DRIVER_TYPE  TMC2208
#define Z_DRIVER_TYPE  TMC2208
#define E0_DRIVER_TYPE TMC2208
along with
#define Z2_DRIVER_TYPE TMC2208
#define E1_DRIVER_TYPE TMC2208

and set #define NUM_Z_STEPPER_DRIVERS 2
I have this set in config.adv.h
#if AXIS_IS_TMC(Z2)
    #define Z2_CURRENT      650 
    #define Z2_CURRENT_HOME 400 
    #define Z2_MICROSTEPS    16 
    #define Z2_RSENSE         0.11
    #define Z2_CHAIN_POS     -1
I have the same numbers for Z and for E1

When I flash the firmware and try to lower or raise the bed, 1 of the z axis motors is only turning half as much
I have tried to add e-steps for Z2 or E1 by modifying the line:
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 800, 93 } to #define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 800, 93 [800] }
but i get an error when compiling, there are too many numbers defined.

### Bug Timeline

When I installed a skr 1.3 mobo and set both z motors to a separate stepper drive

### Expected behavior

I expected the two Z motors to turn in sync

### Actual behavior

1 of the motors only turns half as much as the other, which then leads to a tipped bed

### Steps to Reproduce

1. Define Z2 as stepper driver tmc2208
2. Enabled Dual Z alignment
3. Flash firmware
4. Try and move bed down or up

### Version of Marlin Firmware

2.0.9.1 and bugfix-2.0.x

### Printer model

Crealty Ender 5 Plus

### Electronics

Skr 1.3 Motherboard

### Add-ons

n/a

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[configfiles.zip](https://github.com/MarlinFirmware/Marlin/files/6928600/configfiles.zip)

ellensp, 2021-08-04:
Incorrect syntax in #define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 800, 93 [800] }

Needs a extra prameter, ie and number after a comma.

#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 800, 93, 800 }

ellensp, 2021-08-04:
you need #define Z2_DRIVER_TYPE TMC2208 not #define E1_DRIVER_TYPE TMC2208

This only way this can't work is if you have faulty driver. Or have left jumper installed where they shouldn't be 
Enable #define Z2_DRIVER_TYPE TMC2208
retest
Swap drivers and retest,  does the fault move, yes its a driver issue.

sjasonsmith, 2021-08-09:
I suggest you reproduce the problem, then run `M122` and examine the output. It should report two Z motors. Make sure they share the same value for `msteps`.

If they have different values, then it may be incorrect jumpers as @ellensp suggested, or some other type of TMC communication issue.

closed 2021-08-09 by thinkyhead