MarlinFirmware/Marlin #22506: [BUG] UBL G29 multiple probing not improving accuracy

ajanky, 2021-08-04:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Setup is a custom large format machine with custom electronics. The printer is working. ZPROBE is an inductive sensor with clear filtered signal and consistent results (M48: Mean: -3.165667 Min: -3.169 Max: -3.163 Range: 0.006, Standard Deviation: 0.002630).
I wondered about relatively wide varying probe values and tried to improve accuracy with multiple probing (x2, x3) but this has shown a strange behavior. Seems like **only the first probing is a "real" measurement, the 2nd/3rd (at the same position) is always very close to the theoretical value**. See Logic Analyzer measurement attached.

**Background**: since its "large format" (ca 700x700mm) a warped bed is very likely, 2mm between edges is realistically. UBL can manage this (proven prints). However, sometimes probing values are way off. You hear this on the Z-Motor when its moving much shorter to the ZProbe than usual. I wanted to identify the problem and found this with ,ultiple probing.

G29 T mesaurements are: 
Recv:     ( 10,470)              (630,470)
Recv:         0       1       2       3
Recv:  3 | -3.933  -2.289  -2.015  -3.203
Recv:    |
Recv:  2 | -4.114  -2.778  -2.593  -3.513
Recv:    |
Recv:  1 | -3.585  -2.975  -2.958  -3.593
Recv:    |
Recv:  0 |[-2.797] -2.990  -3.253  -3.548
Recv:         0       1       2       3
Recv:     ( 10, 10)              (630, 10)

Probing step count 'til trigger of a triple probing run was:
#1: 1914 - 2399 - 2399
#2: 2411 - 2403 - 2399
#3: 2599 - 2418 - 2398
#4: 2629 - 2404 -2399
#5: 2439 - 2399 - 2403
...

### Bug Timeline

tested on current version, very likely on older versions

### Expected behavior

with multiple probing 2/3 very close measurements

### Actual behavior

first measurement seems real distance, 2nd / 3rd measurement very close to Z_CLEARANCE value

### Steps to Reproduce

**G29 P1 T** and count Z stepper pulses in to-probe direction until probe triggers - should be multiple times very close

### Version of Marlin Firmware

2.0.9.1

### Printer model

custom

### Electronics

custom LPC1768

### Additional information & file uploads

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/6930268/configs.zip)
[image]

thinkyhead, 2021-08-06:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

ajanky, 2021-08-06:
done. Debug info attached + DSView logic analyzer file. 
- multiple probing set to "3"
[debug triple probing2.txt](https://github.com/MarlinFirmware/Marlin/files/6945816/debug.triple.probing2.txt)
[TripleProbing_DSLogic-la-210806-152251.zip](https://github.com/MarlinFirmware/Marlin/files/6945854/TripleProbing_DSLogic-la-210806-152251.zip)

thinkyhead, 2021-08-07:
> Download Marlin bugfix-2.0.x to test with the latest code.

According to your log you are not running `bugfix-2.0.x` but `2.0.9.1`.

> Debug info attached

It looks like the probe is simply returning almost the same exact value on each probe, indicating that you simply have a very consistent probe. A test with `M48` should return a standard deviation pretty close to 0.

Your probe's Z offset could be improved. The probe result closest to 0 is -1.8. So you should subtract 2 from your probe Z offset to try to and get all your points close to 0.

ajanky, 2021-08-11:
HMMM. Yes probe is pretty precise, see M48 results in the "Bug Description" 
I'm currently questioning the logic of the multiple probing:

- In the picture at probing point # 1 the bed moves up (or probe down) and triggers after -say- 2000 steps. value stored. Bed moves down [clearance] steps (2400 for me), moves up again, get trigger at 2399, stores , etc. - calculating average.

- after last probing it moves down [clearance] again and moves over to probing point # 2. 

- Bed has is bend here and is closer to Z-Probe. When probing =  moving up it triggers at 1500 steps, value stored. Moves down [clearance], moves up [clearance] + triggers a.s.o. 

- only the first probing gives info about the bed warp, all other masurements are only [clearance] values and diluting the accuracy

or am I totally on the wrong track?

[image]