MarlinFirmware/Marlin #22510: [BUG] Geeetech A20M Filament feeding motors do not work - Marlin 2.0.9.1

olekpiwowar, 2021-08-04:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Latest release
PlatformIO
2.0.9.1

Geeetech A20M Filament feeding motors do not work - Marlin 2.0.9.1

I installed the latest version of Marlin 2.0.9.1 using the latest configuration for my Geeetech A20M printer.
Installation went smoothly using Visual Studio Code.

Everything works fine except the motors that provide filament. **Motors do not start.**
Please help.

### Bug Timeline

New

### Expected behavior

Of course, I expected proper working :)

### Actual behavior

Two motors feeding two different filaments do not work. They do not start when using the filament replacement option and when trying to print.

### Version of Marlin Firmware

2.0.9.1

### Printer model

Geeetech A20M

### Electronics

Stock

### Add-ons

No

### Your Slicer

Cura

snakesrules94, 2021-08-05:
Hello,
both stepper are not working or only second one ? In that case , have a look here : https://github.com/MarlinFirmware/Marlin/issues/22166

olekpiwowar, 2021-08-05:
> Hello,
> both stepper are not working or only second one ? In that case , have a look here : #22166

Both stepper are not working.

thinkyhead, 2021-08-06:
Would you be able to test some `bugfix-2.0.x` commits to narrow this down?

The strategy is to find a commit from some point in the "distant" past where the feature works. Then, test a commit from halfway between that date and today…. And then you keep going to the commit half-way in between your "known to work" commit and your "known to be broken" commit until you find the exact day where it broke.

If you started from a point 256 commits in the past, it would take no more than 8 tests to find the exact commit that broke it.

olekpiwowar, 2021-08-06:
> Would you be able to test some `bugfix-2.0.x` commits to narrow this down?
>
> The strategy is to find a commit from some point in the "distant" past where the feature works. Then, test a commit from [...]

Is it about uploading older versions of Merlin, and checking which motors have stopped working?

thinkyhead, 2021-08-06:
That's the idea. It is a way that users with no coding skills can still discover the likely cause.

Meanwhile, I will test the current Marlin on my A20M and see what I can figure out.

olekpiwowar, 2021-08-06:
> That's the idea. It is a way that users with no coding skills can still discover the likely cause.
> 
> Meanwhile, I will test the current Marlin on my A20M and see what I can figure out.

I'll come back from work and try different versions. Thanks for your commitment to the topic.

thinkyhead, 2021-08-07:
I've added a small patch to `bugfix-2.0.x` so give the latest code a try to see if it is any better.

olekpiwowar, 2021-08-08:
> I've added a small patch to `bugfix-2.0.x` so give the latest code a try to see if it is any better.

Thank you very much. I check and it seems to be working properly.

closed 2021-08-08 by thisiskeithb