MarlinFirmware/Marlin #22512: [BUG] SINGLENOZZLE breaks fan speed control

basilfx, 2021-08-04:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

PR #19152 [always](Temperature::set_fan_speed) sets `fan = 0` if `#define SINGLENOZZLE`. That means that if we have a [chamber fan](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/module/temperature.cpp#L1464) (or any other fan), it will not work, and FAN0 is always changed for any invocation of `Temperature::set_fan_speed`.

This was introduced by #19152.

### Bug Timeline

This was introduced by #19152

### Expected behavior

Fan PWM control (e.g. chamber) should work with `#define SINGLENOZZLE`.

### Actual behavior

- When changing the enclosure temperature, it modifies FAN0
- Any fan control (from the menu) always changes FAN0.

### Steps to Reproduce

1. Configure Marlin 2.0.9.1 with `#define SINGLENOZZLE` and a `FAN2_PIN` for chamber cooling.

### Version of Marlin Firmware

2.0.9.1

### Printer model

AM8

### Electronics

SKR Pro V1.1

thinkyhead, 2021-08-06:
Now patched! Please test `bugfix-2.0.x` and if the issue is fixed then this can be closed.

basilfx, 2021-08-07:
This fixes it for me. Thanks!

commit 526b6cdf (2021-08-06): 🐛 Fix fan index for Singlenozzle, chamber fan

Fixes #22512
Followup to #19152, #19519
commit fefde2a6 (2021-08-09): 🐛 Fix fan index for Singlenozzle, chamber fan

Fixes #22512
Followup to #19152, #19519
closed 2021-08-07 by basilfx