MarlinFirmware/Marlin #22513: DWIN SET for Marlin Bug fix 2.0 for CR-10S PRO V2? I cant find it?

sparksplur, 2021-08-05:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Cant find the Marlin dwin set for bug fix 2.0 cr10s pro v2, ive been looking for a couple hours

### Expected behavior

Finding DWIN SET

### Version of Marlin Firmware

Newest bug fix 2.0

### Printer model

cr10s pro v2

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

Cura

### Additional information & file uploads

Just trying to find the DWIN Set so i can flash the LCD for the bug fix 2.0 for cr10s pro v2

thisiskeithb, 2021-08-05:
It doesn't look like we have a config for the CR-10S Pro, so a DWIN set wouldn't be available: https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality

Someone will need to contribute a config & DWIN files.

thinkyhead, 2021-08-06:
The link to `DWIN_SET` above applies to the Ender-3 V2, and that is the only Creality DWIN screen currently supported. If there are differences in other screens, we will add those whenever they are contributed by somebody in the world.

closed 2021-08-06 by thinkyhead

thisiskeithb, 2021-08-06:
> The link to DWIN_SET above applies to the Ender-3 V2

Sorry, but your edit to my comment was confusing and that's not where the link was supposed to go, so I removed my comment.

InsanityAutomation, 2021-08-06:
https://github.com/InsanityAutomation/Marlin/tree/CrealityDwin_2.0 The code here will land in head at some point. There is a 7z archive with the dwin_set folder.