MarlinFirmware/Marlin #22518: [BUG] Zero extruders is broken

IvanBayan, 2021-08-05:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I tried to compile bugfix-2.0.x branch with EXTRUDERS set to 0 (commit id 35df24e1cbf5b71166580f28389a7c4bd7f54120), but compilation failed.
It works with 1 year old  release, but now I got errors during compilation:
```In file included from Marlin/src/HAL/AVR/../../inc/MarlinConfig.h:49:0,
                 from Marlin/src/HAL/AVR/HAL_SPI.cpp:34:
Marlin/src/HAL/AVR/../../inc/SanityCheck.h:3198:1: error: static assertion failed: DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Should be 3+0)
 static_assert(COUNT(sanity_arr_1) <= DISTINCT_AXES, "DEFAULT_AXIS_STEPS_PER_UNIT has too many elements." _EXTRA_NOTE);
 ^
Marlin/src/HAL/AVR/../../inc/SanityCheck.h:3205:1: error: static assertion failed: DEFAULT_MAX_FEEDRATE has too many elements. (Should be 3+0)
 static_assert(COUNT(sanity_arr_2) <= DISTINCT_AXES, "DEFAULT_MAX_FEEDRATE has too many elements." _EXTRA_NOTE);
 ^
Marlin/src/HAL/AVR/../../inc/SanityCheck.h:3212:1: error: static assertion failed: DEFAULT_MAX_ACCELERATION has too many elements. (Should be 3+0)
 static_assert(COUNT(sanity_arr_3) <= DISTINCT_AXES, "DEFAULT_MAX_ACCELERATION has too many elements." _EXTRA_NOTE);
 ^
In file included from Marlin/src/HAL/AVR/../../inc/MarlinConfig.h:49:0,
                 from Marlin/src/HAL/AVR/HAL.cpp:24:
Marlin/src/HAL/AVR/../../inc/SanityCheck.h:3198:1: error: static assertion failed: DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Should be 3+0)
 static_assert(COUNT(sanity_arr_1) <= DISTINCT_AXES, "DEFAULT_AXIS_STEPS_PER_UNIT has too many elements." _EXTRA_NOTE);
 ^
Marlin/src/HAL/AVR/../../inc/SanityCheck.h:3205:1: error: static assertion failed: DEFAULT_MAX_FEEDRATE has too many elements. (Should be 3+0)
 static_assert(COUNT(sanity_arr_2) <= DISTINCT_AXES, "DEFAULT_MAX_FEEDRATE has too many elements." _EXTRA_NOTE);
 ^
Marlin/src/HAL/AVR/../../inc/SanityCheck.h:3212:1: error: static assertion failed: DEFAULT_MAX_ACCELERATION has too many elements. (Should be 3+0)
 static_assert(COUNT(sanity_arr_3) <= DISTINCT_AXES, "DEFAULT_MAX_ACCELERATION has too many elements." _EXTRA_NOTE);
 ^
In file included from Marlin/src/HAL/AVR/../../inc/MarlinConfig.h:49:0,
                 from Marlin/src/HAL/AVR/MarlinSerial.cpp:39:
Marlin/src/HAL/AVR/../../inc/SanityCheck.h:3198:1: error: static assertion failed: DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Should be 3+0)
 static_assert(COUNT(sanity_arr_1) <= DISTINCT_AXES, "DEFAULT_AXIS_STEPS_PER_UNIT has too many elements." _EXTRA_NOTE);
 ^
Marlin/src/HAL/AVR/../../inc/SanityCheck.h:3205:1: error: static assertion failed: DEFAULT_MAX_FEEDRATE has too many elements. (Should be 3+0)
 static_assert(COUNT(sanity_arr_2) <= DISTINCT_AXES, "DEFAULT_MAX_FEEDRATE has too many elements." _EXTRA_NOTE);
 ^
Marlin/src/HAL/AVR/../../inc/SanityCheck.h:3212:1: error: static assertion failed: DEFAULT_MAX_ACCELERATION has too many elements. (Should be 3+0)
 static_assert(COUNT(sanity_arr_3) <= DISTINCT_AXES, "DEFAULT_MAX_ACCELERATION has too many elements." _EXTRA_NOTE);
 ^
cc1plus: warning: unrecognized command line option '-Wno-register'
cc1plus: warning: unrecognized command line option '-Wno-register'
*** [.pio/build/mega2560/src/src/HAL/AVR/HAL_SPI.cpp.o] Error 1
*** [.pio/build/mega2560/src/src/HAL/AVR/HAL.cpp.o] Error 1
cc1plus: warning: unrecognized command line option '-Wno-register'
*** [.pio/build/mega2560/src/src/HAL/AVR/MarlinSerial.cpp.o] Error 1
In file included from Marlin/src/HAL/AVR/../../inc/MarlinConfig.h:49:0,
                 from Marlin/src/HAL/AVR/Servo.cpp:55:
Marlin/src/HAL/AVR/../../inc/SanityCheck.h:3198:1: error: static assertion failed: DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Should be 3+0)
 static_assert(COUNT(sanity_arr_1) <= DISTINCT_AXES, "DEFAULT_AXIS_STEPS_PER_UNIT has too many elements." _EXTRA_NOTE);
 ^
Marlin/src/HAL/AVR/../../inc/SanityCheck.h:3205:1: error: static assertion failed: DEFAULT_MAX_FEEDRATE has too many elements. (Should be 3+0)
 static_assert(COUNT(sanity_arr_2) <= DISTINCT_AXES, "DEFAULT_MAX_FEEDRATE has too many elements." _EXTRA_NOTE);
 ^
Marlin/src/HAL/AVR/../../inc/SanityCheck.h:3212:1: error: static assertion failed: DEFAULT_MAX_ACCELERATION has too many elements. (Should be 3+0)
 static_assert(COUNT(sanity_arr_3) <= DISTINCT_AXES, "DEFAULT_MAX_ACCELERATION has too many elements." _EXTRA_NOTE);
 ^
cc1plus: warning: unrecognized command line option '-Wno-register'
*** [.pio/build/mega2560/src/src/HAL/AVR/Servo.cpp.o] Error 1
```

Probably it relaed to DerAndere1 patch.
I tried to modify config in the next way:
```
#define DEFAULT_AXIS_STEPS_PER_UNIT { 200, 100, 200 }
#define DEFAULT_MAX_FEEDRATE { 2000, 2000, 1500 }
#define DEFAULT_MAX_ACCELERATION { 700, 700, 700 }
#define AXIS_RELATIVE_MODES  {  false, false, false }
```

But got another error:
```Marlin/src/gcode/motion/G6.cpp: In static member function 'static void GcodeSuite::G6()':
Marlin/src/gcode/motion/G6.cpp:43:49: error: 'xyze_bool_t {aka struct XYZEval<bool>}' has no member named 'e'
     if (parser.seen('E')) planner.last_page_dir.e = !!parser.value_byte();
                                                 ^
At global scope:
cc1plus: warning: unrecognized command line option '-Wno-register'
*** [.pio/build/mega2560/src/src/gcode/motion/G6.cpp.o] Error 1```

With EXTRUERS defined set to 1 code compiles succesfully.

### Expected behavior

I expected code to be compiled.

### Actual behavior

Compilation fails with errors.

### Steps to Reproduce

1. #define EXTRUDERS 0
2. Try to compile

### Version of Marlin Firmware

commit 35df24e1cbf5b71166580f28389a7c4bd7f54120 of bugfix-2.x.x branch

ellensp, 2021-08-05:
It is not broken, when you remove an axis you now have to edit many things to remove the E parameters
As the error message is tell you.
#define DEFAULT_MAX_ACCELERATION      { 3000, 3000, 100, 10000 } becomes #define DEFAULT_MAX_ACCELERATION      { 3000, 3000, 100 }
#define DEFAULT_MAX_FEEDRATE          { 300, 300, 5, 25 } becomes #define DEFAULT_MAX_FEEDRATE          { 300, 300, 5}
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 400, 500 } becomes #define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 400 }
#define MANUAL_FEEDRATE { 50*60, 50*60, 4*60, 2*60 } becomes   #define MANUAL_FEEDRATE { 50*60, 50*60, 4*60 }
#define AXIS_RELATIVE_MODES { false, false, false, false } becomes #define AXIS_RELATIVE_MODES { false, false, false }

ellensp, 2021-08-05:
You have #define DIRECT_STEPPING enabled. This is having issue with 0 extruders.

ellensp, 2021-08-05:
This gets around the issue

```DIFF
diff --git a/Marlin/src/gcode/motion/G6.cpp b/Marlin/src/gcode/motion/G6.cpp
index 168dc28abd..3605c7b296 100644
--- a/Marlin/src/gcode/motion/G6.cpp
+++ b/Marlin/src/gcode/motion/G6.cpp
@@ -40,7 +40,9 @@ void GcodeSuite::G6() {
     if (parser.seen('X')) planner.last_page_dir.x = !!parser.value_byte();
     if (parser.seen('Y')) planner.last_page_dir.y = !!parser.value_byte();
     if (parser.seen('Z')) planner.last_page_dir.z = !!parser.value_byte();
-    if (parser.seen('E')) planner.last_page_dir.e = !!parser.value_byte();
+    #if EXTRUDERS > 0
+      if (parser.seen('E')) planner.last_page_dir.e = !!parser.value_byte();
+    #endif
   }

   // No index means we just set the state
diff --git a/Marlin/src/module/planner.cpp b/Marlin/src/module/planner.cpp
index 52ca76308c..32a113dde0 100644
--- a/Marlin/src/module/planner.cpp
+++ b/Marlin/src/module/planner.cpp
@@ -3051,7 +3051,7 @@ bool Planner::buffer_line(const xyze_pos_t &cart, const_feedRate_t fr_mm_s, cons
       PAGE_UPDATE_DIR(X);
       PAGE_UPDATE_DIR(Y);
       PAGE_UPDATE_DIR(Z);
-      PAGE_UPDATE_DIR(E);
+      TERN_(HAS_EXTRUDERS, PAGE_UPDATE_DIR(E));
     }

     // If this is the first added movement, reload the delay, otherwise, cancel it.
diff --git a/Marlin/src/module/stepper.cpp b/Marlin/src/module/stepper.cpp
index 0ff909d7cc..f2321a1053 100644
--- a/Marlin/src/module/stepper.cpp
+++ b/Marlin/src/module/stepper.cpp
@@ -1669,7 +1669,7 @@ void Stepper::pulse_phase_isr() {
               PAGE_SEGMENT_UPDATE(X, (b >> 6) & 0x3);
               PAGE_SEGMENT_UPDATE(Y, (b >> 4) & 0x3);
               PAGE_SEGMENT_UPDATE(Z, (b >> 2) & 0x3);
-              PAGE_SEGMENT_UPDATE(E, (b >> 0) & 0x3);
+              TERN_(HAS_EXTRUDERS, PAGE_SEGMENT_UPDATE(E, (b >> 0) & 0x3));
             } break;
             default: break;
           }
@@ -1855,7 +1855,7 @@ uint32_t Stepper::block_phase_isr() {
           PAGE_SEGMENT_UPDATE_POS(X);
           PAGE_SEGMENT_UPDATE_POS(Y);
           PAGE_SEGMENT_UPDATE_POS(Z);
-          PAGE_SEGMENT_UPDATE_POS(E);
+          TERN_(HAS_EXTRUDERS, PAGE_SEGMENT_UPDATE_POS(E));
         }
       #endif
       TERN_(HAS_FILAMENT_RUNOUT_DISTANCE, runout.block_completed(current_block));

```

IvanBayan, 2021-08-05:
>It is not broken, when you remove an axis you now have to edit many things to remove the E parameters
As I wrote in initial bug report I tried to compile with:
```#define DEFAULT_AXIS_STEPS_PER_UNIT { 200, 100, 200 }
#define DEFAULT_MAX_FEEDRATE { 2000, 2000, 1500 }
#define DEFAULT_MAX_ACCELERATION { 700, 700, 700 }
#define AXIS_RELATIVE_MODES  {  false, false, false }
```
And got an error:
```Marlin/src/gcode/motion/G6.cpp: In static member function 'static void GcodeSuite::G6()':
Marlin/src/gcode/motion/G6.cpp:43:49: error: 'xyze_bool_t {aka struct XYZEval<bool>}' has no member named 'e'
     if (parser.seen('E')) planner.last_page_dir.e = !!parser.value_byte();
```

>You have #define DIRECT_STEPPING enabled. This is having issue with 0 extruders.
No I have not:
> grep DIRECT_STEPPING Marlin/*.h
```Marlin/Configuration_adv.h:// #define DIRECT_STEPPING
Marlin/Configuration_adv.h:#if BOTH(SDSUPPORT, DIRECT_STEPPING)
```

>This gets around the issue
Will you submit it as pull request? Otherwise I can test and submit it by myself.

ellensp, 2021-08-05:
Marlin/src/gcode/motion/G6.cpp is  disabled if you don't have DIRECT_STEPPING
All the code is in #if ENABLED(DIRECT_STEPPING) block.

I cant grep your config, you didn't provide them.

IvanBayan, 2021-08-05:
My bad, I had enabled DIRECT_STEPPING and started to disable all un-necessary options when I got an error. Probably I saved gcc errors from the first attempt.
Here is a log from the latest attempt with zero extruders:
```(penv) > grep DIRECT_STEPPING Marlin/*.h
Marlin/Configuration_adv.h:// #define DIRECT_STEPPING
Marlin/Configuration_adv.h:#if BOTH(SDSUPPORT, DIRECT_STEPPING)
(penv) > platformio run -e mega2560  
Processing mega2560 (platform: atmelavr; board: megaatmega2560; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega2560.html
PLATFORM: Atmel AVR (3.0.0) > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
HARDWARE: ATMEGA2560 16MHz, 8KB RAM, 248KB Flash
DEBUG: Current (avr-stub) On-board (avr-stub, simavr)
PACKAGES: 
 - framework-arduino-avr 5.1.0 
 - toolchain-atmelavr 1.50400.190710 (5.4.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 10 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.5.0
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <Wire> 1.0
|-- <SPI> 1.0
|-- <EEPROM> 2.0
|-- <SoftwareSerial> 1.0
Building in release mode
Compiling .pio/build/mega2560/src/src/HAL/AVR/HAL.cpp.o
Compiling .pio/build/mega2560/src/src/HAL/AVR/HAL_SPI.cpp.o
Compiling .pio/build/mega2560/src/src/HAL/AVR/MarlinSerial.cpp.o
Compiling .pio/build/mega2560/src/src/HAL/AVR/Servo.cpp.o
In file included from Marlin/src/HAL/AVR/../../inc/MarlinConfig.h:49:0,
                 from Marlin/src/HAL/AVR/HAL_SPI.cpp:34:
Marlin/src/HAL/AVR/../../inc/SanityCheck.h:589:1: error: static assertion failed: AXIS_RELATIVE_MODES must contain X Y Z elements.
 static_assert(COUNT(arm) == LOGICAL_AXES, "AXIS_RELATIVE_MODES must contain " _LOGICAL_AXES_STR "elements.");
 ^
Marlin/src/HAL/AVR/../../inc/SanityCheck.h:3198:1: error: static assertion failed: DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Should be 3+0)
 static_assert(COUNT(sanity_arr_1) <= DISTINCT_AXES, "DEFAULT_AXIS_STEPS_PER_UNIT has too many elements." _EXTRA_NOTE);
 ^
Marlin/src/HAL/AVR/../../inc/SanityCheck.h:3205:1: error: static assertion failed: DEFAULT_MAX_FEEDRATE has too many elements. (Should be 3+0)
 static_assert(COUNT(sanity_arr_2) <= DISTINCT_AXES, "DEFAULT_MAX_FEEDRATE has too many elements." _EXTRA_NOTE);
 ^
Marlin/src/HAL/AVR/../../inc/SanityCheck.h:3212:1: error: static assertion failed: DEFAULT_MAX_ACCELERATION has too many elements. (Should be 3+0)
 static_assert(COUNT(sanity_arr_3) <= DISTINCT_AXES, "DEFAULT_MAX_ACCELERATION has too many elements." _EXTRA_NOTE);
 ^
In file included from Marlin/src/HAL/AVR/../../inc/MarlinConfig.h:49:0,
                 from Marlin/src/HAL/AVR/HAL.cpp:24:
Marlin/src/HAL/AVR/../../inc/SanityCheck.h:589:1: error: static assertion failed: AXIS_RELATIVE_MODES must contain X Y Z elements.
 static_assert(COUNT(arm) == LOGICAL_AXES, "AXIS_RELATIVE_MODES must contain " _LOGICAL_AXES_STR "elements.");
 ^
Marlin/src/HAL/AVR/../../inc/SanityCheck.h:3198:1: error: static assertion failed: DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Should be 3+0)
 static_assert(COUNT(sanity_arr_1) <= DISTINCT_AXES, "DEFAULT_AXIS_STEPS_PER_UNIT has too many elements." _EXTRA_NOTE);
 ^
Marlin/src/HAL/AVR/../../inc/SanityCheck.h:3205:1: error: static assertion failed: DEFAULT_MAX_FEEDRATE has too many elements. (Should be 3+0)
 static_assert(COUNT(sanity_arr_2) <= DISTINCT_AXES, "DEFAULT_MAX_FEEDRATE has too many elements." _EXTRA_NOTE);
 ^
Marlin/src/HAL/AVR/../../inc/SanityCheck.h:3212:1: error: static assertion failed: DEFAULT_MAX_ACCELERATION has too many elements. (Should be 3+0)
 static_assert(COUNT(sanity_arr_3) <= DISTINCT_AXES, "DEFAULT_MAX_ACCELERATION has too many elements." _EXTRA_NOTE);
 ^
cc1plus: warning: unrecognized command line option '-Wno-register'
cc1plus: warning: unrecognized command line option '-Wno-register'
*** [.pio/build/mega2560/src/src/HAL/AVR/HAL.cpp.o] Error 1
*** [.pio/build/mega2560/src/src/HAL/AVR/HAL_SPI.cpp.o] Error 1
In file included from Marlin/src/HAL/AVR/../../inc/MarlinConfig.h:49:0,
                 from Marlin/src/HAL/AVR/MarlinSerial.cpp:39:
Marlin/src/HAL/AVR/../../inc/SanityCheck.h:589:1: error: static assertion failed: AXIS_RELATIVE_MODES must contain X Y Z elements.
 static_assert(COUNT(arm) == LOGICAL_AXES, "AXIS_RELATIVE_MODES must contain " _LOGICAL_AXES_STR "elements.");
[... 7 lines cut ...]
Marlin/src/HAL/AVR/../../inc/SanityCheck.h:3212:1: error: static assertion failed: DEFAULT_MAX_ACCELERATION has too many elements. (Should be 3+0)
 static_assert(COUNT(sanity_arr_3) <= DISTINCT_AXES, "DEFAULT_MAX_ACCELERATION has too many elements." _EXTRA_NOTE);
 ^
cc1plus: warning: unrecognized command line option '-Wno-register'
*** [.pio/build/mega2560/src/src/HAL/AVR/MarlinSerial.cpp.o] Error 1
In file included from Marlin/src/HAL/AVR/../../inc/MarlinConfig.h:49:0,
                 from Marlin/src/HAL/AVR/Servo.cpp:55:
Marlin/src/HAL/AVR/../../inc/SanityCheck.h:589:1: error: static assertion failed: AXIS_RELATIVE_MODES must contain X Y Z elements.
 static_assert(COUNT(arm) == LOGICAL_AXES, "AXIS_RELATIVE_MODES must contain " _LOGICAL_AXES_STR "elements.");
 ^
[... cut ...]
 ^
Marlin/src/HAL/AVR/../../inc/SanityCheck.h:3212:1: error: static assertion failed: DEFAULT_MAX_ACCELERATION has too many elements. (Should be 3+0)
 static_assert(COUNT(sanity_arr_3) <= DISTINCT_AXES, "DEFAULT_MAX_ACCELERATION has too many elements." _EXTRA_NOTE);
 ^
cc1plus: warning: unrecognized command line option '-Wno-register'
*** [.pio/build/mega2560/src/src/HAL/AVR/Servo.cpp.o] Error 1
```

Configs are pasted below:
[MC.zip](https://github.com/MarlinFirmware/Marlin/files/6938458/MC.zip)

ellensp, 2021-08-05:
Provided Configs do not have modifications to 
DEFAULT_MAX_ACCELERATION
DEFAULT_MAX_FEEDRATE
DEFAULT_AXIS_STEPS_PER_UNIT 
AXIS_RELATIVE_MODES

one those are all patched you get a standard config error 
Marlin/src/HAL/AVR/../../inc/SanityCheck.h:1910:6: error: #error "USE_CONTROLLER_FAN requires a CONTROLLER_FAN_PIN. Define in Configuration_adv.h."

you have also stripped out all spaces and indenting and done weird multi line comments everywhere, making comparisons to the standard config quite impossible

you also have stuff like "git show" randomly in variables!

and you added \ to the end of comment, really confusing marlin.

ellensp, 2021-08-05:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6939629/Configuration.zip)

I have cleaned up Configuration.h and removed all the issues
I have removed all the issues from Configuration_adv.h but Have not cleaned up and its a nightmare.

You need to set CONTROLLER_FAN_PIN in Configuration.h to a pin. or disable USE_CONTROLLER_FAN

ellensp, 2021-08-05:
And other that your config issues, for which you win worst config ever award, It Compiles with extruders set to 0 without issues!

ellensp, 2021-08-05:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-08-05 by ellensp

IvanBayan, 2021-08-05:
You are too kind granting me 'worst config ever' award,  unfortunately I can't accept this. All indentation changes and line breaks was made by vscode automatically. I didn't thought that someone will compare my config with standard one, so didn't care to preserve formatting.. I've checked your config, it compiles with 0 extruders, thank you for your help. At least it helped to find #22520