MarlinFirmware/Marlin #22519: [FR] (UBL w. sample from homing)

Wallby, 2021-08-05:
### Describe the feature you want

Hello,

I thought it might be nice if UBL would (re)use the "Z homing sample" (e.g. sampled from the center of the bed), and then samples the four corners of the bed. That way 5 points are samples in total (enough to construct a mesh I think..?

thinkyhead, 2021-08-06:
Note that the Z homing point, where Z0 is established, is rarely coincident with the center of the mesh. Moreover, it is most common to home Z with a cold bed, whereas `G29` should be done on a heated bed.

Wallby, 2021-08-06:
> Note that the Z homing point, where Z0 is established, is rarely coincident with the center of the mesh.

That depends on the model of the printer. Still reusing the point can reduce the number of samples needed in G29 (unless you think that G29 should not be done "directly after G28".

NOTE: also part of this feature request is that this could potentially reduce the # samples needed. i.e. currently min # samples for UBL is 9 (min 3 per axis). With the "cross pattern" I suggested that would be only 5.

Wallby, 2021-10-02:
I've reformatted this request (changed title and description).
I.e. if one of the points required for bed leveling is already known from z homing.. skip the point in bed leveling.

Wallby, 2021-10-09:
> Note that the Z homing point, where Z0 is established, is rarely coincident with the center of the mesh. Moreover, it is most common to home Z with a cold bed, whereas `G29` should be done on a heated bed.

I personally use bed leveling to do homing as well (i.e. the option on the LCD menu always homes and then levels).

Wallby, 2021-10-09:
What I would like to be able to do (not sure if it makes sense though) is..

`#define OVERRIDE_GRID_POINT_WITH_Z0 { 1,1 } //< i.e. instead of sampling to figure out grid point 1x1, read Z0`

Wallby, 2021-10-10:
This feature (or something alike) actually became a requirement instead of a "would be nice" when my IR probe no longer detected the bed "at the right time" (i.e. now.. it crashes into the bed, because there is a hole for a thermistor at the center of the bed).

I changed where Z0 samples (-10mm offset on the X axis now), but bed leveling still samples the same point.