MarlinFirmware/Marlin #22520: [BUG] DIRECT_STEPPING fails to handle 0 extruders.

ellensp, 2021-08-05:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

DIRECT_STEPPING fails to handle 0 extruders.

### Version of Marlin Firmware

latest bugfix

### Additional information & file uploads

Was noticed when fighting with issue #22518

ellensp, 2021-08-05:
This gets around the issue

```DIFF
diff --git a/Marlin/src/gcode/motion/G6.cpp b/Marlin/src/gcode/motion/G6.cpp
index 168dc28abd..3605c7b296 100644
--- a/Marlin/src/gcode/motion/G6.cpp
+++ b/Marlin/src/gcode/motion/G6.cpp
@@ -40,7 +40,9 @@ void GcodeSuite::G6() {
     if (parser.seen('X')) planner.last_page_dir.x = !!parser.value_byte();
     if (parser.seen('Y')) planner.last_page_dir.y = !!parser.value_byte();
     if (parser.seen('Z')) planner.last_page_dir.z = !!parser.value_byte();
-    if (parser.seen('E')) planner.last_page_dir.e = !!parser.value_byte();
+    #if HAS_EXTRUDERS
+      if (parser.seen('E')) planner.last_page_dir.e = !!parser.value_byte();
+    #endif
   }

   // No index means we just set the state
diff --git a/Marlin/src/module/planner.cpp b/Marlin/src/module/planner.cpp
index 52ca76308c..32a113dde0 100644
--- a/Marlin/src/module/planner.cpp
+++ b/Marlin/src/module/planner.cpp
@@ -3051,7 +3051,7 @@ bool Planner::buffer_line(const xyze_pos_t &cart, const_feedRate_t fr_mm_s, cons
       PAGE_UPDATE_DIR(X);
       PAGE_UPDATE_DIR(Y);
       PAGE_UPDATE_DIR(Z);
-      PAGE_UPDATE_DIR(E);
+      TERN_(HAS_EXTRUDERS, PAGE_UPDATE_DIR(E));
     }

     // If this is the first added movement, reload the delay, otherwise, cancel it.
diff --git a/Marlin/src/module/stepper.cpp b/Marlin/src/module/stepper.cpp
index 0ff909d7cc..f2321a1053 100644
--- a/Marlin/src/module/stepper.cpp
+++ b/Marlin/src/module/stepper.cpp
@@ -1669,7 +1669,7 @@ void Stepper::pulse_phase_isr() {
               PAGE_SEGMENT_UPDATE(X, (b >> 6) & 0x3);
               PAGE_SEGMENT_UPDATE(Y, (b >> 4) & 0x3);
               PAGE_SEGMENT_UPDATE(Z, (b >> 2) & 0x3);
-              PAGE_SEGMENT_UPDATE(E, (b >> 0) & 0x3);
+              TERN_(HAS_EXTRUDERS, PAGE_SEGMENT_UPDATE(E, (b >> 0) & 0x3));
             } break;
             default: break;
           }
@@ -1855,7 +1855,7 @@ uint32_t Stepper::block_phase_isr() {
           PAGE_SEGMENT_UPDATE_POS(X);
           PAGE_SEGMENT_UPDATE_POS(Y);
           PAGE_SEGMENT_UPDATE_POS(Z);
-          PAGE_SEGMENT_UPDATE_POS(E);
+          TERN_(HAS_EXTRUDERS, PAGE_SEGMENT_UPDATE_POS(E));
         }
       #endif
       TERN_(HAS_FILAMENT_RUNOUT_DISTANCE, runout.block_completed(current_block));

```

thinkyhead, 2021-08-06:
Zero extruders support is always ongoing. Please submit patches wherever they are found wanting.

DerAndere1, 2021-10-06:
This, plus extension to more than 3 LINEAR_AXES would be ideal.

closed 2021-12-18 by github-actions[bot]