MarlinFirmware/Marlin #22526: SKR 1.3 MAX31865  H SPI MINTEMP triggered HEATHER_ID:  E0

pietja5, 2021-08-05:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

MAX31865 is not working   Marlin-bugfix-2.0.x 2021-08-05 
SKR1.3 +TFT35 
Case 1 
conf.h :
#define SERIAL_PORT -1
#define SERIAL_PORT_2 0 
#define MOTHERBOARD BOARD_BTT_SKR_V1_3 
#define TEMP_SENSOR_0 -5
#define MAX31865_SENSOR_OHMS_0      100   // (Ω) Typically 100 or 1000 (PT100 or PT1000)
#define MAX31865_CALIBRATION_OHMS_0 430   // (Ω) Typically 430 for Adafruit PT100; 4300 for Adafruit PT1000

#define SDSUPPORT
//#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

conf_adv.h
  //#define SDCARD_CONNECTION LCD
MAX31865.h:
SPI_MODE1 

pins_BTT_SKR_common:

#define TEMP_0_CS_PIN   P0_16
// #ifndef TEMP_0_PIN
//   #define TEMP_0_PIN                    P0_24_A1  // A1 (T1) - (68) - TEMP_0_PIN
// #endif

   #define SDCARD_CONNECTION            ONBOARD

MAX 31865 connected to:
SDI  P0.18
CS    P0.16
CLK  P0.15
SDO P0.17 
VCC and GND 

Compilation - no errors  
file .bin >> sd card>> SKR>> restart >> no printer attached  on TFT35 

Case 2 >>
#define TEMP_SENSOR_0  1 
//#define TEMP_0_CS_PIN   P0_16
 #ifndef TEMP_0_PIN
   #define TEMP_0_PIN                    P0_24_A1  // A1 (T1) - (68) - TEMP_0_PIN
 #endif
Compilation - no errors  
file .bin >> sd card>> SKR>> restart >> READY 

Some suggestions? 

### Expected behavior

MAX 31865 work  with SKR 1.3

### Actual behavior

no comunication / freez of mainboard

### Version of Marlin Firmware

Marlin-bugfix-2.0.x

### Printer model

SKR 1.3 

### Electronics

SKR 1.3

thinkyhead, 2021-08-06:
There are some new options in `Configuration_adv.h` that you might also try adjusting…
```cpp
/**
 * Configuration options for MAX Thermocouples (-2, -3, -5).
 *   FORCE_HW_SPI:   Ignore SCK/MOSI/MISO pins and just use the CS pin & default SPI bus.
 *   MAX31865_WIRES: Set the number of wires for the probe connected to a MAX31865 board, 2-4. Default: 2
 *   MAX31865_50HZ:  Enable 50Hz filter instead of the default 60Hz.
 */
//#define TEMP_SENSOR_FORCE_HW_SPI
//#define MAX31865_SENSOR_WIRES_0 2
//#define MAX31865_SENSOR_WIRES_1 2
//#define MAX31865_50HZ_FILTER
```

pietja5, 2021-08-06:
Cof.adv.h< #define TEMP_SENSOR_FORCE_HW_SPI
#define MAX31865_SENSOR_WIRES_0 2>
Conf.h
<#define TEMP_0_CS_PIN   P0_16
// #ifndef TEMP_0_PIN
//   #define TEMP_0_PIN                    P0_24_A1  // A1 (T1) - (68) - TEMP_0_PIN
// #endif
>
Compilation - no errors
file .bin >> sd card>> SKR>> restart >> no printer attached on TFT35

thinkyhead, 2021-08-07:
You can set `SDCARD_CONNECTION` to `LCD` in your configuration instead of editing `pins_BTT_SKR_common.h`.

thinkyhead, 2021-08-07:
It looks like support for MAX Thermocouples is not set up yet for BTT SKR boards. I don't know what else would be required to make it work. Maybe you can sign on to the Marlin Discord and ask some other smart people for help. https://discord.gg/n5NJ59y

qrstu, 2021-08-08:
@pietja5 
I cold also no get working on EXT2 CS (MAX31855) BTT SKR 1.4, I used SPI Pins (doesn't exist on BTT SKR 1.3)
Try to connect to SDCARD_CONNECTION ONBOARD
Use this as help:
https://github.com/GadgetAngel/MAX31865-SKR-V1-3-GUIDE
In the last 4 weeks some changes took place on bugfix 2.0.x for MAX318155 and MAX318165 which make the setup much easier.
Check --> #if HAS_MAX_TC --> pins_BTT_SKR_common.h
#define TEMP_SENSOR_FORCE_HW_SPI --> Configuration_adv.h
#define MAX31865_SENSOR_WIRES_0 2 --> Configuration_adv.h
Regards
Martin

jinhong-, 2021-08-22:
@pietja5 I struggled for a day to get this to work as well. I noticed @GadgetAngel's guide is quite outdated as compared to the new code changes in marlin's bugfix 2.0 branch. I will share with you how I got mine working, though I have a different skr board (SKR E3 Turbo), hopefully this will help you successfully setup yours as well

Here is @GadgetAngel's [original guide for SKR 1.3](https://github.com/GadgetAngel/MAX31865-SKR-V1-3-GUIDE). I recommend reading it to look for the correct pins to use to connect your MAX31865. Note that I am using **software SPI**. Also, I believe this only works with 2 wire PT100 out of the box

Here are the changes I did to get mine working

## Changes made to pins_*.h (find your appropriate board). In here you will define the pins you are using to connect the SKR board to your MAX31865
```
#if HAS_MAX_TC
  #define TEMP_0_CS_PIN   P1_21
  #define TEMP_0_MISO_PIN P0_11
  #define TEMP_0_MOSI_PIN P0_10
  #define TEMP_0_SCK_PIN  P1_20
#endif
```
## Changes made to Configuration.h
```
#define MAX31865_SENSOR_OHMS_0      100   // (Ω) Typically 100 or 1000 (PT100 or PT1000)
#define MAX31865_CALIBRATION_OHMS_0 430   // (Ω) Typically 430 for Adafruit PT100; 4300 for Adafruit PT1000
#define LIB_MAX31865 1 // Enable custom library for MAX31865
```
## Changes made to features.ini
Add the following line under the `[features]` section
```
LIB_MAX31865                           = https://github.com/GadgetAngel/Adafruit-MAX31865-V1.1.0-Mod-M.git
```
## To enable 50hz filter
To enable 50hz filter in countries that have 50hz AC power frequency
### Changes made to Configuration_adv.h
```
#define MAX31865_50HZ_FILTER
```
### Changes made to temperature.cpp
Replace 
```
      #if defined(LIB_INTERNAL_MAX31865) && ENABLED(MAX31865_50HZ_FILTER)
        max31865_0.enable50HzFilter(1);
      #endif
```
with 
```
      #if ENABLED(MAX31865_50HZ_FILTER)
        max31865_0.enable50Hz(1);
      #endif
```

I am not sure why the codes baked into Marlin did not work. Perhaps @slowbro or @GadgetAngel can shed some light?

jinhong-, 2021-08-22:
For some reason after I put everything back together it decides to stop working again. Seems quite strange. Must have somehow fluked it. I noticed that If i pulled out the power pin and cycled the power, the temperature readings will start showing up. Somehow the max31865 needs to be reset

jinhong-, 2021-08-23:
I added a PR to update `MAX31865.cpp`, which improves the software SPI handling

slowbro, 2021-08-23:
Thanks for the PR, @jinhong-, it is very much appreciated! I added some notes on the changes there.

For @GadgetAngel's guide, lots of it is out of date; we use an internal MAX31865 library now, so the `features.ini` and `LIB_MAX31865` parts are not needed. For the 50Hz filter, only `MAX31865_50HZ_FILTER` is needed; the calls are made on the backend. See here: https://github.com/MarlinFirmware/Marlin/blob/2c45ac3c1b2acb1f98a92dd80160c560d08600a8/Marlin/src/libs/MAX31865.cpp#L230-L232

pietja5, 2021-08-24:
Hi
Unfortunately no changes on my side
Firmware download 16:20 24.08.2021 (Marlin-bugfix-2.0.x)
changes: 
platformio.ini:
`default_envs = LPC1768`
features.ini:
` - no changes `
configuration.h
```
#define SERIAL_PORT -1
#define SERIAL_PORT_2 0
#define BAUDRATE_2 250000   // Enable to override BAUDRATE

#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_BTT_SKR_V1_3
#endif

#define TEMP_SENSOR_0 -5

#define MAX31865_SENSOR_OHMS_0      100   // (Ω) Typically 100 or 1000 (PT100 or PT1000)
#define MAX31865_CALIBRATION_OHMS_0 430   // (Ω) Typically 430 for Adafruit PT100; 4300 for Adafruit PT1000

#define SDSUPPORT
```
```
#define X_DRIVER_TYPE  TMC2208
#define Y_DRIVER_TYPE  TMC2208
#define Z_DRIVER_TYPE  TMC2208
```

Configuration_adv.h
```
//#define TEMP_SENSOR_FORCE_HW_SPI
 //#define MAX31865_SENSOR_WIRES_0 2
//#define MAX31865_SENSOR_WIRES_1 2
#define MAX31865_50HZ_FILTER
```
```
#define MONITOR_DRIVER_STATUS
#define TMC_DEBUG
#define SHOW_TEMP_ADC_VALUES
#define MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED 10

 #define UTF_FILENAME_SUPPORT
 #define LONG_FILENAME_HOST_SUPPORT
 #define SDCARD_CONNECTION LCD
```
temperature.ccp: 
  ```
no changes
```
pins_BTT_SKR_common.h: 
```
#if HAS_MAX_TC
  #define TEMP_0_CS_PIN   P1_19
  #define TEMP_0_MISO_PIN P1_18
  #define TEMP_0_MOSI_PIN P1_21
  #define TEMP_0_SCK_PIN  P1_20

```
MAX31865.ccp :

`, SPI_MODE0 // CPOL0 CPHA1 ///there was error during compilation,  I pick 0 according to MAX31865-SKR-V1-3-GUIDE`

file .bin >> sd card>> SKR>> restart >> no printer attached on TFT35
Restart >> MINTEMP triggered HEATER_ID: E0

under this configuration I connec osciloscope and check CLK, MOSI/ MISO - no signal  - both sd card and max31865 
not possible to check sd card

when 

`#define TEMP_SENSOR_0 1 `
regular CLK/ MOSI/ MISO  signal  on SD CARD 
possible to check files on sd card.

GadgetAngel, 2021-08-24:
> Thanks for the PR, @jinhong-, it is very much appreciated! I added some notes on the changes there.
>
> For @GadgetAngel's guide, lots of it is out of date; we use an internal MAX31865 library now, so [...]

If @slowbro could do what @jinhong- did above and tell me what needs to change in [my guide](https://github.com/GadgetAngel/MAX31865-SKR-V1-3-GUIDE) then I will make the changes.  

I am now using Klipper for my firmware on my printers, but I will be more than glad to update [my guide](https://github.com/GadgetAngel/MAX31865-SKR-V1-3-GUIDE) to reflect the changes in the Marlin firmware if someone provides me with the details on how to change [my guide](https://github.com/GadgetAngel/MAX31865-SKR-V1-3-GUIDE).  [My repository](https://github.com/GadgetAngel/MAX31865-SKR-V1-3-GUIDE/pulls) does have a pull-request that someone can fill out (@slowbro or @jinhong- ) once the steps are known, then I can update the guide.

jinhong-, 2021-08-31:
@pietja5 which branch of marlin did you try? can you try [this version](https://github.com/jinhong-/Marlin/tree/max31865-enhancements) with your configuration? I have good success getting this to work on my SKR E3 Turbo, which is the same family as what you are using

pietja5, 2021-09-22:
Hello 
I canfirm that **MAX31865 enhancements #22682** works

closed 2022-04-12 by github-actions[bot]