MarlinFirmware/Marlin #22527: [BUG] NeoPixel not working on btt skr mini e3 v2.

00ihsan, 2021-08-06:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

When i configure NeoPixel and try to compile the code its not compiling

### Bug Timeline

today

### Expected behavior

expected to have working neopixels

### Actual behavior

not compiling for my chip

### Steps to Reproduce

1. Buy a btt skr mini e3 v2.
2. connect neopixel pins as required.
3. change marlin.
4. no compilation.

### Version of Marlin Firmware

2.0.9.1

### Printer model

Creality Ender #

rstalde, 2021-08-06:
Look for the following code-lines:

// Support for Adafruit NeoPixel LED driver
#define NEOPIXEL_LED //RST NEO
#if ENABLED(NEOPIXEL_LED)
  #define NEOPIXEL_TYPE   NEO_GRB  // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
  //#define NEOPIXEL_PIN     4     // LED driving pin
  //#define NEOPIXEL2_TYPE NEOPIXEL_TYPE
  //#define NEOPIXEL2_PIN    5
  #define NEOPIXEL_PIXELS 19       // Number of LEDs in the strip. (Longest strip when NEOPIXEL2_SEPARATE is disabled.)	//RST NEO
  #define NEOPIXEL_IS_SEQUENTIAL   // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
  #define NEOPIXEL_BRIGHTNESS 127  // Initial brightness (0-255)
  #define NEOPIXEL_STARTUP_TEST  // Cycle through colors at startup //RST NEO

The lines with the Remark "//RST NEO" are changed.
Look for the Jumper that enable the optional Step-Down-Module. You need this module if you have more than 12 (?) Neopixel.
Look on the user's guide for this board on BBT's github (https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/hardware/BTT%20SKR%20MINI%20E3%20V2.0/Hardware)

thisiskeithb, 2021-08-06:
> ### Steps to Reproduce
> 1. Buy a btt skr mini e3 v2.
> 2. connect neopixel pins as required.
> 3. change marlin.
> 4. no compilation.

@00ihsan: You have to be more specific for steps three and four.

What did you change? Please attach your configs and any other modified files.

Which environment are you compiling with?

What’s the error message?

00ihsan, 2021-08-06:
code:

// Support for Adafruit NeoPixel LED driver
#define NEOPIXEL_LED
#if ENABLED(NEOPIXEL_LED)
  #define NEOPIXEL_TYPE   NEO_RGB  // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
  //#define NEOPIXEL_PIN    PA8        // LED driving pin
  //#define NEOPIXEL2_TYPE NEOPIXEL_TYPE
  //#define NEOPIXEL2_PIN    5
  #define NEOPIXEL_PIXELS 72       // Number of LEDs in the strip. (Longest strip when NEOPIXEL2_SEPARATE is disabled.)
  #define NEOPIXEL_IS_SEQUENTIAL   // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
  #define NEOPIXEL_BRIGHTNESS 127  // Initial brightness (0-255)
  #define NEOPIXEL_STARTUP_TEST  // Cycle through colors at startup

  // Support for second Adafruit NeoPixel LED driver controlled with M150 S1 ...
  //#define NEOPIXEL2_SEPARATE
  #if ENABLED(NEOPIXEL2_SEPARATE)
    #define NEOPIXEL2_PIXELS      15  // Number of LEDs in the second strip
    #define NEOPIXEL2_BRIGHTNESS 127  // Initial brightness (0-255)
    #define NEOPIXEL2_STARTUP_TEST    // Cycle through colors at startup
  #else
    //#define NEOPIXEL2_INSERIES      // Default behavior is NeoPixel 2 in parallel
  #endif

  // Use some of the NeoPixel LEDs for static (background) lighting
  //#define NEOPIXEL_BKGD_INDEX_FIRST  0              // Index of the first background LED
  //#define NEOPIXEL_BKGD_INDEX_LAST   30              // Index of the last background LED
  #define NEOPIXEL_BKGD_COLOR { 255, 255, 255, 0 }  // R, G, B, W
  #define NEOPIXEL_BKGD_ALWAYS_ON                   // Keep the backlight on when other NeoPixels are off
#endif

output:

In file included from Marlin/src/HAL/STM32F1/../../inc/MarlinConfig.h:50:0,
                 from Marlin/src/HAL/STM32F1/HAL.cpp:30:
Marlin/src/HAL/STM32F1/../../inc/../HAL/STM32F1/inc/SanityCheck.h:43:4: error: #error "NEOPIXEL_LED (Adafruit NeoPixel) is not supported for HAL/STM32F1. Comment out this line to proceed at your own risk!"
   #error "NEOPIXEL_LED (Adafruit NeoPixel) is not supported for HAL/STM32F1. Comment out this line to proceed at your own risk!"
    ^~~~~
In file included from Marlin/src/HAL/STM32F1/../../inc/MarlinConfig.h:50:0,
                 from Marlin/src/HAL/STM32F1/HAL_SPI.cpp:32:
Marlin/src/HAL/STM32F1/../../inc/../HAL/STM32F1/inc/SanityCheck.h:43:4: error: #error "NEOPIXEL_LED (Adafruit NeoPixel) is not supported for HAL/STM32F1. Comment out this line to proceed at your own risk!"
   #error "NEOPIXEL_LED (Adafruit NeoPixel) is not supported for HAL/STM32F1. Comment out this line to proceed at your own risk!"
    ^~~~~
In file included from Marlin/src/HAL/STM32F1/../../inc/MarlinConfig.h:50:0,
                 from Marlin/src/HAL/STM32F1/MarlinSerial.cpp:25:
Marlin/src/HAL/STM32F1/../../inc/../HAL/STM32F1/inc/SanityCheck.h:43:4: error: #error "NEOPIXEL_LED (Adafruit NeoPixel) is not supported for HAL/STM32F1. Comment out this line to proceed at your own risk!"
   #error "NEOPIXEL_LED (Adafruit NeoPixel) is not supported for HAL/STM32F1. Comment out this line to proceed at your own risk!"
    ^~~~~
In file included from Marlin/src/HAL/STM32F1/../../inc/MarlinConfig.h:50:0,
                 from Marlin/src/HAL/STM32F1/Servo.cpp:25:
Marlin/src/HAL/STM32F1/../../inc/../HAL/STM32F1/inc/SanityCheck.h:43:4: error: #error "NEOPIXEL_LED (Adafruit NeoPixel) is not supported for HAL/STM32F1. Comment out this line to proceed at your own risk!"
   #error "NEOPIXEL_LED (Adafruit NeoPixel) is not supported for HAL/STM32F1. Comment out this line to proceed at your own risk!"
    ^~~~~
In file included from Marlin/src/HAL/STM32F1/../../inc/MarlinConfig.h:50:0,
                 from Marlin/src/HAL/STM32F1/SPI.cpp:43:
Marlin/src/HAL/STM32F1/../../inc/../HAL/STM32F1/inc/SanityCheck.h:43:4: error: #error "NEOPIXEL_LED (Adafruit NeoPixel) is not supported for HAL/STM32F1. Comment out this line to proceed at your own risk!"
   #error "NEOPIXEL_LED (Adafruit NeoPixel) is not supported for HAL/STM32F1. Comment out this line to proceed at your own risk!"
    ^~~~~
In file included from Marlin/src/HAL/STM32F1/dogm/../../../inc/MarlinConfig.h:50:0,
                 from Marlin/src/HAL/STM32F1/dogm/u8g_com_stm32duino_swspi.cpp:21:
Marlin/src/HAL/STM32F1/dogm/../../../inc/../HAL/STM32F1/inc/SanityCheck.h:43:4: error: #error "NEOPIXEL_LED (Adafruit NeoPixel) is not supported for HAL/STM32F1. Comment out this line to proceed at your own risk!"
   #error "NEOPIXEL_LED (Adafruit NeoPixel) is not supported for HAL/STM32F1. Comment out this line to proceed at your own risk!"
    ^~~~~
In file included from Marlin/src/HAL/STM32F1/../../inc/MarlinConfig.h:50:0,
                 from Marlin/src/HAL/STM32F1/eeprom_bl24cxx.cpp:29:
Marlin/src/HAL/STM32F1/../../inc/../HAL/STM32F1/inc/SanityCheck.h:43:4: error: #error "NEOPIXEL_LED (Adafruit NeoPixel) is not supported for HAL/STM32F1. Comment out this line to proceed at your own risk!"
   #error "NEOPIXEL_LED (Adafruit NeoPixel) is not supported for HAL/STM32F1. Comment out this line to proceed at your own risk!"
    ^~~~~
*** [.pio/build/STM32F103RC_btt_USB_maple/src/src/HAL/STM32F1/HAL.cpp.o] Error 1
*** [.pio/build/STM32F103RC_btt_USB_maple/src/src/HAL/STM32F1/MarlinSerial.cpp.o] Error 1
*** [.pio/build/STM32F103RC_btt_USB_maple/src/src/HAL/STM32F1/HAL_SPI.cpp.o] Error 1
*** [.pio/build/STM32F103RC_btt_USB_maple/src/src/HAL/STM32F1/SPI.cpp.o] Error 1
[... cut ...]
   #error "NEOPIXEL_LED (Adafruit NeoPixel) is not supported for HAL/STM32F1. Comment out this line to proceed at your own risk!"
    ^~~~~
*** [.pio/build/STM32F103RC_btt_USB_maple/src/src/HAL/STM32F1/eeprom_flash.cpp.o] Error 1
==================== [FAILED] Took 4.01 seconds ====================

Environment                Status    Duration
-------------------------  --------  ------------
STM32F103RC_btt_USB_maple  FAILED    00:00:04.008
============== 1 failed, 0 succeeded in 00:00:04.008 ==============

thisiskeithb, 2021-08-06:
From the log you just posted:

> #error "NEOPIXEL_LED (Adafruit NeoPixel) is not supported for HAL/STM32F1. Comment out this line to proceed at your own risk!"

Comment out that line to proceed at your own risk.

I'd recommend compiling with a non-maple environment since NeoPixels are not supported otherwise.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-08-06 by thisiskeithb

00ihsan, 2021-08-06:
Thanks! Neopixel works but the speaker does not work properly anymore and ui is slower.