MarlinFirmware/Marlin #22535: [BUG] .sconsign39.dlbite and .sconsign39.tmp files are not found in the directory for the Melzi Optiboot

SherEast, 2021-08-08:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I am trying to build the melzi_optiboot environment for the Sanguino ATMega1284p mainboard on my Ender 3 Pro and I cannot finish the build process because the .sconsign39.dlbite file is nowhere to be found in the melzi_optiboot file.

The script goes as follows:
PS C:\Users\[username]\Downloads\Marlin-2.0.x\Marlin-2.0.x> echo "done" >C:\Users\[username]\AppData\Local\Temp\ipc
PS C:\Users\[username]\Downloads\Marlin-2.0.x\Marlin-2.0.x> platformio run -e melzi_optiboot
Processing melzi_optiboot (platform: atmelavr; board: sanguino_atmega1284p; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/sanguino_atmega1284p.html
PLATFORM: Atmel AVR (3.3.0) > Sanguino ATmega1284p (16MHz)
HARDWARE: ATMEGA1284P 16MHz, 16KB RAM, 127KB Flash
DEBUG: Current (simavr) On-board (simavr)
PACKAGES:
 - framework-arduino-avr 5.1.0
 - toolchain-atmelavr 1.70300.191015 (7.3.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 6 compatible libraries
Dependency Graph
|-- <U8glib-HAL> 0.5.0
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <Wire> 1.0
|-- <SPI> 1.0
|-- <EEPROM> 2.0
|-- <SoftwareSerial> 1.0
Building in release mode
*** [Marlin\src\MarlinCore.cpp] C:\Users\[username]\Downloads\Marlin-2.0.x\Marlin-2.0.x\.pio\build\melzi_optiboot\.sconsign39.dblite: No such file or directory

### Bug Timeline

This seems to be a new bug

### Version of Marlin Firmware

Marlin Firmware 2.0.x

### Printer model

Creality Ender 3 Pro

### Electronics

Stock electronics

### Add-ons

Creality BL Touch Bed Leveling Sensor

### Bed Leveling

ABL Linear grid

### Your Slicer

Cura

### Host Software

SD Card (headless)

rhapsodyv, 2021-08-09:
It's just a random platformio error. Just try build again and it will work.

closed 2021-08-09 by thisiskeithb

SherEast, 2021-08-09:
I have tried building it several times and it still isn't working.

rhapsodyv, 2021-08-09:
Try reinstall PIO, or run it from vscode or cmd, instead of PS.

SherEast, 2021-08-09:
I am using VSCode. Should I still reinstall PIO?

rhapsodyv, 2021-08-09:
Try it.
Because something probably is not right in your current setup. I never saw this error show up more than once.