MarlinFirmware/Marlin #22538: [BUG] (Duplication mode in IDEX not working properly)

atharvbadkas, 2021-08-09:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hello,
I have an ender 3 pro with extended X axis(400mm).
I have installed BTT Idex kit with BTT RRF E3 V1.1 With Expansion board.

I also have configured marlin in Dual X carriage mode.
I'm attaching my config files.
Please check if I have missed something there.
Thanks

### Expected behavior

Dual colour mode works as expected after calibrating the X offset in print heads.

In duplication mode,
After print starts,X2 hotend grinds to X_max_endstop
Then motor stops rotating.

In Mirror mode,
After print start, X2 hotend grinds to X_max_endstop
Then start printing above LCD(does not come to bed at X220 location)

### Actual behavior

In duplication mode,
After print starts, X2 hotend should come to X110 location then both hotend start the print in sync.

In mirror mode,
After print starts, X2 hotend should come to X220 location and then start printing.

### Version of Marlin Firmware

V2.0.9.1 also have tried on latest bugfix version

### Printer model

Creality Ender 3 pro

### Electronics

Bigtreetech RRF E3 V1.1 with Expansion board

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

atharvbadkas, 2021-08-09:
[WhatsApp Business Documents.zip](https://github.com/MarlinFirmware/Marlin/files/6952292/WhatsApp.Business.Documents.zip)

My config files

atharvbadkas, 2021-08-10:
Hello,
Can anyone please check my config files and let me know that my configs are right or wrong?

Because I have seen exact setup like mine working normally on Marlin 2.0.6.(from https://enderidex.com/)
Thanks

thinkyhead, 2021-08-13:
The configurations look okay to me. This could certainly just be a bug. IDEX tends to be a work in progress because only a few contributors have one.

Would you be able to test some `bugfix-2.0.x` commits to narrow down when the problem started?

The strategy is to find a commit from some point in the "distant" past where the feature works. Then, test a commit from halfway between that date and today…. And then you keep going to the commit half-way in between your "known to work" commit and your "known to be broken" commit until you find the exact day where it broke.

If you started from a point 256 days in the past, it should take no more than 8 tests to find the exact day when it broke.

atharvbadkas, 2021-08-13:
Sure thing,
I can test some bugfix version from past 256 days as you mentioned.

Just tell me what files I have to change regarding the motherboard.
Files for BTT RRF E3 V1.1, for example pins or some lines in sanity check

thinkyhead, 2021-08-13:
You should just be able to use your same configuration options, maybe turning off some optional things for simplicity. The main thing is just to get IDEX enabled and test the behavior with tool-changing and make sure it moves properly in the different modes. Enable `DEBUG_LEVELING_FEATURE` and send `M111 S247` to get maximum logging to the console.

InsanityAutomation, 2021-08-13:
I got my IDEX machine back up to date and on the workbench last weekend, needed a new heater cartridge for one of the heads. Hoping to dig in this weekend with it working again after the new heater is in.

shtaXveed, 2021-09-12:
any ideas on this yet?

atharvbadkas, 2021-09-12:
One of my friend had discussed this with @InsanityAutomation on discord. He somehow made it work using Marlin 2.0.6. but not fully functional.
 He's Marlin folder(https://drive.google.com/file/d/1ZRwMiPPJaXwuuIlOZM1mU-uPcQRw3obD/view?usp=drivesdk)
I'm also working on this and trying to solve some errors I'm getting while compiling
[image]

thinkyhead, 2021-09-13:
I downloaded and compiled your folder from the Google drive, and got no errors. Try deleting your `.pio` folder and your `$HOME/.platformio/.cache` folder and build again.

thinkyhead, 2021-09-13:
> any ideas on this yet?

Someone with IDEX hardware will need to track down the exact cause. There are a lot of issues to go through and I would just be taking stabs in the dark trying to figure out the issue. But of course… please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands today. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

whoishack, 2021-09-14:
Not resolved in the latest bugfix unfortunately. Someone else please confirm but X2 for me still smashes into the endstop at the end of a home sequence.

thisiskeithb, 2021-09-16:
@InsanityAutomation: Can you look at this?

InsanityAutomation, 2021-09-16:
I got the machine back together and some other issues on it fixed a bit over a week ago intending to chase exactly this, so yes as soon as commercial commitments are done this is first on my list.

shtaXveed, 2021-09-16:
so I've been messing around a little 2.0.9 makes the right hotend ram the right endstop, instead of traveling towards center. and 2.0.8.1 makes the left hotend ram the endstop while the right hotend travels to the center. 

but if I assign duplication mode from my start gcode it will start fine

shtaXveed, 2021-09-16:
note in 2.0.8.1 it prints fine after the right hotend reaches its start position. just grinds while the right hotend is moving

imp67, 2021-09-17:
I see similar issues to shaXveed (above) using build 2.0.8.2.
The same work around in gcode avoids the issue.
E1 does not extrude if DISABLE_INACTIVE_EXTRUDER is defined.
In duplication mode the heads home, then E0 tries to move LEFT in synch with E1 at the start of the print grinding the belt. As it's already homed on the left at that point, the head alignment is maintained and the print proceeds OK from that point.
In mirror mode, the belt grinding doesn't occur, and the print is OK.

Jerronimo2001, 2021-09-24:
I have the same problem with grinding the belt moving E0 in start of duplication mode (2.0.8.2).