MarlinFirmware/Marlin #22539: [BUG] problem with serial communication with the MKS tft display in six-axis mode

jafarniknam, 2021-08-09:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

HI .
I'm using marlin 2.0 , bugfix version for my 6 axes robot (not a 3D printer ) + MKS GEN v1.4 motherboard + MKS TFT32 . TFT firmware is open source Bigtreetech . Serial port is 0 with 115200 baudrate and I'm using AUX 1 of MKS GEN v1.4 motherboard for TFT32. of course Serial communication in pronterface program established without problems in all cases .
when I define LINEAR_AXES 3 , everything is ok but when I define LINEAR_AXES 6 and set depended configs , mks tft 32 will disconnect and show "NO printer attached" message .
What tests did I take ?
Upload marlin with original configurations ( just change motherboard and baudrate) ==> everything is OK
Several baudrates and serial ports ==> "NO printer attached" message
Several pins definition for 6th axis ==> "NO printer attached" message

 You think pins mismatch/redefines?  No I'll check this out before. In the last test I did last night, I came to a strange conclusion. When I activated five axes with an extruder, the problem of serial communication with the display was also solved. Then I tested the sixth axis with an extruder and everything was right.
For now, I have to activate an additional extruder to solve this problem and use futile digital pins (Like EXP1 pins ) to define extruder driver pins.
It was strange for me, too.

I think the main problem is additional axes and cause disconnection .

If I change  #define EXTRUDERS 1 to 0 in Configuration.h  and Make the necessary changes ، Serial communication with LCD will  broken . Of course, in communicating with PC , No problem  .

thanks and sorry for my English !!

### Bug Timeline

NEW

### Version of Marlin Firmware

FIRMWARE_NAME : Marlin bugfix-2.0.x (Aug  8 2021 20:28:49) 

### Printer model

Not a 3D printer . six axes sampler robot 

### Electronics

MKS Gen v1.4 + MKS TFT32

### Additional information & file uploads

6 axis + additional extruder ( NO serial problem )
[config.zip](https://github.com/MarlinFirmware/Marlin/files/6952545/config.zip)

6 axis + 0 Extruder (Serial problem with TFT32 )
[config_2.zip](https://github.com/MarlinFirmware/Marlin/files/6952577/config_2.zip)

ellensp, 2021-08-09:
Its is most probable that the MKS TFT32 doesn't understand the idea of 0 extruders. It was designed for a 3d printer...

ellensp, 2021-08-09:
What does a real serial console say instead of the tft?

jafarniknam, 2021-08-09:
> What does a real serial console say instead of the tft? Like Pronterface or Octoprint Console

thanks for reply . NO problem with that . Serial communication with PC using Pronterface is Ok and says :"printer is online " and commands send successfully  .

ellensp, 2021-08-09:
Then issue is with tft and outside of marlin control.

closed 2021-08-09 by ellensp