MarlinFirmware/Marlin #22540: [FR] download all documentation, gcodes, as pdf or html single file package, opengpl that too

goofyseeker311, 2021-08-09:
### Is your feature request related to a problem? Please describe.

cant download all things related for offline use

### Are you looking for hardware support?

no

### Describe the feature you want

all things marlin should be simply downloadable, single button, for documentation, for offline use

### Additional context

https://marlinfw.org/meta/gcode/ and you notice the problem, no download button, intentional, eh

ellensp, 2021-08-09:
The entire thing is available for download at https://github.com/MarlinFirmware/MarlinDocumentation

goofyseeker311, 2021-08-09:
was not obvious

ellensp, 2021-08-09:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-08-09 by ellensp

goofyseeker311, 2021-08-09:
there is no binary of the documentation, reopen, please

goofyseeker311, 2021-08-09:
> The entire thing is available for download at https://github.com/MarlinFirmware/MarlinDocumentation

no the final documentation is not in there, wrong answer

ellensp, 2021-08-09:
No I will not reopen this. 

There are many reasons for this. but the most important is that this FR should be created in the MarlinDocumentation repository, not the Marlin source code repository.

ellensp, 2021-08-09:
read the readme.
https://github.com/MarlinFirmware/MarlinDocumentation/blob/master/README.md

The Documentation web site is generated from the https://github.com/MarlinFirmware/MarlinDocumentation repository.

goofyseeker311, 2021-08-09:
the main site needs to have direct full download of the documentation in the main download section of the main site, thats why Im not obeying, ever

goofyseeker311, 2021-08-09:
this that you removed the documentation from being online, and I did not have the build environment for the documentation

goofyseeker311, 2021-08-09:
simply I dont care if you have the documentation online or that I can download the source, I need the third option, locally, now

goofyseeker311, 2021-08-09:
women cannot understand law issues, even if they claim to be experts of the law